thesofproject/sof #4354: [BUG][TGL] tplg load failed on TGL chrome SDW device

keqiaozhang, 2021-06-15:
**Describe the bug**
TPLG failed to load widget GHD13.0. This is a regression and happened from last Friday. Daily Test ID:4568
This issue is caused by https://github.com/thesofproject/sof/commit/6dd44d5b4c4b380ce415ec1ad2b3ff9118ceea9f
Google hotword library is added in sof-tgl-sdw-max98373-rt5682.m4.

**To Reproduce**
check the demsg after boot

**Reproduction Rate**
100%

**Expected behavior**
tplg can be loaded successfully without errors

**Impact**
Fail to create the audio pipelines

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 57440285
SOF Branch: main
SOF Commit: 499d0ee31095
Device:  Chrome Volteer
TPLG:sof-tgl-sdw-max98373-rt5682.tplg

**Dmesg**
```
[    4.145822] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.146013] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.146033] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.146486] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.146492] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget DMIC1.IN setup complete
[    4.146501] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    4.146648] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[    4.146651] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.12.DMIC1.IN setup complete
[    4.146665] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    4.146806] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget GHD13.0
[    4.147571] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    4.147599] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    4.147604] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    4.147721] kernel: sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    4.651240] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3008 len 1
[    4.651260] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3008 count 0
[    5.155272] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3001 len 1
[    5.155287] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3001 count 0
[    5.659276] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3000 len 1
[    5.659290] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3000 count 0
[    6.163265] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3005 len 1
[    6.163278] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3005 count 0
[    6.667233] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3004 len 1
[    6.667247] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3004 count 0
[    7.172233] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3008 len 1
[    7.172251] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3008 count 0
[    7.676226] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3001 len 1
[    7.676243] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3001 count 0
[    8.123264] kernel: soundwire_intel soundwire_intel.link.1: IO transfer timed out, cmd 2 device 1 addr 2054 len 1
[    8.123277] kernel: soundwire sdw-master-1: trf on Slave 1 failed:-110 read addr 2054 count 0
[    8.179233] kernel: soundwire_intel soundwire_intel.link.0: IO transfer timed out, cmd 3 device 1 addr 3000 len 1
[    8.179252] kernel: soundwire sdw-master-0: trf on Slave 1 failed:-110 write addr 3000 count 0

``
[dmesg.txt](https://github.com/thesofproject/sof/files/6654226/dmesg.txt)

keqiaozhang, 2021-06-15:
@plbossart, can we create a new tplg(passthrough mode, like sof-tgl-sdw-max98373-rt5682-ci.m4  in tools/topology/development folder) for CI test?

lgirdwood, 2021-06-15:
@keqiaozhang I think we are out of heap. Are you building the Google hotword in to the FW or is it disabled in Kconfig ?

plbossart, 2021-06-15:
@keqiaozhang @RanderWang is this related to SoundWire topologies moving to dynamic pipelines?

keqiaozhang, 2021-06-15:
> @keqiaozhang I think we are out of heap. Are you building the Google hotword in to the FW or is it disabled in Kconfig ?
We didn't build google hotword into FW, but the tplg is changed to build with google hotword as default.  that's the gap and causes the tplg load fails on Volteer.

plbossart, 2021-06-15:
@keqiaozhang @lgirdwood @bzhg I am afraid the commit 6dd44d5b4c4b380ce415ec1ad2b3ff9118ceea9f needs to be reverted. This changed the default to topology to use a 3rd party library that is not present in the firmware that we test in the Intel/public CI.

You need to put the Google-specific topologies in a separate folder and not touch public ones.

cujomalainey, 2021-06-15:
Whoops yep apologies, that was my fault, its only a partial revert of the config lines e.g. https://github.com/thesofproject/sof/commit/6dd44d5b4c4b380ce415ec1ad2b3ff9118ceea9f#diff-5493c06560e041c10cd41f29644f38cda164adb107dafa05b9c74dc45dda08afR91

plbossart, 2021-06-17:
This seems to be fixed @keqiaozhang, can you please double-check and close?

plbossart, 2021-06-22:
No answer and no issue in last daily tests, closing.

closed 2021-06-22 by plbossart