thesofproject/sof #4376: [BUG] IPC error + DSP Panic with CNL_nocodec tests

plbossart, 2021-06-18:
**Describe the bug**

With the simple following unit test, the DSP crashes

````
export TPLG=/lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg 
/root/sof-test/test-case/check-capture.sh -d 1 -l 1 -r 3
````
[dmesg.log](https://github.com/thesofproject/sof/files/6677246/dmesg.log)

````
[   27.977098] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: open stream 3 dir 1
[   27.977105] sof-audio-pci-intel-cnl 0000:00:1f.3: period min 192 max 16384 bytes
[   27.977108] sof-audio-pci-intel-cnl 0000:00:1f.3: period count 2 max 16
[   27.977111] sof-audio-pci-intel-cnl 0000:00:1f.3: buffer max 65536 bytes
[   27.977234] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: hw params stream 3 dir 1
[   27.977237] sof-audio-pci-intel-cnl 0000:00:1f.3: generating page table for 0000000015b457ed size 0x10000 pages 16
[   27.977246] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x240000 successful
[   27.977261] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x40000 successful
[   27.977265] sof-audio-pci-intel-cnl 0000:00:1f.3: period_bytes:0x4000
[   27.977267] sof-audio-pci-intel-cnl 0000:00:1f.3: periods:4
[   27.977277] sof-audio-pci-intel-cnl 0000:00:1f.3: stream_tag 2
[   27.977287] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[   27.977441] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[   27.977445] sof-audio-pci-intel-cnl 0000:00:1f.3: widget PIPELINE.7.DMIC0.IN setup complete
[   27.977452] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[   27.977643] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[   27.977646] sof-audio-pci-intel-cnl 0000:00:1f.3: widget PCM3C setup complete
[   27.977651] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[   27.977836] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[   27.977841] sof-audio-pci-intel-cnl 0000:00:1f.3: widget BUF7.0 setup complete
[   27.977848] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[   27.978097] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[   27.978107] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   27.978483] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[   27.978486] sof-audio-pci-intel-cnl 0000:00:1f.3: widget DMIC0.IN setup complete
[   27.978491] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   27.978632] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   27.978635] sof-audio-pci-intel-cnl 0000:00:1f.3: route BUF7.0 -> PCM3C setup complete
[   27.978640] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   27.978749] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   27.978751] sof-audio-pci-intel-cnl 0000:00:1f.3: route DMIC0.IN -> BUF7.0 setup complete
[   27.978754] sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: complete pipeline PIPELINE.7.DMIC0.IN id 39
[   27.978758] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   27.979085] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   27.979093] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   27.979457] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   27.979460] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[   27.979496] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   27.979867] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[   27.981532] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 1
[   27.982050] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x2024001e successful
[   27.982057] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   27.982204] sof-audio-pci-intel-cnl 0000:00:1f.3: error: ipc error for 0x60040000 size 12
[   27.982210] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[   27.982215]  DMIC: ASoC: trigger FE cmd: 1 failed: -22
[   27.982219] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 0
[   27.982223] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   27.982337] sof-audio-pci-intel-cnl 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[   27.982343] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[   27.982346] sof-audio-pci-intel-cnl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[   27.982350]  DMIC: ASoC: trigger FE cmd: 0 failed: -22
[   27.982725] sof-audio-pci-intel-cnl 0000:00:1f.3: pcm: free stream 3 dir 1
[   27.982732] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   27.982839] sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[   27.982844] sof-audio-pci-intel-cnl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   27.982848] sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[   27.982923] sof-audio-pci-intel-cnl 0000:00:1f.3: error: runtime exception
[   27.982927] sof-audio-pci-intel-cnl 0000:00:1f.3: error: trace point 00004000
[   27.982930] sof-audio-pci-intel-cnl 0000:00:1f.3: error: panic at :0
[... 64 lines cut ...]
[         652.656224] (          13.906249) c3 idc                ......./intel/cavs/idc.c:165  INFO idc_do_cmd()
[         665.208307] (          12.552083) c3 pipe         7.39  ......./pipeline-graph.c:383  INFO pipe reset
Skipped 7592 bytes after the last statement.

````
**To Reproduce**
run the test!

**Reproduction Rate**
100% but after 2 iterations of capture. This points to an allocation/release problem

**Expected behavior**
No IPC error and DSP crash

**Impact**
show-stopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 56ab165beb0d744624fd2ee47c75b40ae13d24cd
    * SOF: af6655e7fac3c56847251553699f354df74bf6a9
2) Name of the topology file
    * Topology: sof-cnl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: UpExtreme

plbossart, 2021-06-18:
Disabling multi-core either by reverting 97bfb24d701c254279e407f3388b02fa77045a68 or using https://github.com/thesofproject/linux/pull/3003 with options snd_sof sof_debug=0x20 in /etc/modprobe.d/alsa-base.conf

Clearly a multicore issue.

plbossart, 2021-06-18:
@ranj063 I am afraid this one in for you...

marc-hb, 2021-06-22:
> With the simple following unit test, the DSP crashes

@fredoh9 ,  @XiaoyunWu6666  is `check-capture.sh -d 1 -l 1 -r 3` not part of any test plan!? @plbossart is this an unusual combination of parameters and is it required? Did you say `-r 2` or more above? 

I wonder what is the difference between:
```
    -r parameter |  --round parameter
	round count
	Default Value: 1
   ...
   -l parameter |  --loop parameter
	loop count
	Default Value: 3
```

plbossart, 2021-06-22:
@marc-hb I just extracted the command line from the run-all-tests.sh log. literal copy/paste.

fredoh9, 2021-06-22:
> @fredoh9 , @XiaoyunWu6666 is `check-capture.sh -d 1 -l 1 -r 3` not part of any test plan!?
This is part of PR test and Daily test. And part of run-test-all.sh
 
> I wonder what is the difference between:
> 
> ```
>     -r parameter |  --round parameter
> 	round count
> 	Default Value: 1
>    ...
>    -l parameter |  --loop parameter
> 	loop count
> 	Default Value: 3
> ```

You mean the difference betwewn -r (round), -l (loop)?
'Round' is going through all the pipeline and repeat, i.e PIPE1 -> PIPE2 -> PIPE3 and repeat(we called this round :) )
'Loop' is repeating with single pipe, i.e looping PIPE1 -> looping PIPE2 -> looping PIPE3

marc-hb, 2021-06-22:
@fredoh9 just explained me why we didn't catch this: it's not that we don't run with these parameters, it's about that specific hardware and/or topology configuration

plbossart, 2021-06-22:
@fredoh9 @marc-hb @ranj063 I added a flag to force dynamic pipelines to be used, it'd be interesting to start running daily tests with this kernel parameter set. If we don't start forcing this validation we'll be relatively blind...

fredoh9, 2021-06-23:
> @fredoh9 @marc-hb @ranj063 I added a flag to force dynamic pipelines to be used, it'd be interesting to start running daily tests with this kernel parameter set. If we don't start forcing this validation we'll be relatively blind...

I'm ready to run a daily-test with the flag to force dynamic pipelines to be used. I will run it right after scheduled daily test. My change is in staging branch in sof-sh repo for now. I ran test run with PR test plan (testplanresult 4831).

```
[    3.377696] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: tplg: pipeline id 1 comp 5 scheduling comp id 4
[    3.377699] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: pipeline PIPELINE.1.SSP0.OUT: period 1000 pri 0 mips 5000 core 0 frames 0 dynamic 1
```

fredoh9, 2021-06-24:
- daily test 4834 is scheduled, daily test
- daily test 4838 is the one, set the flag to force to use dynamic pipeline.

Both daily test are using same SHA1(s), this will be good comparison.

marc-hb, 2021-09-02:
(found this while looking for something else)

@ranj063 do we have a test for this and is it passing now?

XiaoyunWu6666, 2022-01-24:
@plbossart  Hi pierre , we do not have CNL platforms in CI .
Do we still have this issue ? Is it fixed?

mengdonglin, 2022-01-25:
This issue cannot be reproduced after SOF v2.0 and recent daily test on CML_RVP_NOCODEC with multicore and dynamic pipeline (e.g. test ID 9689)

closed 2022-01-25 by mengdonglin