thesofproject/sof #4380: [BUG] Disabling a secondary core and re-enabling it results in IDC timeout

ranj063, 2021-06-21:
**Describe the bug**
Disabling a secondary core and then re-enabling it without a D3 cycle always results in an IDC timeout

**To Reproduce**
1. Schedule a pipeline on core 1 in topology with dynamic pipelines
2. Start aplay/arecord on that pipeline
3. Stop aplay/arecord
4. start aplay/arecord for the second time before the DSP enters D3

**Reproduction Rate**
100%

**Expected behavior**
No IDC timeout

**Impact**
Cannot enable dynamic pipelines with multi-core

[logger.txt](https://github.com/thesofproject/sof/files/6684300/logger.txt)
After the first arecord after pm->restore, the IPC to set up the pipeline widget succeeds
```
[229302124.895833] (      500.312500) c0 ipc                  src/ipc/ipc3/handler.c:634  ipc: pm -> restore
[229305467.187500] (     3342.291748) c0 ipc                  src/ipc/ipc3/handler.c:661  ipc: pm core mask 0x3 -> enable
[229305512.135417] (       44.947918) c0 power              ../cavs/lib/pm_runtime.c:416  ERROR powering up secondary core 1

[229305617.447917] (      105.312500) c1 irq-cavs           ./intel/cavs/interrupt.c:301  ERROR platform_interrupt_init(): 1
[229305630.208333] (       12.760417) c1 edf-schedule       /schedule/edf_schedule.c:260  edf_scheduler_init()
[229305714.531250] (       84.322914) c1 idc                           src/idc/idc.c:317  idc_init() 1
[229305774.375000] (       59.843750) c1 idc                           src/idc/idc.c:328  idc_init() calling platform_idc_init 
[229306017.291667] (      242.916672) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[229306032.656250] (       15.364583) c1 idc                ......./intel/cavs/idc.c:165  idc_do_cmd()
[229306046.250000] (       13.593750) c1 ipc                  src/ipc/ipc3/handler.c:1229 ranjani pipe new

[229306059.427083] (       13.177083) c1 ipc                  src/ipc/ipc3/handler.c:1238 ipc: pipe 2 -> new
[229306072.552083] (       13.125000) c1 pipe               ......./pipeline-graph.c:113  pipeline new pipe_id 2 priority 0
[229306301.197917] (      228.645828) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[229306316.406250] (       15.208333) c1 idc                ......./intel/cavs/idc.c:165  idc_do_cmd()
```

But the second time around, it fails:
```
[230597629.895833] (      183.177078) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[230597671.250000] (       41.354168) c1 idc                ......./intel/cavs/idc.c:165  idc_do_cmd()
[230597701.718750] (       30.468750) c1 idc                ......./intel/cavs/idc.c:235  idc_free()
[230597890.156250] (      188.437500) c1 power              ../cavs/lib/pm_runtime.c:458  ERROR powering down secondary core 1

[232075058.906250] (  1477168.750000) c0 ipc                  src/ipc/ipc3/handler.c:661  ipc: pm core mask 0x3 -> enable
[232075103.750000] (       44.843750) c0 power              ../cavs/lib/pm_runtime.c:416  ERROR powering up secondary core 1

[232075208.437500] (      104.687500) c1 irq-cavs           ./intel/cavs/interrupt.c:301  ERROR platform_interrupt_init(): 1
[232075221.614583] (       13.177083) c1 edf-schedule       /schedule/edf_schedule.c:260  edf_scheduler_init()
[232075305.937500] (       84.322914) c1 idc                           src/idc/idc.c:317  idc_init() 1
[232075366.145833] (       60.208332) c1 idc                           src/idc/idc.c:328  idc_init() calling platform_idc_init 
[232075397.500000] (       31.354166) c1 idc                ......./intel/cavs/idc.c:58   idc_irq_handler(), IPC_IDCTFC_BUSY
[232075414.062500] (       16.562500) c1 idc                ......./intel/cavs/idc.c:165  idc_do_cmd()
[232075427.864583] (       13.802083) c1 ipc                  src/ipc/ipc3/handler.c:1229 ranjani pipe new

[232075441.093750] (       13.229167) c1 ipc                  src/ipc/ipc3/handler.c:1238 ipc: pipe 2 -> new
[232075454.895833] (       13.802083) c1 pipe               ......./pipeline-graph.c:113  pipeline new pipe_id 2 priority 0
[232076613.541667] (     1158.645874) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24218 cpu 378082
[232077863.177083] (     1249.635376) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24218 cpu 400020
[232080363.020833] (     2499.843750) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24218 cpu 400024
[232084113.020833] (     3750.000000) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24218 cpu 400036
[232087863.072917] (     3750.052002) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24218 cpu 400080
[232088102.812500] (      239.739578) c0 idc                           src/idc/idc.c:112  ERROR idc_wait_in_blocking_mode() error: timeout
```

ranj063, 2021-06-21:
@lgirdwood @keyonjie here're the logs for the issue that I'm seeing with enabling/disabling secondary cores without a D3 cycle.

plbossart, 2021-06-21:
@ranj063 is this limited to APL..CNL or does this also carry over to TGL? The flows for core power management are different so wondering if this can happen for TGL as well.

ranj063, 2021-06-21:
BTW the logs "ERROR powering down secondary core 1" and "ERROR powering up secondary core 1" are simply logs. there're no errors while powering up/down

lgirdwood, 2021-06-21:
Smells like cache. I bet we are not writing back global context on core N power down and invalidating it on core N power on. This probably doesn't show during regular boot sine all topology is loaded before any streams.

ranj063, 2021-06-21:
> @ranj063 is this limited to APL..CNL or does this also carry over to TGL? The flows for core power management are different so wondering if this can happen for TGL as well.

I dont know if this can happen on TGL as well. I'd say most likely yes but need to confirm.

plbossart, 2021-06-21:
@ranj063 I would think to some extent it should be 'easier' to handle on TGL, all the power is controlled by firmware.

for TGL-,  I am not sure if the firmware knows that a coreN (N>0) was powered off. Is there a mandatory sequence to send a 'prepare for power-down' IPC before turning the coreN off?

plbossart, 2022-12-12:
@ranj063 can we close this one? I think the conclusion is that multi-core is not supported prior to TGL, right?

ranj063, 2022-12-13:
Yes, let me mark as won't fix and close it.

closed 2022-12-13 by ranj063