thesofproject/sof #4382: [BUG] [TGL] Sporadic playback failure on speaker, headset and HDMI

yongzhi1, 2021-06-21:
**Describe the bug**
All the boards ran for 1 hour without any issue for YouTube streaming, but if we pause and play few times, then there is no sound coming.

**To Reproduce**
Play youtube for few seconds, pause for few seconds to 5 mins, then resume playback and observe no sound.

**Reproduction Rate**
100% (The issue can happen on speaker, headset and HDMI with equal rate.)

**Expected behavior**
Should not lose sound with pause/resume.

**Environment**
Kernel Branch: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.4
SOF Branch: tgl-013-drop-stable
SOF Commit: 9a35f6511
Device: Chrome VOXEL
TPLG:sof-tgl-max98373-rt5682-xperi.tplg

**Note**
The SOF FW include Google KWD and Maxiam DSM IP.

[dmesg.txt](https://github.com/thesofproject/sof/files/6689983/dmesg.txt)

yongzhi1, 2021-06-21:
cc @RDharageswari @ranj063

keyonjie, 2021-06-22:
Hi @yongzhi1 can you attach sof-logger logs also?

lgirdwood, 2021-06-22:
@yongzhi1 can you also confirm if the DSM is in the output path here. I know we have the same pause test in CI but it may not be holding for 5 secs and it wont be using DSM.

yongzhi1, 2021-06-22:
> Hi @yongzhi1 can you attach sof-logger logs also?

Sure, the dmesg.txt was provided by validation, here are the logs I captured:

[dmesg-003.txt](https://github.com/thesofproject/sof/files/6694334/dmesg-003.txt)
[dsp-log-003.txt](https://github.com/thesofproject/sof/files/6694335/dsp-log-003.txt)

yongzhi1, 2021-06-22:
> @yongzhi1 can you also confirm if the DSM is in the output path here. I know we have the same pause test in CI but it may not be holding for 5 secs and it wont be using DSM.

Yes, but the same issue can be reproduced on headset and HDMI as well:
[dmesg-hdmi.txt](https://github.com/thesofproject/sof/files/6694355/dmesg-hdmi.txt)

lyakh, 2021-06-23:
From the DSP log:
```
[   783510433.293176] (       10420.000000) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 112955
[   783515376.522146] (        4943.229004) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 95011
[   783520980.428174] (        5603.906250) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout
```
that's the first occurrence of the `ERROR idc_wait_in_blocking_mode() error: timeout` error and 112955 and 95011 deltas are huge. On TGL the timer ticks with 38.4MHz, i.e. those deltas are like 3 and 2.5ms roughly. Looks like some code path is blocking interrupts for too long.

yongzhi1, 2021-06-23:
Thanks @lyakh for the insight , so it seems the kernel side IPC error is because idc_do_cmd() get blocked to process the pipeline param cmd.

With the following proof-of-concept, the issue seems to be fixed, but I am sure you can provide the right fix :)
https://github.com/yongzhi1/sof/commit/f61f493b64f9099110e3a9f5b046f293eecdd6ab

[idc-temp-fix-log.txt](https://github.com/thesofproject/sof/files/6702307/idc-temp-fix-log.txt)

lyakh, 2021-06-24:
> Thanks @lyakh for the insight , so it seems the kernel side IPC error is because idc_do_cmd() get blocked to process the pipeline param cmd.
> 
> With the following proof-of-concept, the issue seems to be fixed, but I am sure you can provide the right fix :)
> [yongzhi1@f61f493](https://github.com/yongzhi1/sof/commit/f61f493b64f9099110e3a9f5b046f293eecdd6ab)

Unfortunately I don't have a *simple* correct fix. We have some clues which code paths are blocking interrupts for a long time in some scenarios and it would be easy enough to find out which one is doing that here, but fixing them is often rather difficult.

lgirdwood, 2021-06-25:
@yongzhi1 there is some incoming multicore fixes, probably coming Monday - we should retest with these.
@ranj063 fyi.

ranj063, 2021-06-28:
I have a theory here. I think what happens is when the playback stream is paused(stooped in chrome), the DSP enters D0i3 because the wov stream is running. And then when the stream is unpaused, the DSP transitions from D0i3->D0i0 and fails to send IDC to the secondary core. Could it be that we dont restore the secondary cores after a D0i3 cycle?

libinyang, 2021-06-28:
I still can't reproduce this issue on Delbin. I have done the test for more than 1 hour with dozens of pause/resume on youtube.

- I have updated the fw and tplg. 
    7aac4106dd4d2a90b69fdb4d40d7ad18  sof-tgl.ri
    e451294e818f8860647ee2aa2d0e57b5  sof-tgl-max98373-rt5682.tplg
- I have enabled the GHWD by adding “--enable-features=EnableDspHotword” in /etc/chrome_dev.conf and reboot.

ranj063, 2021-06-28:
@yongzhi1 confirmed that problem can only be reproduced with WOV enabled and that confirms my theory. Here's the long explanation for the issue:

The chrome tplg has 1 pipeline on core 1 and the rest on core 0 with woV is enabled. When a pipeline running on core 0 is started and then paused (chrome stops streams when paused), wov is the only pipeline left running and the DSP enters D0i3.

For the DSP to be in D0i3, only core 0 should be active. So technically we should power off coreX before entering and power it back up upon exiting D0i3 from the host. This is missing today. So does the FW enter D0i3 thinking coreX is active or does it power it off on its own?

Now assuming we power off CoreX before D0i3 and power it back on after D0i3 exit, the pipelines on CoreX need to be set up again. The only time we set up pipelines is after a D3 exit. I am not sure how we can set up the Corex pipelines without a D3 cycle. 

The most logical fix is to use dynamic pipelines. So we take down unused pipelines along with coreX when entering D0i3. But it is a massive feature and would require careful validation.

@lgirdwood @mwasko FYI

ranj063, 2021-06-28:
@lgirdwood I have one outstanding issue with dynamic pipelines and multicore today (and thats only based on my testing). We need to more testing to rule out more issues.

yongzhi1, 2021-06-28:
To summarize my understanding, as a fix, the dynamic pipelines fit the use case/runtime paradigm perfectly, it may represent the least effort to "extend" D0ix support to CoreX right now.

libinyang, 2021-06-29:
I can reproduce this issue now with the help from @sathya-nujella to enable "Ok Google" feature. 

@ranj063 's explanation makes sense and should be the root cause. @yongzhi1 Could you try https://github.com/thesofproject/sof/pull/4390 on your side? I will set my environment to build the new fw/tplg on my side at the same time.

mwasko, 2021-06-29:
> Now assuming we power off CoreX before D0i3 and power it back on after D0i3 exit, the pipelines on CoreX need to be set up again. The only time we set up pipelines is after a D3 exit. I am not sure how we can set up the Corex pipelines without a D3 cycle.

The FW is power gating cores when they are not used, so in this scenario CoreX has already been power off prior to entry to D0i3. If the core power off would cause that issue then I would expect to see the same problem in scenario with pause of pipeline on Core 1 and then resume (with second pipeline running on Core 0). The unique thing that we perform in D0i3 with WoV is switch to WOV CRO clock and resume HPRO clock on D0i3 exit.

ranj063, 2021-06-29:
> The FW is power gating cores when they are not used,

@mwasko when is a core "in use"? When a pipeline is active or does it simply mean a pipeline and its components are initialized and scheduled to be on that core?

yongzhi1, 2021-06-29:
> Now assuming we power off CoreX before D0i3 and power it back on after D0i3 exit, the pipelines on CoreX need to be set up again. The only time we set up pipelines is after a D3 exit. I am not sure how we can set up the Corex pipelines without a D3 cycle.
> 

Did another experiment, the results seem to agree with above hypothesis:

Step 1: repro the no sound issue with youtube.
Step 2: close the DUT lid to trigger "sleeps".
Step 3: wait a little while, open the lid, the youtube playback works fine.

mwasko, 2021-06-30:
> > Now assuming we power off CoreX before D0i3 and power it back on after D0i3 exit, the pipelines on CoreX need to be set up again. The only time we set up pipelines is after a D3 exit. I am not sure how we can set up the Corex pipelines without a D3 cycle.
> 
> Did another experiment, the results seem to agree with above hypothesis:
> 
> Step 1: repro the no sound issue with youtube.
> Step 2: close the DUT lid to trigger "sleeps".
> Step 3: wait a little while, open the lid, the youtube playback works fine.
> (The blocking condition happened in Step 1 was unblocked with Step 2 & 3)

The hypothesis assume that in the process of D0i3 transition the pipeline on CoreX (other then 0) is being destroyed and need to be setup again on D0i3 exit. The experiment with close LID (sleep) confirm that after D3 (FW gets reloaded and initialized again) we can recover. This is expected but does not confirm the hypothesis - most of other FW-Driver error states would be recovered as well in the process of D3 transition.

@yongzhi1, @libinyang can we try the original reproduction scenario with D0i3 but with dynamic pipelines enabled? If the issue will not reproduce than it will be a good hint that we have a problem with pipelines on CoreX during D0i3 transition.

yongzhi1, 2021-06-30:
Hi, @mwasko, I agree with you that closing the lid means DSP FW reload, so it's equal to a reset.

For the "ERROR idc_wait_in_blocking_mode() error: timeout", 

1. Can be reproduced by on-line recorder app
2. Can be reproduced by cold reboot alone 

[dsp-cold-reboot-dmic.log](https://github.com/thesofproject/sof/files/6742875/dsp-cold-reboot-dmic.log)

Dynamic pipelines need some porting effort as we are on v5.4, will update you.

slawblauciak, 2021-07-02:
@yongzhi1 please try this PR and tell me if you can still reproduce the issue
https://github.com/thesofproject/sof/pull/4444

lgirdwood, 2021-07-02:
@ujfalusi it looks like this kernel error is wrong, should it be "no volatile kcontrol found" ?

> [    7.401991] sof-audio-pci 0000:00:1f.3: error: volatile control found for dynamic widget SMART_AMP1.0
> [    7.402037] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
> [    7.402041] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22

ujfalusi, 2021-07-02:
@lgirdwood, it depends, but I think in this case it is printed because:
`dynamic widgets cannot have volatile kcontrols` it is from `sof_set_pipe_widget()`

My guess is that the topology is not correct?

lgirdwood, 2021-07-02:
> @lgirdwood, it depends, but I think in this case it is printed because:
> `dynamic widgets cannot have volatile kcontrols` it is from `sof_set_pipe_widget()`

ok, can we update the kernel message ?

> 
> My guess is that the topology is not correct?

Could be, @yongzhi1 can you check the topology here ? Could be a bug here too.

ujfalusi, 2021-07-02:
> > @lgirdwood, it depends, but I think in this case it is printed because:
> > `dynamic widgets cannot have volatile kcontrols` it is from `sof_set_pipe_widget()`
> 
> ok, can we update the kernel message ?

Now that I know why it is printed, it is clear and it states the reason fore the error.
I can try to come up with something, but so far I could not find a compact enough one to fit under 100 char column.

yongzhi1, 2021-07-03:
> @yongzhi1 please try this PR and tell me if you can still reproduce the issue
> #4444

Hi, @slawblauciak , thanks for the patch, with everything remains the same as the original tgl-013-drop-stable + #4444, I can still repro the issue: [dsp-pr4444-log.txt](https://github.com/thesofproject/sof/files/6759066/dsp-pr4444-log.txt)

keyonjie, 2021-07-05:
Hi @yongzhi1 can you try to disable LPS:
remove the 1st line 'CONFIG_CAVS_LPS=y' in src/arch/xtensa/configs/override/tigerlake_chrome.config and rebuild the FW.

yongzhi1, 2021-07-05:
> Hi @yongzhi1 @libinyang can you try to disable LPS:
> remove the 1st line 'CONFIG_CAVS_LPS=y' in src/arch/xtensa/configs/override/tigerlake_chrome.config and rebuild the FW.

With the .config update as suggested, the issue is no longer reproducible: [dsp-pr4444-log-02.txt](https://github.com/thesofproject/sof/files/6764663/dsp-pr4444-log-02.txt). During my test, only observed no sound once with the trace flooded with:

```
[    19919999.781367] (       16010.051758) c0 dw-dma                 src/drivers/dw/dma.c:1046 dw_dma_avail_data_size() size is 0!
[    19920019.677200] (          19.895832) c0 dai          11.67          src/audio/dai.c:1140 WARN dai_copy(): nothing to copy
[    19935991.447398] (       15971.770508) c0 dai          11.67          src/audio/dai.c:1136 WARN dai_copy(): Copy_bytes 768 + free bytes 32 < period bytes 1536, possible glitch
```

sathya-nujella, 2021-07-16:
Hello,
Thank you @ranj063 and team,

For TGL alone, we got approval from Google to take the below change as CHROMIUM only patch for Chrome-kernel 5.4 and got merged there (CHROMIUM only: can not go upstream)  as a fix for this issue  (for future platforms - dynamic pipelines, not this WA)

Diff which is taken for ref:

```
diff --git a/sound/soc/sof/intel/cnl.c b/sound/soc/sof/intel/cnl.c
index 5da40c82636a..c387eb3da1f9 100644
--- a/sound/soc/sof/intel/cnl.c
+++ b/sound/soc/sof/intel/cnl.c
@@ -195,18 +195,6 @@ int cnl_ipc_send_msg(struct snd_sof_dev *sdev, struct snd_sof_ipc_msg *msg)

        hdr = msg->msg_data;

-       /*
-        * Use mod_delayed_work() to schedule the delayed work
-        * to avoid scheduling multiple workqueue items when
-        * IPCs are sent at a high-rate. mod_delayed_work()
-        * modifies the timer if the work is pending.
-        * Also, a new delayed work should not be queued after the
-        * CTX_SAVE IPC, which is sent before the DSP enters D3.
-        */
-       if (hdr->cmd != (SOF_IPC_GLB_PM_MSG | SOF_IPC_PM_CTX_SAVE))
-               mod_delayed_work(system_wq, &hdev->d0i3_work,
-                                msecs_to_jiffies(SOF_HDA_D0I3_WORK_DELAY_MS));
-
        return 0;
 }

```

https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/3025608

@sathyap-chrome :  FYI

yongzhi1, 2021-07-19:
Close as [CR #3025608](https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/3025608) got merged.

sathya-nujella, 2021-08-08:
> Hello,
> Thank you @ranj063 and team,
>
> For TGL alone, we got approval from Google to take the below change as CHROMIUM only patch for Chrome-kernel 5.4 and got merged there (CHROMIUM only: can not go [...]

Google told us, as kernel WA can not go upstream: WA is accepted for temporary and asked to provide upstream SOF/driver solution for this BUG.
Adding ADL tag and re-opening the bug. Thank you!

sathya-nujella, 2021-08-08:
@Vamshigopal  - FYI

keyonjie, 2021-08-10:
> 
> Google told us, as kernel WA can not go upstream: WA is accepted for temporary and asked to provide upstream SOF firmware / driver solution for this BUG.
> Adding ADL tag and re-opening the bug. Thank you!

Hi @sathya-nujella do you mean that we need the upstream version for TGL also, or only for ADL?

kv2019i, 2021-08-17:
Agreed with Karol to move this P1 ADL issue to @bkokoszx .

abonislawski, 2021-09-27:
> Could it be that we dont restore the secondary cores after a D0i3 cycle?

Yes, we dont even power up secondary cores on wake up :)
With @bkokoszx we managed to fully restore secondary cores, we will share debug code soon for testing purposes.

bkokoszx, 2021-09-27:
Here is debug branch mentioned by @abonislawski :
https://github.com/bkokoszx/sof/tree/tgl-013-drop-stable-secondary_core_restore

@yongzhi1 Are you able to try to reproduce issue with above code?

Vamshigopal, 2021-09-28:
@bkokoszx In ADL we saw little different issue,  logs 
[wov_spk.txt](https://github.com/thesofproject/sof/files/7244148/wov_spk.txt)

Test : if we try speaker playback after GHW(s0ix)
Observation : speaker playback is muted.
fw_branch: origin/adl-003-drop-stable

We have not reported this issue because we are seeing this behavior only with google kpd, with intel clap issue is not observed.
with your debug branch https://github.com/bkokoszx/sof/tree/tgl-013-drop-stable-secondary_core_restore
the issue is no longer reproduced with google kpd.

keyonjie, 2021-09-30:
> @bkokoszx In ADL we saw little different issue, logs
> [wov_spk.txt](https://github.com/thesofproject/sof/files/7244148/wov_spk.txt)
> from logs Error signature is matching with original TGL issue.
> 
> Test : if we try speaker playback after GHW(s0ix)
> Observation : speaker playback is muted.
> fw_branch: origin/adl-003-drop-stable
> 
> We have not reported this issue because we are seeing this behavior only with google kpd, with intel clap issue is not observed.
> with your debug branch https://github.com/bkokoszx/sof/tree/tgl-013-drop-stable-secondary_core_restore
> the issue is no longer reproduced with google kpd.

@Vamshigopal I think adl-003-drop-stable is already aligned with the latest tgl-013-drop-stable so the issue you mentioned should already been fixed, can you check the top commit of your adl-003-drop-stable branch?

@bkokoszx please correct me if I am wrong.

sathyap-chrome, 2021-09-30:
@keyonjie we saw the issue even when tgl013 FW resigned for ADL.
with or without the WA submitted in TGL.

keyonjie, 2021-09-30:
> @keyonjie we saw the issue even when tgl013 FW resigned for ADL. with or without the WA submitted in TGL.

got it, then looks @bkokoszx and @abonislawski 's fix for the LPS should be the correct way to go, we should get them for all TGL+ WoV related platforms.

bkokoszx, 2021-09-30:
@keyonjie 
adl-003-drop-stable is based on tgl-013-drop-stable. Additionally, it contains only two more topology commits, and minor change in FW referring to check buffer allocation in comp_prepare() function. So basically, there are mostly the same.

sathyap-chrome, 2021-10-04:
Tested the series with below topology config on ADL Brya platform:

SPk pipe -> dump codec  (no algo running) - CORE 0
DMIC -> Intelligo - CORE1
HS pipe -> no algo running - CORE2
HDMI pipelines -> no algo running - CORE3
Results here:
[Audio_multicore_testing.xlsx](https://github.com/thesofproject/sof/files/7276098/Audio_multicore_testing.xlsx)

We also tested known issues raised in TGL when multicore was enabled, attached results here:
[sof_issues_on_ADL.xlsx](https://github.com/thesofproject/sof/files/7276102/sof_issues_on_ADL.xlsx)

Expect the known alsa conformance faulire, rest all use cases are PASS
kindly help merge this and also make a adl-003-hotfix branch to include them

cc @Vamshigopal @kv2019i @bkokoszx

kv2019i, 2021-11-25:
@yongzhi1 @bkokoszx , ok to close now?

lgirdwood, 2021-12-21:
Can reopen if needed.

closed 2021-12-21 by lgirdwood