thesofproject/sof #4383: [BUG][ADL]dai_config error when aplaying HDMI1 on ADL

XiaoyunWu6666, 2021-06-22:
**Describe the bug**
When running aplay on HDMI1 , dai_config error occured

**To Reproduce**
on ADLP_RVP_SDW, run aplay   -Dhw:0,5 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
on ADLP_RVP_HDA , run  aplay   -Dhw:0,3 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: b1c762a1
SOF Branch: main
SOF Commit: 86c9a037e212
Topology: sof-hda-generic-4ch-kwd.tplg for ADLP_RVP_HDA ; sof-adl-rt711-4ch.tpl for ADLP_RVP_SDW
 Platform: **ADLP_RVP_HDA ;ADLP_RVP_SDW**

**Screenshots or console output**
[dmesg]
```shell
ubuntu@sh-adlp-rvp-hda-01:~$ aplay -Dhw:0,5 48K_Let_It_Go.wav -vvv
Playing WAVE '48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1407: Unable to install hw params:
```
[sof-logger]
```
[   153826040.189586] (           0.000000) c0 dma-trace             src/trace/dma-trace.c:343  ERROR FW ABI 0x3013000 DBG ABI 0x5003000 tag v1.8-rc1-236-g86c9a037e212 src hash 0x6560584a (ldc hash 0x6560584a)
[         238.124991] (         238.124985) c0 ll-schedule        ./schedule/ll_schedule.c:393  INFO task add 0x9e126740 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690>
[         255.729157] (          17.604166) c0 ll-schedule        ./schedule/ll_schedule.c:397  INFO task params pri 4 flags 0 start 500000 period 500000
[         281.458322] (          25.729166) c0 ll-schedule        ./schedule/ll_schedule.c:309  INFO new added task->start 3377618 at 3351437
[         298.541655] (          17.083332) c0 ll-schedule        ./schedule/ll_schedule.c:312  INFO num_tasks 2 total_num_tasks 2
[         951.666629] (         653.125000) c0 ipc                  src/ipc/ipc3/handler.c:634  INFO ipc: pm -> restore
[        4861.874807] (        3910.208252) c0 component             src/ipc/ipc3/helper.c:418  INFO comp new dai <c2b00d27-ffbc-4150-a51a-245c79c5e54b> type 2 id 9.57
[        4909.114388] (          47.239582) c0 hda-dai      3.6   /drivers/intel/hda/hda.c:69   INFO hda_probe()
[        4932.499804] (          23.385416) c0 dai                           src/lib/dai.c:164  INFO dai_get type 3 index 6 new sref 1
[        4955.416470] (          22.916666) c0 hda-dma            ..../intel/hda/hda-dma.c:790  INFO hda-dmac :7 -> probe
[        5041.093550] (          85.677078) c0 dma                           src/lib/dma.c:121  INFO dma_get() ID 7 sref = 1 busy channels 0
[        5690.156024] (         649.062500) c0 ipc                  src/ipc/ipc3/handler.c:536  INFO ipc: dai 3.6 -> config
[        5709.687273] (          19.531250) c0 dai                           src/lib/dai.c:164  INFO dai_get type 3 index 6 new sref 2
[        5727.603939] (          17.916666) c0 hda-dai      3.6   /drivers/intel/hda/hda.c:36   INFO hda_set_config(): channels 2 rate 48000
[        5745.781022] (          18.177082) c0 dai                           src/lib/dai.c:188  INFO dai_put type 3 index 6 new sref 1
[        5763.958104] (          18.177082) c0 ipc                      src/ipc/ipc3/dai.c:159  INFO ipc_comp_dai_config() dai type = 3 index = 6
[        5783.906020] (          19.947916) c0 dai          9.57        src/ipc/ipc3/dai.c:223  INFO dai_config() dai type = 3 index = 6 dd 0x9e122000
[        6482.812242] (         698.906250) c0 ipc                  src/ipc/ipc3/handler.c:536  INFO ipc: dai 3.6 -> config
[        6502.187242] (          19.375000) c0 dai                           src/lib/dai.c:164  INFO dai_get type 3 index 6 new sref 2
[        6519.739324] (          17.552082) c0 hda-dai      3.6   /drivers/intel/hda/hda.c:36   INFO hda_set_config(): channels 2 rate 48000
[        6537.499740] (          17.760416) c0 dai                           src/lib/dai.c:188  INFO dai_put type 3 index 6 new sref 1
[        6556.614323] (          19.114582) c0 ipc                      src/ipc/ipc3/dai.c:159  INFO ipc_comp_dai_config() dai type = 3 index = 6
warn: failed to fread() 3 params from the log for src/ipc/ipc3/dai.c:223

```

kv2019i, 2021-06-23:
This has same rootcause as https://github.com/thesofproject/linux/issues/2992

plbossart, 2021-06-23:
@kv2019i for complete idiots like me, what is the root cause? I am not sure I get how we can have a DAI_CONFIG issue, there's nothing hardware-specific about those configs or dependent on i915, is there?

kv2019i, 2021-06-24:
@plbossart This is not at all so obvious indeed. But it's not a DAI config issue but kernel failure for hw_params. You can get the same result with any HDA platform by adding "options snd_sof_intel_hda_common codec_mask=0x1" and trying to play to one of the HDMI/DP PCMs.

The snd_hdac_ext_bus_get_link() call with fail in sof/intel/hda-dai.c:hda_link_hw_params() and -EINVAL is returned to user-space.

XiaoyunWu6666, 2021-06-29:
It seems that the error still happens in inner daily -ADL tests but logs changed a bit:

 id :4930    model=ADLS_RVP_HDA&testcase=check-playback-3rounds
[demsg]
```
[  276.613327] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  276.613777] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  276.613811] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  276.614260] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  276.614275] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget HDA4.OUT setup complete
[  276.614300] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  276.614683] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  276.614700] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp1 Pin: -22
[  276.614719] kernel:  iDisp1: ASoC: soc_pcm_hw_params() failed (-22)
[  276.614727] kernel:  HDMI1: ASoC: dpcm_fe_dai_hw_params failed (-22)
[  276.614738] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 3 dir 0
[  276.615219] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 3 dir 0
[  277.476694] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  277.476754] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  277.882392] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x90040000: GLB_TRACE_MSG
[  277.882691] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x90040000: GLB_TRACE_MSG
[  277.993947] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x90040000: GLB_TRACE_MSG
[  277.994330] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x90040000: GLB_TRACE_MSG
[  278.476699] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  278.476780] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  278.976686] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  278.976768] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG

```

lgirdwood, 2021-06-30:
@kv2019i sorry (another complete idiot here), it sounds like you are saying we are sending the wrong config to FW because the GFX driver is not ready or giving us a bad config ?

kv2019i, 2021-06-30:
@lgirdwood No, no, this is not a DAI config issue. It's a kernel hw_params error. When i915 is not available, the codec driver is ASoC dummy and that cannot provide a HDA link, so hw_params fails.

Let me move this to kernel.