thesofproject/sof #4392: [BUG] glitches with SSP interface during pause, stop and start operations

kkarask, 2021-06-23:
**Describe the bug**
Glitches was found in  test: 02_20_TestPtpCaptureStreamStopStartOperation48000Hz32b32b2ch
While pipelines are working pause, stop and start operation are done on capture stream.

**Topology:**
```
pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```

**To Reproduce**
Run test 02_20_TestPtpCaptureStreamStopStartOperation48000Hz32b32b2ch with Diagnostic driver
with parameters:
--playback_iterations=75

**Reproduction Rate**
2/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: d5d2b41c63ab0715c5f122ac1cfcdbaed8df40b1

3) Name of the platform(s) on which the bug is observed.
    * Platform: WHL, APL

[image]

[WHL_02_20_TestPtpCaptureStreamStopStartOperation48000Hz32b32b2ch.zip](https://github.com/thesofproject/sof/files/6700992/WHL_02_20_TestPtpCaptureStreamStopStartOperation48000Hz32b32b2ch.zip)

slawblauciak, 2021-06-29:
@kkarask does the glitch only occur at the point of starting/stopping the stream? Or are they randomly scattered around in the stream?

kkarask, 2021-06-29:
@slawblauciak glitches are visible between the start/stop operation like here:

[image]

closed 2025-07-24 by github-actions[bot]