thesofproject/sof #4400: [BUG]APL_UP2_NOCODEC_ZEPHYR ipc timeout when CTX_SAVE

XiaoyunWu6666, 2021-06-25:
**Describe the bug**

**To Reproduce**
TPLG=sof-apl-nocodec.tplg ~/sof-test/test-case/verify-kernel-boot-log.sh

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 695d444a
SOF Branch: main
SOF Commit: 
86c9a037e212
Topology:  sof-apl-nocodec.tplg
Platform:APL_UP2_NOCODEC_ZEPHYR

**Screenshots or console output**
[dmesg]
```
[    9.358937] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[    9.863111] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x40010000 size 76
[    9.863343] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[    9.863356] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[    9.863459] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00000000
[    9.863530] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[    9.863601] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[    9.863668] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[    9.863745] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ctx_save ipc error -110, proceeding with suspend
[    9.863772] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x4]=0x1010303 successful
[    9.863787] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0x4]=0x303 successful
```

lgirdwood, 2021-06-25:
@XiaoyunWu6666 thanks, btw - can you add the Zephyr label for anything zephyr related. Thanks !

kv2019i, 2021-06-29:
@lyakh  routing to you as you have an APL.

XiaoyunWu6666, 2021-07-02:
some logs on TGLH_RVP_NOCODEC_ZEPHYR look quite similar to logs on APL_UP2_NOCODEC 

Shall I open some new issues for TGLH_RVP_NOCODEC?

lgirdwood, 2021-07-02:
I dont think we are not calling the Zephyr power OFF, but going through the SOF power OFF only (meaning Zephyr will still have active IP. IRQs etc).
@dcpleung fyi - is there a z_power_off() ready with the CAVS power off logic ready ? This CAVS logic can be copy and pasted from SOF (some of it is assembly).

dcpleung, 2021-07-06:
I don't think we have anything PM related for CAVS in Zephyr.

lyakh, 2021-07-14:
Just ran both tests on an UP2 with a PCM512x codec with an SMP tweak with an older 5.11-ish kernel - no reproduction. Then repeated the tests with a current 5.14-ish - and both tests failed. check-runtime-pm-status.sh failed with
```
2021-07-14 12:28:53 UTC [ERROR] Caught kernel log error
===========================>>
[  464.672008] kernel:  SSP5-Codec: ASoC: no users capture at close - state 0
```
and check-suspend-resume.sh failed with
```
[  604.110390] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x40010000 size 76
[  604.110408] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[  604.110415] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  604.110419] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  604.110423] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  604.110426] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[  604.110429] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[  604.110433] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[  604.110482] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  604.110486] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
```

lgirdwood, 2021-07-14:
@lyakh @dcpleung part of the power down is that we need to disable all IP that can cause bus transactions prior to D3. i.e. we need to ensure the timer is OFF (the enable bit must be clear). Today, SOF takes care of most IP, but some IP (like timer) is under Zephyr control today.

dcpleung, 2021-07-14:
Zephyr doesn't even have a way to shutdown the device (though there is `sys_reboot()`). The model assumes that the hardware would run forever.

I wonder if it is possible to utilize PM deep sleep to do this, by enabling PM in Zephyr and implementing `pm_power_state_set()`. Deep sleep is only allowed when there is need to power off the DSP. Then you can do shutdown instead.

lgirdwood, 2021-07-16:
@dcpleung ```pm_power_state_set()``` wont work on it's own as we need to make sure we are disabling all IP, masking IRQs etc. e.g. the timer must be switched off so that it will never assert during the shutdown flow, and scheduling needs to be shutdown prior to timer etc..

dcpleung, 2021-07-16:
Just for clarification, when you say shutdown, do you mean the SoC is totally off, so next time it starts from scratch (like power-cycle)? If so, it might be possible to fake deep sleep into a shutdown.

lgirdwood, 2021-07-16:
I mean the DSP is totally OFF and it does start from scratch next time. The DSP FW must power off and shutdown all it's internal facing IP before the host driver puts the DSP PCI device into D3. Not doing this can lead failure to enter D3 or D0.

kv2019i, 2021-08-10:
Potential bugfix https://github.com/thesofproject/sof/pull/4618

kv2019i, 2021-08-17:
@XiaoyunWu6666 This should be fixed now, can you confirm? The latest daily tests no longer show this.

1994lwz, 2021-08-18:
This bug was verified on the today's daily version, and it did not reproduce, so close it.

closed 2021-08-18 by 1994lwz