thesofproject/sof #4411: [BUG][TGLU][SDW]Dai failed to alloc dma buffer when running multiple-pipeline on TGLU-RVP_SDW

XiaoyunWu6666, 2021-06-28:
**Describe the bug**
dai failed to alloc dam buffer whn running multiple-pipeline on TGLI_RVP_SDW[sh-tglu-rvp-sdw-01]

**To Reproduce**
 TPLG=sof-tgl-rt711-rt1308-4ch.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 5b851f48
SOF Branch: main
SOF Commit: bccecb1a8eed

**Screenshots or console output**
[sof-logger]
```
[     1959349.765892] (        1909.739502) c0 pipe         2.11  ....../pipeline-params.c:215  INFO pipe params dir 1 frame_fmt 0 buffer_fmt 0 rate 48000
[     1959369.140892] (          19.375000) c0 pipe         2.11  ....../pipeline-params.c:219  INFO pipe params stream_tag 4 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[     1959478.776304] (         109.635414) c0 dai          2.10        src/ipc/ipc3/dai.c:84   INFO dai_data_config() dai type = 4 index = 3 dd 0x9e122200
[     1959506.745053] (          27.968748) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[     1959523.984636] (          17.239582) c0 dai          2.10           src/audio/dai.c:543  ERROR dai_params(): failed to alloc dma buffer
[     1959542.578385] (          18.593750) c0 pipe         2.11  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 6
[     1959561.797134] (          19.218750) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 2 comp 6 params failed -12
[     1959580.026300] (          18.229166) c0 pipe         2.11  ......./pipeline-graph.c:338  INFO pipe reset
[     1959663.359630] (          83.333328) c0 pga          2.7      src/audio/component.c:63   INFO comp_set_state(), state already set to 1
[     1959683.411713] (          20.052082) c0 dai          2.10           src/audio/dai.c:687  INFO dai_reset()
[     1959700.078379] (          16.666666) c0 dai          2.10     src/audio/component.c:63   INFO comp_set_state(), state already set to 1
```
[dmesg]
```
2021-06-27 22:37:10 UTC [REMOTE_INFO] Testing: DMIC [hw:0,10]
2021-06-27 22:37:10 UTC [REMOTE_COMMAND] arecord   -D hw:0,10 -c 4 -r 48000 -f S16_LE /dev/null -q
2021-06-27 22:37:10 UTC [REMOTE_INFO] Testing: DMIC16kHz [hw:0,11]
2021-06-27 22:37:10 UTC [REMOTE_COMMAND] arecord   -D hw:0,11 -c 4 -r 16000 -f S16_LE /dev/null -q
2021-06-27 22:37:10 UTC [REMOTE_INFO] Testing: Jack In [hw:0,1]
2021-06-27 22:37:10 UTC [REMOTE_COMMAND] arecord   -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q
2021-06-27 22:37:10 UTC [REMOTE_INFO] wait 0.5s for aplay_opts()
2021-06-27 22:37:11 UTC [REMOTE_INFO] checking pipeline status
2021-06-27 22:37:11 UTC [REMOTE_INFO] Letting playback/capture run for 5s
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
2021-06-27 22:37:16 UTC [REMOTE_INFO] checking pipeline status again
2021-06-27 22:37:16 UTC [REMOTE_ERROR] Running process count is 8, but 9 is expected
```

plbossart, 2021-06-28:
@XiaoyunWu6666 please do not conflate two bugs on separate platforms that have nothing to do with each other. TGP-SDW and JSL-nocodec are as similar as diesel or electric cars. Please file two separate issues.

plbossart, 2021-06-28:
@lgirdwood @ranj063 the difference between tests points to the following commits

6/28 Daily test 4904: FAIL SOF commit  bccecb1a8eed

6/26 Daily test 4887: PASS SOF commit 2d70f100c7ad

git log --oneline 2d70f100c7ade0def40fa14d4de9458ae7ed32f5..HEAD
(not tested) 0cf1f8854 (HEAD -> main, origin/main) buffer: fix double list deletion detected by valgrind.
bccecb1a8 dai: only process DAI config for comps on cores that enabled
928dba085 pipeline/buffer: alloc them in the runtime shared zone
29f0aeb91 arch: extensa: cpu: reset stale trace point during power down
7b06921b6 pipeline_graph: no need to free pipeline components during pipeline_free()
299fe20d8 schedule: edf/ll: remove list del/init during schedule free
da79c5073 ipc: ipc3: handler: fix error handling for pm_core_enable

lgirdwood, 2021-06-28:
@plbossart there was a double free in DMIC that @ranj063 has just fixed (now merged).
@XiaoyunWu6666 can you check again with the latest main branch.

XiaoyunWu6666, 2021-06-29:
> @XiaoyunWu6666 please do not conflate two bugs on separate platforms that have nothing to do with each other. TGP-SDW and JSL-nocodec are as similar as diesel or electric cars. Please file two separate issues.

Yes , let's leave this for TGLU_RVP_SDW and open a new one for JSL_RVP_NOCODEC.

XiaoyunWu6666, 2021-06-29:
> @plbossart there was a double free in DMIC that @ranj063 has just fixed (now merged).
> @XiaoyunWu6666 can you check again with the latest main branch.

It still happens . inner test 4947?model=TGLU_VOLT_SDW&testcase=multiple-pipeline-all
[sof-logger]
```
[      903895.432832] (      903895.437500) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      903914.391165] (          18.958332) c0 host         1.0           src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      903931.995331] (          17.604166) c0 pipe         1.5   ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 0
[      903951.214080] (          19.218750) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 1 comp 0 params failed -12
[      905251.734862] (        1300.520752) c0 dai          7.35        src/ipc/ipc3/dai.c:253  ERROR dai_config(): No memory for dai_config.
[      905269.911944] (          18.177082) c0 ipc                      src/ipc/ipc3/dai.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed
[      905725.068176] (         455.156219) c0 dai          8.41        src/ipc/ipc3/dai.c:253  ERROR dai_config(): No memory for dai_config.
[      905742.828592] (          17.760416) c0 ipc                      src/ipc/ipc3/dai.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed
[      908698.557642] (        2955.729004) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      908717.151391] (          18.593750) c0 host         4.19          src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      908734.390973] (          17.239582) c0 pipe         4.24  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 19
[      908753.765973] (          19.375000) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 4 comp 19 params failed -12
[      911556.734611] (        2802.968750) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      911575.484611] (          18.750000) c0 host         2.6           src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      911592.828360] (          17.343750) c0 pipe         2.11  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 6
```

lgirdwood, 2021-06-29:
@XiaoyunWu6666 thanks, I think 4 OOM issues in 2 days means we need to check the heap map trace info on TGL.

XiaoyunWu6666, 2021-06-30:
TGLH_0A5E_SDW with this issue too #4433

XiaoyunWu6666, 2021-07-14:
dma buffer issue seems to disappear from daily test result .
Let's close it and reopen once it happened again

closed 2021-07-14 by XiaoyunWu6666