thesofproject/sof #4418: [BUG] [TGL] XRUN on speaker/headset path with the IGO integrated firmware while starting the PCM99 capture

RDharageswari, 2021-06-28:
**Describe the bug**
With the IGO integrated firmware and topology, XRUN is observed in the playback pipeline, be it headset/the speaker path with DSM module.

**To Reproduce**
Start playback on one terminal(speaker with DSM or the headset pipeline)
localhost ~ # aplay -Dhw:0,0 Titanium_low1.wav -c2 -r48000 -fs24_le
start capture on other terminal:
localhost ~ # arecord -Dhw:0,99 try.wav -c4 -r48000 -fs32_le -d60 &
The moment the capture stream is opened
Playback returns I/O error:
aplay: pcm_write:2059: write error: Input/output error

**Reproduction Rate**
Always.
Very easy to repro.

**Expected behavior**
playback is on core 0 and IGO module runs on core 1.
The moment capture stream is started the playback on core 0 stalls.
Is it that core 0 is blocked here?

**Impact**
blocker for IGO integration

**Environment**
Tgl_13release + private patches in google repo+ Intelligo module(
have shared all the binaries through email as they are closed source.
Please let me know if you need more details.

**Screenshots or console output**
localhost ~ # ./sof-logger -l sof-tgl.ldc
TIMESTAMP (us) DELTA C# COMPONENT LOCATION CONTENT ktime=75.835s @ 2021-06-21 18:39:54 -0700 PDT
[ 105098936.344577] ( 0.000000) c0 dma-trace src/trace/dma-trace.c:335 ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.8-rc2-45-gd143acb3ece8 src hash 0x38d78a96 (ldc hash 0x38d78a96)
[ 20826327.349519] ( 20826328.000000) c0 dw-dma src/drivers/dw/dma.c:1100 ERROR dw_dma_get_data_size(): xrun detected
[ 20826344.641185] ( 17.291666) c0 dai 1.5 src/audio/dai.c:1069 ERROR dai_report_xrun(): underrun due to no data available
[ 20826361.984935] ( 17.343750) c0 dai 1.5 src/audio/dai.c:686 ERROR comp_underrun(): dev->comp.id = 5, source->avail = 384, copy_bytes = 0
[ 20826481.412013] ( 119.427078) c0 pipe 1.6 ....../pipeline-stream.c:168 ERROR pipeline_copy(): ret = -61, start->comp.id = 5, dir = 1
[ 20826500.422429] ( 19.010416) c0 pipe 1.6 ..../pipeline-schedule.c:63 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!

slawblauciak, 2021-06-29:
Here are my observations:

I was successfully able to reproduce the issue. At the point of problem occurring (I/O error in aplay) I could see an xrun in the FW trace logs, followed by a constant flow of warnings from the system agent, which shows a consistent problem in the DSP core taking too long to process.

`[    11611479.642768] (       31989.634766) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 370857`

I can get the same warnings to appear without aplay, just arecord running.

@RDharageswari
Reading the logs, it appears the period for this pipeline is 16 ms. Is that correct?

---

Performance counters show that the IGO component takes upwards of 3.8M cycles to complete a copy. Now, if one would assume that each instruction takes exactly one cycle, that'd make it 9.5 ms, which is well below 16 ms. The problem is that very assumption however, since various instructions take different amounts of cycles to complete and it all depends on the state of the DSP pipeline.
If we were to assume 1.5 cycles per instruction, that's already over 14 ms.

This finding in conjunction with the agent constantly complaining every other period reinforces my suspicion that the IGO simply takes too long to complete, as a result making the DSP core unable to respond to various events properly, such as IDC communication, which in turn causes other cores trying to communicate with it to stall, in the end, manifesting itself as this xrun.

I will continue debugging this, to ensure that the initial measurements are correct, this time using alternative method.

fuyuntsuo, 2021-06-30:
Thanks @slawblauciak . Your observation is correct that IGO  component takes about 9.5 ms to complete. That's why we moved it to Core 1 (c32a0e032fc04df2edaab0dd1b7f012f9c50d73e) so that other components and IDC shouldn't be affected by IGO's long computing time. We just don't understand why IDC still complains and why smart-amp turns out to be xrun even in this case.

Here is a information might help. Before moving IGO to Core 1, we used to extend smart-amp period to 16ms (776977852af006bf3f9bad4cc3ff8156b0c60450), prolonging buffer (we gusse) so that no xrun occurs. However since IGO is running on Core 1 no, there is no need to apply such patch. 

I am working on integrating IGO component on SOF. Please ping me if needed. Thanks.

fuyuntsuo, 2021-06-30:
Another question: does the performance counter count cycle or instruction? If cycle then 3.5M cycles should be okay. If instruction ,then cycle-per-instruction matters as you pointed out.

keyonjie, 2021-06-30:
@fuyuntsuo it's big information that the IGO component takes about 9.5 ms to complete each iteration, this will for sure blocks other tasks (e.g. 1ms SA task) regular runs.

ranj063, 2021-06-30:
> @fuyuntsuo it's big information that the IGO component takes about 9.5 ms to complete each iteration, this will for sure blocks other tasks (e.g. 1ms SA task) regular runs.

@keyonjie SA is disabled in the chrome prpducts.

keyonjie, 2021-06-30:
> > @fuyuntsuo it's big information that the IGO component takes about 9.5 ms to complete each iteration, this will for sure blocks other tasks (e.g. 1ms SA task) regular runs.
> 
> @keyonjie SA is disabled in the chrome prpducts.

Okay, I see that from @slawblauciak 's comment:
```
[ 11611479.642768] ( 31989.634766) c0 sa src/lib/agent.c:79 WARN validate(), ll drift detected, delta = 370857
```

Anyway, we still have bunch of pipelines scheduled in 1ms period.

@fuyuntsuo how could it happen to takes 9.5ms for each IGO .copy()? it's ~40 times of DSM or GHD!

fuyuntsuo, 2021-06-30:
Hi @keyonjie, let's make this assumption here if the answer is not blocking this issue. We could discuss why IGO takes much cycles via email since Intel has signed NDA with us. Can anyone from Intel bridge us?

fuyuntsuo, 2021-06-30:
May I know what does SA stand for? And why it is affected given IGO is running on Core 1.

ranj063, 2021-06-30:
> May I know what does SA stand for? And why it is affected given IGO is running on Core 1.

@fuyuntsuo SA is the system agent. It is not enabled in the chrome product FW. So its not a concern

cujomalainey, 2021-06-30:
@keyonjie GHD and DSM are all on Core 0, IGO is on Core 1, so SA aside, Core 1 should not block work on Core 0. So why is core 0 blocked on messages?

keyonjie, 2021-06-30:
@cujomalainey @fuyuntsuo yes, as all DSP cores share the same timer for timer driven tasks scheduling, let's talk about the details via email.

fuyuntsuo, 2021-06-30:
@RDharageswari @harshapriya-n @sathya-nujella Could you help bridge @cujomalainey and me with Intel engineers who are enthusiastically supporting this issue?

keyonjie, 2021-06-30:
@RDharageswari @sathya-nujella please create a thread about this, including me and Slawomir please.

sathya-nujella, 2021-06-30:
> @RDharageswari @sathya-nujella please create a thread about this, including me and Slawomir please.

Hi @keyonjie ,  Done.

keyonjie, 2021-06-30:
Hi @fuyuntsuo @RDharageswari @sathya-nujella I analyze the LL scheduler part and make some refinement in the PR #4436 , hope it helps.

slawblauciak, 2021-06-30:
Alright. I was at least partially incorrect in my last comment.
@fuyuntsuo your question regarding the CCOUNT actually made me realize that. Yes, it is cycles, meaning, the IGO copy does take exactly 9.5 ms. As such, my initial suspicion is completely incorrect.

However, after a short chat with Keyon, we came to conclusion that given the current scheduling paradigm, the FW cannot support different periods on different cores, given the way the timer is rearmed at a synchronized point in time.
This design problem is addressed by the PR here https://github.com/thesofproject/sof/pull/4436
I was able to test it and confirm that it is an effective solution for this issue.

RDharageswari, 2021-06-30:
Hi @keyonjie @slawblauciak,
I did test the patches from https://github.com/thesofproject/sof/pull/4436 and don't see the XRUN at my end with the IGO module and the speaker with DSM in the pipeline

fuyuntsuo, 2021-07-01:
Thanks for @slawblauciak @keyonjie @RDharageswari prompt action on this issue. I've also tested this patch on my TGL DUT and the issue has been solved. LL timer is no long complaining and speaker is not getting xrun neither.

@slawblauciak Yes. It would have been surprising if the performance counter shows instructions count instead of cycle count. Thanks for explanation!

RDharageswari, 2021-07-14:
I still see some issue in my testing. with the tgl-13 branch with the fixes above.
If i have the following configs enabled in kernel, I dont see any issues.
+CONFIG_SND_SOC_SOF_DEBUG=y
+CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE=y
+CONFIG_SND_SOC_SOF_DEVELOPER_SUPPORT=y

If i dont have the above in kernel i see the following issue:
Start playback on one terminal(speaker with DSM)
localhost ~ # aplay -Dhw:0,0 Titanium_low1.wav -c2 -r48000 -fs24_le
start capture on other terminal:
localhost ~ # arecord -Dhw:0,99 try.wav -c4 -r48000 -fs32_le -d60 &
The moment the capture stream is opened, playback stops.
There are no dmesg error logs.
I couldnt collect any fw trace logs as the issue goes away if i enable tracing.

While, the following sequence still works:
localhost ~ # arecord -Dhw:0,99 try.wav -c4 -r48000 -fs32_le -d60 &
Start playback(speaker with DSM)
localhost ~ # aplay -Dhw:0,0 Titanium_low1.wav -c2 -r48000 -fs24_le
PS: Capture is on core 1 while pb is on core0

slawblauciak, 2021-07-19:
> I still see some issue in my testing. with the tgl-13 branch with the fixes above.
> If i have the following configs enabled in kernel, I dont see any issues.
> +CONFIG_SND_SOC_SOF_DEBUG=y [...]

This looks like an entirely different issue altogether. The xrun doesn't seem to appear anymore, as the stream keeps going, even though there's nothing coming through the speakers. I don't know about you, but I also noticed occasional DSP panics originating from DMA, right here:

`static void dw_dma_verify_transfer(struct dma_chan_data *channel,
				   struct dma_cb_data *next)`

More specifically, when trying to dereference `lli->ctrl_hi`, in the while loop here:

`		while (lli->ctrl_hi & DW_CTLH_DONE(1)) {
			lli->ctrl_hi &= ~DW_CTLH_DONE(1);
			dw_chan->lli_current =
				(struct dw_lli *)dw_chan->lli_current->llp;
			lli = platform_dw_dma_lli_get(dw_chan->lli_current);
		}`

For some reason, current `lli` is 0. So, we get an oopsie.
I think this finding makes it highly probably that the source of the playback going silent is in DMA, the DSP panic merely being a secondary symptom.

lgirdwood, 2021-07-19:
> More specifically, when trying to dereference `lli->ctrl_hi`, in the while loop here:
> 
> ```c++
> 		while (lli->ctrl_hi & DW_CTLH_DONE(1)) {
> 			lli->ctrl_hi &= ~DW_CTLH_DONE(1);
> 			dw_chan->lli_current =
> 				(struct dw_lli *)dw_chan->lli_current->llp;
> 			lli = platform_dw_dma_lli_get(dw_chan->lli_current);
> 		}
> ```
> 
> For some reason, current `lli` is 0. So, we get an oopsie.
> I think this finding makes it highly probable that the source of the playback going silent is in DMA, the DSP panic merely being a secondary symptom.

So this could be a race between synchronizing the SRAM with the DMA HW LLI DONE bits and a pipeline state change (for start/stop etc). We probably need a spinlock around this and the state change (we may be missing a lock in one of these places).

fuyuntsuo, 2021-07-19:
We are able to reproduce the issue on our side, but the observation is quite different:

1. It does not relate to the kernel config. Whether the debug config is enabled or not, the issue can be reproduced.
2. This issue is barely reproducible using ChromeOS App (such as online audio record plus Youtube). Combined 1&2, it's very likely to be a timing or a memory issue.
3. The speaker xrun is not 100% reproducible. So the following comments might be "false positive" (aka. issue fixed) as we have been doing our best to avoid this from happening.
3. The issue seems fixed if we disable `igo_nr` component by using `sof-tgl-max98373-rt5682.tplg` instead of `sof-tgl-max98373-rt5682-igonr.tplg`
4. The issue is still reproducible if we do solo buffer consume/produce in `igo_nr` component. (by comment out [igo_nr.c#L624](https://github.com/thesofproject/sof/blob/0c21218b379d033d82616a651a7e702583762709/src/audio/igo_nr/igo_nr.c#L624)) 
5. Then we suspect the topology: inside `intel-generic-dmic-kwd.m4`, `DMIC_16k` and `DMIC_48k` are using 20ms and 16ms buffer respectively. By setting `INTEL_GENERIC_DMIC_KWD_PERIOD = KWD_PIPE_SCH_DEADLINE_US` (i.e., `INTEL_GENERIC_DMIC_KWD_PERIOD = 20000`). The issue seems fixed. (but why?)

Combined with what @slawblauciak mentioned, does DMA (of DMIC) support these kind of setting as mentioned in 6? We are not familiar with this part. Please comment if any of the items above doesn't make sense.

slawblauciak, 2021-07-22:
This new issue of playback audio going away is not related to the initially reported xrun problem. The xrun was caused by incorrect scheduling of multicore tasks in the firmware. This new problem however, is caused by something wrong going on in the DMA driver. The tasks and scheduling in this particular case are functioning correctly. As to what exactly is going on, I'm still trying to track down. But the xrun itself is fixed by a change in scheduler, which is already merged with the TGL-013 branch.

As to whether the DMIC pipeline periods you're using are okay, I don't really see any issue there.

fuyuntsuo, 2021-07-22:
Thanks for clarification @slawblauciak . Indeed the DMIC period workaround is a false positive. We did reproduce the issue later.

From what we have seen with tracev, the `read_ptr` below suddenly becomes to zero for unknown reason. Then XRUN occurs.

However we also find that both `channel->dma` and `DW_SAR(channel->index)` do not change compared to the previous `dw_dma_free_data_size()` being called on the same core 0. 

It seems to be a register reading issue, which requires hardware knowledge. So we decide to stop here.

```C
static int dw_dma_free_data_size(struct dma_chan_data *channel)
{
	struct dw_dma_chan_data *dw_chan = dma_chan_get_data(channel);
	int32_t read_ptr = dma_reg_read(channel->dma, DW_SAR(channel->index));
	int32_t write_ptr = dw_chan->ptr_data.current_ptr;
	int size;

	size = read_ptr - write_ptr;
	if (size < 0)
		size += dw_chan->ptr_data.buffer_bytes;

	if (!size)
		tr_info(&dwdma_tr, "dw_dma_free_data_size() size is 0!");

	tr_dbg(&dwdma_tr, "SAR %x writer 0x%x free 0x%x avail 0x%x", read_ptr,
	       write_ptr, size, dw_chan->ptr_data.buffer_bytes - size);

	return size;
}
```
Since

keyonjie, 2021-07-23:
Is the google KWD enabled with the test? Can we try this with Google Assistant disabled?
On the other side, is it possible to try this with the SOF main branch? We have several later DMIC multi-core related fixes on the main IIRC.

RDharageswari, 2021-07-26:
Hi Keyon,
I am able to repro the issue without Ghoogle Assistant also.

keyonjie, 2021-08-02:
@RDharageswari @fuyuntsuo I can't reproduce this issue with https://github.com/thesofproject/sof/pull/4575 applied.

Without that PR, I saw params error, so was unable to reproduce it.

sathya-nujella, 2021-08-02:
+ @yongzhi1  - Yong has tried. Requesting @yongzhi1  to share his observations too.

sathya-nujella, 2021-08-02:
Hi,
My observations on the issue: https://github.com/thesofproject/sof/issues/4418#issuecomment-879612135   with with #4575 applied.

1) It is not easy to reproduce the issue described above.  But please see (3) below.
I did this and could not catch the issue.

Youtube playback 
+
loop for 100 times with PCM99 capture.

//No issues here.

2)
To stress and look for different ways to repro issue: I introduced suspend/resume, I hit this DSP panic issue - https://github.com/thesofproject/sof/issues/4584

3)
I hit this specific issue (only once):
Steps and problem symptom:
Youtube playback
suspend-resume once
start PCM99 capture
Playback stops and no errors in dmesg.
Firmware logs show these errors:

[     6002035.907334] (          13.385416) c0 host         11.63         src/audio/host.c:377  no bytes to copy, 16 avail in buffer, 384 free in DMA
[     6003972.313507] (        1936.406128) c0 dw-dma                 src/drivers/dw/dma.c:1100 ERROR dw_dma_get_data_size(): xrun detected
[     6003981.844757] (           9.531250) c0 dai          11.67          src/audio/dai.c:1072 ERROR dai_report_xrun(): overrun due to no space available
[     6003990.803090] (           8.958333) c0 dai          11.67          src/audio/dai.c:704  ERROR comp_overrun(): sink->free = 0, copy_bytes = 0
[     6003999.657256] (           8.854166) c0 pipe         11.68 ....../pipeline-stream.c:184  pipe trigger cmd 8
[     6004010.021839] (          10.364583) c0 dai          11.67          src/audio/dai.c:963  dai_comp_trigger_internal(), XRUN
[     6004019.240588] (           9.218750) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:841  ssp_trigger() cmd 8
[     6004029.396838] (          10.156250) c0 dw-dma                 src/drivers/dw/dma.c:407  dw_dma_stop(): dma 1 channel 0 stop
[     6004044.605171] (          15.208333) c0 pipe         11.68 ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 67, dir = 0
[     6004053.251004] (           8.645833) c0 pipe         11.68 ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!

yongzhi1, 2021-08-02:
Tried the playback immediately followed by recording test with SOF FW trace debug turned off:

```
#!/bin/bash

for i in {1..10};

do
b=$(( ( RANDOM % 10 ) + 15 ))
aplay -Dhw:0,0 /home/test.wav &
arecord -Dhw:0,99 try.wav -c4 -r48000 -fs32_le -d60 -vv
sleep $b
echo "sleep $b round $i"
done
```

1. Observed DSP panic only once [dsp-panic.txt](https://github.com/thesofproject/sof/files/6919329/dsp-panic.txt)
2. Did not observe any other error in dmesg log.
3. Did not observe muted playback.
4. Observed high pitched noise during playback, repro rate: > 1/10 iterations.

ranj063, 2021-08-02:
@sathya-nujella @yongzhi1 can you please attach the dmesg with dynamic logs enabled?

yongzhi1, 2021-08-02:
With SOF debug enabled, here is the log captured at the moment when noise was heard

[dmesg-noise.txt](https://github.com/thesofproject/sof/files/6919578/dmesg-noise.txt)

Re-run the test with DSP trace log enabled:

```
localhost ~ # aplay -Dhw:0,0 /home/test.wav 
Playing WAVE '/home/test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2059: write error: Input/output error
```

[dsp-underrun.log](https://github.com/thesofproject/sof/files/6919549/dsp-underrun.log)

lgirdwood, 2021-08-03:
@RDharageswari @yongzhi1 looks like the SSP start is taking about 100uS to complete (10% of an LL time slice) and this could be having a knock on effect for the other components. i.e. a temporary over MCPS budget that causes an xrun as buffers fill/empty by HW.
```
[      117426.037001] (        2515.520752) c0 pipe         1.6   ....../pipeline-stream.c:184  pipe trigger cmd 1
[      117537.599496] (         111.562492) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:841  ssp_trigger() cmd 1
[      117548.016162] (          10.416666) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:630  ssp_pre_start()
[      117559.630745] (          11.614583) c0 mn                 ./drivers/intel/ssp/mn.c:258  mclk_rate 24576000, mclk_source_clock 0
[      117567.911995] (           8.281250) c0 mn                 ./drivers/intel/ssp/mn.c:220  mclk_id 0 mdivr_val 1
[      117579.266161] (          11.354166) c0 mn                 ./drivers/intel/ssp/mn.c:320  find_mn for freq 24576000 bclk 12288000
[      117588.953661] (           9.687500) c0 mn                 ./drivers/intel/ssp/mn.c:617  bclk_rate 12288000, *out_scr_div 2, m 1, n 1
[      117598.224494] (           9.270833) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:686  ssp_set_config(), sscr0 = 0xc7d0017f
[      117607.443243] (           9.218750) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:758  ssp_start()
[      117707.703656] (         100.260414) c0 ll-schedule        ./schedule/ll_schedule.c:413  task add 0xbe2d0180 pipe-task <f11818eb-e92e-4082-82a3-dc54c604ebb3>
[      117716.870322] (           9.166666) c0 ll-schedule        ./schedule/ll_schedule.c:417  task params pri 0 flags 0 start 0 period 1000
[      117734.161988] (          17.291666) c0 ll-schedule        ./schedule/ll_schedule.c:329  new added task->start 7407812 at 7401504
[      117742.339071] (           8.177083) c0 ll-schedule        ./schedule/ll_schedule.c:332  num_tasks 3 total_num_tasks 3
[      122419.682635] (        4677.343750) c0 dw-dma                 src/drivers/dw/dma.c:1100 ERROR dw_dma_get_data_size(): xrun detected
[      122430.463885] (          10.781250) c0 dai          1.5            src/audio/dai.c:1069 ERROR dai_report_xrun(): underrun due to no data available
[      122439.474301] (           9.010416) c0 dai          1.5            src/audio/dai.c:686  ERROR comp_underrun(): dev->comp.id = 5, source->avail = 560, copy_bytes = 0
```

plbossart, 2021-08-03:
@lgirdwood this can be optimized by doing an early_clk enablement for the SSP, so that the trigger only starts the data transfers
But we're still going to fail if we don't have a pre-rolled pipeline....

plbossart, 2021-08-03:
I have no idea what the last wait_delay(PLATFORM_SSP_DELAY) is for. 

This was added in commit 5d42e0fdf0 but for the 'slave mode' according to the commit message? It's very small anyways, 2400-3000-4800 cycles depending on the platform.

I am not sure it makes sense though to wait inside of a spinlock-protected section, this piece of code has always looked controversial to me.

plbossart, 2021-08-03:
We can make the wait_delay(PLATFORM_SSP_DELAY)  conditional on the slave mode.

lgirdwood, 2021-08-03:
> We can make the wait_delay(PLATFORM_SSP_DELAY) conditional on the slave mode.

Looking at the ordering flow in dai.c and ssp.c it looks like a NOP. i.e. we are not delaying in between any DMA/SSP IP programming flows. We only delay at the end for trigger(start) and at the beginning for trigger(stop).

plbossart, 2021-08-03:
Not following @lgirdwood if we wait to let the FIFO status settle doesn't this precisely impact the DMA flows?

ranj063, 2021-08-03:
@lgirdwood @plbossart maybe the SSP changes are needed anyway but I do want to call out that the original xrun issue is no longer reproducible. Even the muted playback when Igo capture is started is fixed with the latest backports. What remains is an occasional DSP panic with stress tests.

lgirdwood, 2021-08-03:
> Not following @lgirdwood if we wait to let the FIFO status settle doesn't this precisely impact the DMA flows?

AFAICT we are delaying for no reason, the DMA is started before the SSP e.g.
1) DMA start
2) SSP start
3) SSP FIFO Delay
4) IPC reply.

> @lgirdwood @plbossart maybe the SSP changes are needed anyway but I do want to call out that the original xrun issue is no longer reproducible. Even the muted playback when Igo capture is started is fixed with the latest backports. What remains is an occasional DSP panic with stress tests.

ok, this is good - but we should still look at removing these delays and validate on slave mode.

bardliao, 2021-08-04:
> > I am not sure it makes sense though to wait inside of a spinlock-protected section, this piece of code has always looked controversial to me.
> 
> Yeah, my thought it was to wait for some SSP/DMA handshake. It's 65uS on TGL. We also do this in `ssp_stop()`
> 
> I'm assuming there are no objections if we remove this ?
> 
> @keyonjie @bardliao any inputs here, you guys have worked on the SSP driver too.

It looks like the delay is for synchronization from the commit message, so I guess it is necessary. But as @plbossart mentioned, we can make it conditional on the slave mode

lgirdwood, 2021-08-04:
> It looks like the delay is for synchronization from the commit message, so I guess it is necessary. But as @plbossart mentioned, we can make it conditional on the slave mode

I dont see this code doing anything, its a delay for no reason (which could be lost with all the refactoring since Tomek left). I could see the point if the delay was in between DMA/SSP IP programming, but it's not. It's at the beginning of ssp_stop() and the end of ssp_start(). 

We can do this in consumer mode, but I would also like to test this here too.

bkokoszx, 2021-08-04:
Referring to @ranj063 comment: 
https://github.com/thesofproject/sof/issues/4418#issuecomment-892006190

Just to be on the same page: I also cannot reproduce xrun and mute playback issue with:
https://github.com/thesofproject/sof/pull/4575

@sathya-nujella @RDharageswari 
Can you confirm that with above changes you can reproduce only DSP panic issue occasionally?

sathya-nujella, 2021-08-04:
> 
> @sathya-nujella @RDharageswari
> Can you confirm that with above changes you can reproduce only DSP panic issue occasionally?

Hi,
For DSP panic, there were two reports:
1) @yongzhi1 reported and he observed once:  https://github.com/thesofproject/sof/issues/4418#issuecomment-891300799

2) I tried to see if any consistent way and got some test sequence which can repro - so created a new BUG:  https://github.com/thesofproject/sof/issues/4584

lgirdwood, 2021-08-04:
I have a feeling here these could possibly be all related. I am suspicious of cache coherency around core 0 LL pipelines and core 1 module using different scheduling time slices. i.e. 1ms vs ~12ms (according to trace). Both core 0 work and core 1 work will share some buffers for processing but core 0 will access the buffer 12 times to the core 1 time and we need to make sure that all accesses are coherent (we could be missing wb/inv at one point).

fuyuntsuo, 2021-08-05:
If I am not mistaken, all the IGO related issue is still tied to the topology `max98373-rt5682`. Maybe we can try to reproduce these issues on another topology, like, `max98357a-rt5682`. For reasons I don't understand, syndrome and reproduce rate could be totally different between different topologies (given the identical FW).

For example, there was a mystery issue I encountered few weeks ago: before applying @keyonjie 's PR #4436 :
* `max98373-rt5682` hit SSP XRUN all the time (100%) 
#4418
* `max98357a-rt5682` hit occasional DSP panic during solo PCM99 capture using arecord (<80%). I didn't raise this issue because it's solved by applying PR #4436 .

bkokoszx, 2021-08-06:
A little update:
With https://github.com/thesofproject/sof/pull/4575 sometimes I am still reproduce a problem with zeroing ```lli``` mentioned by @slawblauciak here: 
https://github.com/thesofproject/sof/issues/4418#issuecomment-882562833

keyonjie, 2021-08-09:
@bkokoszx I found that this may be related with the igo_nr component, just raised it to @fuyuntsuo and he is debugging it from the 3rd party library side at the same time.

fuyuntsuo, 2021-08-13:
status update: it's buffer clearing that causes speaker XRUN. I've provided two patches (NG and OK) to demonstrate this issue even in absence of IGO library.

fuyuntsuo, 2021-08-13:
I also find that inside `alloc_heap_buffer()` there is a `CONFIG_DEBUG_BLOCK_FREE` to zero the allocated memory. Is this CONFIG enabled by default?

keyonjie, 2021-08-13:
I checked the heap allocation, the dma lli buffer is next to the igo handler buffer (but no overlap), it is still strange to me that writing the igo handler buffer impacts the DW dma.

fuyuntsuo, 2021-08-13:
I am thinking of another possibility: xcc toolchain. I have tried to build the same firmware with different optimization (for size instead of for performance) and the XRUN issue is gone. Calling for an Xtensa expert for this issue :)

lgirdwood, 2021-08-13:
> I checked the heap allocation, the dma lli buffer is next to the igo handler buffer (but no overlap), it is still strange to me that writing the igo handler buffer impacts the DW dma.
> 
> @lgirdwood looks the cache line and the alignment here are playing tricks.

Could be true - can we make sure both DW DMA and IGO are doing the allocation with cache line size alignment. I would also state that the DMA LLI is written back by DMAC HW and may have a larger writeback size (i.e. may not play well with L1$ or uncache aliases).

Can we try the following patch that makes all LLI cache line size and adds padding of one unused LLI at the start and one unused LLI at the end of the LLI allocation.

```diff
diff --git a/src/drivers/dw/dma.c b/src/drivers/dw/dma.c
index 317def9fc..300328cff 100644
--- a/src/drivers/dw/dma.c
+++ b/src/drivers/dw/dma.c
@@ -558,8 +558,8 @@ static int dw_dma_set_config(struct dma_chan_data *channel,
 			rfree(dw_chan->lli);
 
 		dw_chan->lli = rballoc_align(0, SOF_MEM_CAPS_RAM | SOF_MEM_CAPS_DMA,
-					sizeof(struct dw_lli) * channel->desc_count,
-					PLATFORM_DCACHE_ALIGN);
+					sizeof(struct dw_lli) * channel->desc_count + 2,
+					PLATFORM_DCACHE_ALIGN * 2);
 		if (!dw_chan->lli) {
 			tr_err(&dwdma_tr, "dw_dma_set_config(): dma %d channel %d lli alloc failed",
 			       channel->dma->plat_data.id,
@@ -569,11 +569,16 @@ static int dw_dma_set_config(struct dma_chan_data *channel,
 		}
 	}
 
-	/* initialise descriptors */
-	bzero(dw_chan->lli, sizeof(struct dw_lli) * channel->desc_count);
-	lli_desc = dw_chan->lli;
-	lli_desc_head = dw_chan->lli;
-	lli_desc_tail = dw_chan->lli + channel->desc_count - 1;
+	/*
+	 * initialise LL descriptors by padding one unused LLI descriptor at
+	 * the start and one unused LL descriptor at the end. The DMAC HW may
+	 * have a larger IO size (TBD) for LLI writeback than the cacheline
+	 * size so add some padding to avoid clobbering adjacent users.
+	 */
+	bzero(dw_chan->lli, sizeof(struct dw_lli) * channel->desc_count + 2);
+	lli_desc = dw_chan->lli + 1;
+	lli_desc_head = dw_chan->lli + 1;
+	lli_desc_tail = dw_chan->lli + channel->desc_count - 2;
 
 	/* configure msize if burst_elems is set */
 	if (config->burst_elems) {
@@ -769,7 +774,7 @@ static int dw_dma_set_config(struct dma_chan_data *channel,
 
 	/* write back descriptors so DMA engine can read them directly */
 	dcache_writeback_region(dw_chan->lli,
-				sizeof(struct dw_lli) * channel->desc_count);
+				sizeof(struct dw_lli) * channel->desc_count + 2);
 
 	channel->status = COMP_STATE_PREPARE;
 	dw_chan->lli_current = dw_chan->lli;
diff --git a/src/include/sof/drivers/dw-dma.h b/src/include/sof/drivers/dw-dma.h
index b96e7adcd..866263da9 100644
--- a/src/include/sof/drivers/dw-dma.h
+++ b/src/include/sof/drivers/dw-dma.h
@@ -150,11 +150,10 @@ struct dw_lli {
 	uint32_t sstat;
 	uint32_t dstat;
 
-	/* align to 32 bytes to not cross cache line
-	 * in case of more than two items
-	 */
-	uint32_t reserved;
-} __packed;
+	/* align and pad to cache line bytes per LLI */
+	uint8_t cache_line[PLATFORM_DCACHE_ALIGN - sizeof(uint32_t) * 7];
+
+}  __attribute__((packed, aligned(PLATFORM_DCACHE_ALIGN)));
```

lgirdwood, 2021-08-13:
> I am thinking of another possibility: xcc toolchain. I have tried to build the same firmware with different optimization (for size instead of for performance) and the XRUN issue is gone. Calling for an Xtensa expert for this issue :)

Interesting, something fishy is going, I suspect the DMAC may be involved and changing the CC -O levels would certainly move TEXT/DATA around.

keyonjie, 2021-08-16:
> +	bzero(dw_chan->lli, sizeof(struct dw_lli) * channel->desc_count + 2);
> +	lli_desc = dw_chan->lli + 1;
> +	lli_desc_head = dw_chan->lli + 1;
> +	lli_desc_tail = dw_chan->lli + channel->desc_count - [...]

It requires some corrections, after that, the issue is W/A by the patch. @fuyuntsuo

fuyuntsuo, 2021-08-16:
@keyonjie @lgirdwood  thanks!

keyon, would you send the corrected patch to @lancedkhou, @sathya-nujella, @RDharageswari and me? So that we can verify whether the open issues can be closed with this W/A.

lgirdwood, 2021-08-16:
@keyonjie can you send a PR with your updates. I think the DMA does have a minimum IO size and alignment. We dont hit this with audio buffers as they are all > 64 bytes.

keyonjie, 2021-08-16:
@fuyuntsuo @lgirdwood the draft PR with the W/A here: https://github.com/thesofproject/sof/pull/4640

fuyuntsuo, 2021-08-16:
Hi @keyonjie @lgirdwood , I've tested it at my end and the XRUN doesn't occur anymore. Thank you both.

Apologize for being a bit pushy on this issue, whose root cause is totally beyond my expectation. Hopefully we could close all the other issues with help of this W/A.

keyonjie, 2021-08-17:
@fuyuntsuo good that it fix the issue at your end.

Yep, the root cause of the issue looks so interesting, we will not stop here, we need to go forward and pursue an explicit explanation of it.

[9 more replies]

closed 2022-01-25 by mengdonglin