thesofproject/sof #4419: [BUG] [TGL] Fw panic when we copy the firmware & topology binary to the file system first time and boot it

RDharageswari, 2021-06-28:
**Describe the bug**
Fw panic when we copy the firmware & topology binary to the file system first time and boot it

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
whenever we copy the firmware & topology for the first time on a board.
The problem goes away if we reboot

**Expected behavior**
No firmware panic and the sound card should bind successfully

**Impact**
showstopper

**Environment**
tgl-13 release.

**Screenshots or console output**
[    8.051731] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40070000 size 12
[    8.051740] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    8.051947] sof-audio-pci 0000:00:1f.3: error: runtime exception
[    8.051948] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[    8.051950] sof-audio-pci 0000:00:1f.3: error: panic at :0
[    8.051951] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

RDharageswari, 2021-06-28:
[FW_panic_while_boot.txt](https://github.com/thesofproject/sof/files/6729194/FW_panic_while_boot.txt)

lgirdwood, 2021-06-29:
@RDharageswari can you confirm that you copy files (how ? over NFS, scp ?) and then immediately reboot (how) ?
This could also be kernel issue or FS cache. Have you tried copy files then unload/reload modules ?
@ujfalusi anything in kernel flow for FW loading that would break here ?

bkokoszx, 2021-06-29:
@lgirdwood @RDharageswari 
I was able to reproduce this issue. I remove tplg and fw files, then I copy tplg/fw using scp and reboot by ```sudo reboot```. Sometimes after reboot I get dsp panic in dmesg. 

Once I also had a case, that just after reboot I did not have panic, but then I tried to run: 
```
aplay -Dhw:0,0 tmp.wav -c2 -r48000 -fs24_le -v
```
and after that I got similar panic.

ujfalusi, 2021-06-29:
Can it be not related to firmware copying and it is just a coincident? Or is it more likely to happen after delete fw/tplg, copy new fw/tplg, then reboot?
Fwiw, in the log the panic happens when loading the topology, specifically with a timeout of sending (SOF_IPC_GLB_PM_MSG|SOF_IPC_PM_CORE_ENABLE) message (`ipc timed out for 0x40070000 size 12`), it is hard to say if it is for the `pipeline->core` or for the `swidget->core` and also from where it happens.

Can it be reproduced if _only_ the firmware or the tplg file is deleted/copyied ?

After the first DSP crash you can reboot repeatably without any issue?

Can it be caused by unclean shutdown of the DSP in a way that same stale data remain from the previous boot and when we load the new tplg, the firmware tries to interpret it and fails?

Can you try to delete the fw/tplg files and then copy the same ones back? Does that also triggers a crash?
Can you try to update the fw/tplg and then shut down and then power on? Does the crash remain?
Can you reboot repeatedly without touching the fw/tplg and see if there is really no DSP crash?

I have been doing locally: update the fw/tplg, unload all drivers, load the drivers again and now the new firmware is in use. Can you try the same, so update the fw/tplg then remove/insert the modules and see if the crash happens also like that?

ranj063, 2021-06-29:
@RDharageswari can you confirm this is tgl-013 specific problem?

RDharageswari, 2021-06-29:
I have seen this problem only with tgl-13. Did not hit it before.

plbossart, 2021-06-29:
This reminds me of the odd behavior with firmware being changed reported by @marc-hb https://github.com/thesofproject/linux/pull/2954

Some sort of firmware caching issue?

marc-hb, 2021-06-29:
I'm confused by the description of this bug: is the same, valid and sane firmware used all the time in every single reproduction steps?

The problem solved (for me) by https://github.com/thesofproject/linux/pull/2954 is much simpler to describe:
1. Load a firmware that invokes panic() _voluntarily at a certain point._
2. That panicky firmware gets CACHED until the next reboot, no way to replace it. The kernel drivers keep reloading the same version, it's stuck to it no matter what's in `/lib/firmware/`, including nothing at all.

I'm resuming the investigation on 2954 now in case it helps this too.

PS: why is this P1? It affects only developers and no product, correct?

ranj063, 2021-06-29:
> PS: why is this P1? It affects only developers and no product, correct?

no the issue is seen on a TGL product

marc-hb, 2021-06-29:
I meant: this does not seem like it would affect a _consumer_, sorry for the confusion.

lgirdwood, 2021-06-30:
> Some sort of firmware caching issue?

This is what I was thinking too, VFS cache could do this but ```sudo reboot``` done by @bkokoszx should do a ```sync``` and writeback to disk.

@bkokoszx can you try and execute ```sync fw_file_name``` to forcibly flush VFS cache version of FW prior to reboot.

bkokoszx, 2021-06-30:
@lgirdwood 
I did so and I was able to reproduce this issue with ```sync fw_file``` before ```sudo reboot```.

I was also trying to analyze exception and stack logs from dmesg with ```sof-tgl.lst``` file. I do not have any special conclusions yet, but we can see that:
```
EXCVADDR 0xbe1d0000
```
points to the:
```
be1d0000 <_sof_core_s_start>
```
, which is used in: ```cavs_pm_runtime_core_en_memory()``` and ```cavs_pm_runtime_core_dis_memory()``` functions. I will try to debug more.

ranj063, 2021-06-30:
@bkokoszx https://github.com/thesofproject/linux/pull/3018 fixes the issue. I still need to refine it to align with other multi-core changes but the underlying problem is to disables the secondary cores before shutdown

marc-hb, 2021-06-30:
According to the bug description the problem happens when the host boots. How can a DSP shutdown fix make a difference?

ranj063, 2021-06-30:
its not just a boot, its a reboot. So shutdown can make a difference. We've seen issues similar to this in the past and we have a shutdown callback for the PCI device to put the DSP in D3 already. This just adds on to it

RDharageswari, 2021-06-30:
Today i was still able to repro the issue, even with https://github.com/thesofproject/linux/pull/3018

RDharageswari, 2021-07-01:
Even when the same firmware file is overwritten we hit the issue.
This issue is not consistently observed every time.

RDharageswari, 2021-07-01:
with the following changes @bkokoszx suggested:
diff --git a/src/platform/intel/cavs/lib/pm_runtime.c b/src/platform/intel/cavs/lib/pm_runtime.c
index aba17421135a..84970c6366fc 100644
--- a/src/platform/intel/cavs/lib/pm_runtime.c
+++ b/src/platform/intel/cavs/lib/pm_runtime.c
@@ -328,8 +328,8 @@ static inline void cavs_pm_runtime_core_dis_memory(uint32_t index)
        /* Address is calculated for index (0 for the primary core) minus one
         * since _sof_core_s_start is first secondary core stack address
         */
-       core_memory_ptr = (char *)&_sof_core_s_start
-               + (index - 1) * SOF_CORE_S_SIZE;
+       core_memory_ptr = cache_to_uncache((char *)&_sof_core_s_start
+               + (index - 1) * SOF_CORE_S_SIZE);

        cavs_pm_memory_hp_sram_power_gate(core_memory_ptr, SOF_CORE_S_SIZE,
                                          false);
@@ -346,8 +346,8 @@ static inline void cavs_pm_runtime_core_en_memory(uint32_t index)
        /* Address is calculated for index (0 for the primary core) minus one
         * since _sof_core_s_start is first secondary core stack address
         */
-       core_memory_ptr = (char *)&_sof_core_s_start
-               + (index - 1) * SOF_CORE_S_SIZE;
+       core_memory_ptr = cache_to_uncache((char *)&_sof_core_s_start
+               + (index - 1) * SOF_CORE_S_SIZE);

        cavs_pm_memory_hp_sram_power_gate(core_memory_ptr, SOF_CORE_S_SIZE,
                                          true);
I am not hitting the failure...tried multiple times..will ocntinue my testing

marc-hb, 2021-07-01:
Additional comments in #4441

marc-hb, 2021-07-02:
I just got to the bottom of the other issue: the failure to reload a new firmware after a firmware panic, see latest commit message in `release_firmware()` fix https://github.com/thesofproject/linux/pull/2954

tl;dr: it was not "caching" but "batching"

This could interact with this different issue and make it even harder to test, so I suggest testing with https://github.com/thesofproject/linux/pull/2954 too.

bkokoszx, 2021-07-02:
I've also tried change mentioned by @RDharageswari (https://github.com/thesofproject/sof/issues/4419#issuecomment-872588622) and I did not have reproduction (tried several times ~20).

slawblauciak, 2021-07-02:
Relevant PR created here: https://github.com/thesofproject/sof/pull/4446

slawblauciak, 2021-07-08:
The PR is unfortunately not correct as it prevents SRAM power gating. However, that does point at a probable area where the fault exists. There appears to be a race condition happening with SRAM power gating. Simply adding as much as one additional trace point for debug makes the issue go away, making this problem extremely difficult to debug. However, that does present a potential workaround: simply delaying the power gating mechanism very slightly.
Unfortunately I have no other conclusions on this matter at this point, I'm still debugging it.

slawblauciak, 2021-07-08:
@RDharageswari please try this PR https://github.com/thesofproject/sof/pull/4466

It makes the delay for power gating HP SRAM longer.

sathya-nujella, 2021-07-12:
> @RDharageswari please try this PR #4466
> 
> It makes the delay for power gating HP SRAM longer.

Hi @slawblauciak 
FYI,
I got a script which does the below:
chrome DUT:
       loop {
	mount -o remount,rw /
	cp /root/sof-tgl.ri /lib/firmware/intel/sof/community/
	cp /root/sof-tgl-max98373-rt5682.tplg /lib/firmware/intel/sof-tplg/voxel/
         sync
        reboot 
       }
Within 10 cycles of the script, I was able to repro the original issue with tgl_13.

   Using this script, I have verified PR #4466 : 'DSP panic' issue is **not reproducible**.  Thanks!

marc-hb, 2021-07-12:
@sathya-nujella I really don't understand how copying the firmware files can make any difference. Linux does not have any `/lib/firmware` "memory" that persists across reboots, if it had that would be a serious (security?) issue.

Can you please try to reproduce with the same loop but without the copies? (and without #4466). Everything else the same.

In other words I don't believe the title of this bug.

sathya-nujella, 2021-07-12:
> @sathya-nujella I really don't understand how copying the firmware files can make any difference. Linux does not have any `/lib/firmware` "memory" that persists across reboots, if it had that would [...]

Yes @marc-hb . 
Validation teams has tried normal reboot cycles without copying any firmware & topologies: no issues reported.

Issue did not report in previous TGL releases, we got reports only in TGL_13 based.

marc-hb, 2021-07-12:
> Validation teams has tried normal reboot cycles without copying any firmware & topologies: no issues reported.

Maybe this was just because the original firmware is not affected, only the new one.

slawblauciak, 2021-07-12:
@marc-hb I was able to reproduce the issue as well. It only occurs after the FW binary is replaced.

marc-hb, 2021-07-12:
> It only occurs after the FW binary is replaced.

With a different version? If not then it could be timing issue, has anyone tried to replace another, unrelated firmware file?

slawblauciak, 2021-07-12:
> > It only occurs after the FW binary is replaced.
> 
> With a different version? If not then it could be a timing issue, has anyone tried to replace another, unrelated firmware file?
> 
> A delay can change the state of the DSP when the reboot starts.

No, the same version even.

marc-hb, 2021-07-12:
OK then this must be a timing issue. Can you please try to copy similar and unrelated files in the same directories and see if it reproduces the same?

closed 2025-07-24 by github-actions[bot]