thesofproject/sof #4420: [Query] How can I use sof-logger with Zephyr .ri image

iuliana-prodan, 2021-06-28:
**Describe the query**
I have a zephyr.ri image, obtained by building SOF with Zephyr.
This is loaded and booted and I have my sof card available:
```
root@imx8qmmek:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 [59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofwm8960audio [sof-wm8960-audio], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@imx8qmmek:~# 
```

How can I use sof-logger to check the traces obtained in SOF?
What .ldc file can I use?
How can I obtained the .ldc file from SOF with Zephyr? Is it possible?

**To Reproduce**
I've tried reading the trace with the ldc file obtained from SOF with XTOS and I'm getting:
```
root@imx8qmmek:~# ./sof-logger -t -l /lib/firmware/imx/sof-gcc/sof-imx8.ldc 2>&1 | tee trace.txt
error: src hash value from version file (0x0) differ from src hash version saved in dictionary (0x2fd09907).
root@imx8qmmek:~# ./sof-logger  -l /lib/firmware/imx/sof-gcc/sof-imx8.ldc 2>&1 | tee trace.txt
error: Unable to open in file /sys/kernel/debug/sof/etrace: No such file or directory
root@imx8qmmek:~# 
```

marc-hb, 2021-06-28:
I'm working on this, will hopefully submit something working this week. Related:

Add new mtrace_printf() and use it for DMA init error handling and demoting banner to INFO level #4389
Restore early tracing #4334
ipc_send_queued_msg() has a hard dependency on the EDF scheduler #4356

Also take note of
[BUG]Empty or stuck DMA trace - Initial FW ABI banner message not found in logger.data.txt #4333

marc-hb, 2021-06-28:
Sorry forgot to mention that there is a partial solution that already works: https://github.com/zephyrproject-rtos/zephyr/pull/35186

At least it works for Intel hardware, dunno about IMX sorry.

iuliana-prodan, 2021-06-29:
> Sorry forgot to mention that there is a partial solution that already works: [zephyrproject-rtos/zephyr#35186](https://github.com/zephyrproject-rtos/zephyr/pull/35186)
> 
> At least it works for Intel hardware, dunno about IMX sorry.

Thanks, @marc-hb!
For now, I'll try the partial solution, see if it works for i.MX.

marc-hb, 2021-07-06:
BTW you'll need to revert 4e8743f76f82 `zephyr: imx: split initialization` that just got merged. I'm curious why you needed that, maybe it was just a copy/paste of cavs commit cf8e35f860a73? I submitted a revert of the latter in #4452

iuliana-prodan, 2021-07-08:
> BTW you'll need to revert [4e8743f](https://github.com/thesofproject/sof/commit/4e8743f76f8278b897fc6f4af2751b8d2964fe62) `zephyr: imx: split initialization` that just got merged in #4217 . I'm curious why you needed that, maybe it was just a copy/paste of cavs commit [cf8e35f](https://github.com/thesofproject/sof/commit/cf8e35f860a735fda9b423a2fa71d24ae2e309db)? I submitted a revert of the latter in #4452

@marc-hb I've tried you're commits from #4452 and I've reverted mine - [here's ](https://github.com/iuliana-prodan/sof/commit/194f0b1bb60a8e7ee41cfab06993a54926d55cbd) the commit from my branch, but now I cannot load the topology:
```
root@imx8qmmek:~# ./loadSof.sh 
[   57.816714] sof-audio-of 556e8000.dsp: DT DSP detected
[   57.827732] imx-dsp imx-dsp: NXP i.MX DSP IPC initialized
[   57.841229] sof-audio-of 556e8000.dsp: warning: unknown sof_ext_man header type 5 size 0x20
[   57.849816] sof-audio-of 556e8000.dsp: warning: unknown sof_ext_man header type 3 size 0x30
[   57.858286] sof-audio-of 556e8000.dsp: Firmware info: version 1:8:0-194f0
[   57.865102] sof-audio-of 556e8000.dsp: Firmware: ABI 3:19:0 Kernel ABI 3:17:0
[   57.872278] sof-audio-of 556e8000.dsp: warn: FW ABI is more recent than kernel
[   57.882182] sof-audio-of 556e8000.dsp: Firmware info: version 1:8:0-194f0
[   57.888975] sof-audio-of 556e8000.dsp: Firmware: ABI 3:19:0 Kernel ABI 3:17:0
[   57.896111] sof-audio-of 556e8000.dsp: warn: FW ABI is more recent than kernel
[   58.432421] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x90030000 size 80
[   58.440033] sof-audio-of 556e8000.dsp: info: preventing DSP entering D3 state to preserve context
[   58.448985] sof-audio-of 556e8000.dsp: invalid header size 0x202010. FW oops is bogus
[   58.457092] sof-audio-of 556e8000.dsp: error: unexpected fault 0x40270400 trace 0x40270400
[   58.465388] sof-audio-of 556e8000.dsp: error: can't set params for DMA for trace -110
[   58.473255] sof-audio-of 556e8000.dsp: warning: failed to initialize trace -110
[   58.505797] sof-audio-of 556e8000.dsp: Topology: ABI 3:18:1 Kernel ABI 3:17:0
[   58.513276] sof-audio-of 556e8000.dsp: warn: topology ABI is more recent than kernel
[   59.040417] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x40070000 size 12
[   59.048024] sof-audio-of 556e8000.dsp: info: preventing DSP entering D3 state to preserve context
[   59.057025] sof-audio-of 556e8000.dsp: invalid header size 0x202010. FW oops is bogus
[   59.064882] sof-audio-of 556e8000.dsp: error: unexpected fault 0x40270400 trace 0x40270400
[   59.073163] sof-audio-of 556e8000.dsp: error: core 0 enable ipc failure -110
[   59.080221] sof-audio-of 556e8000.dsp: error: enable core: -110
[   59.086155] sof-audio-of 556e8000.dsp: ASoC: failed to load widget PCM0P
[   59.092865] sof-audio-of 556e8000.dsp: ASoC: topology: could not load header: -110
[   59.100447] sof-audio-of 556e8000.dsp: error: tplg component load failed -110
[   59.107609] sof-audio-of 556e8000.dsp: error: failed to load DSP topology -22
[   59.114764] sof-audio-of 556e8000.dsp: ASoC: error at snd_soc_component_probe on 556e8000.dsp: -22
[   59.123739] sof-audio-of 556e8000.dsp: ASoC: failed to probe component -22
[   59.130654] asoc-simple-card sof-sound-wm8960: ASoC: failed to instantiate card -22
[   59.138430] asoc-simple-card: probe of sof-sound-wm8960 failed with error -22
root@imx8qmmek:~# 

```
Is this working on Intel's platforms with Zephyr?

On i.MX, without enabling dma trace, so without [this](https://github.com/iuliana-prodan/sof/commit/194f0b1bb60a8e7ee41cfab06993a54926d55cbd) commit, I can load the topology and have my sof audio card:
```
root@imx8qmmek:~# ./loadSof.sh 
[  157.834000] sof-audio-of 556e8000.dsp: DT DSP detected
[  157.844449] imx-dsp imx-dsp: NXP i.MX DSP IPC initialized
[  157.858352] sof-audio-of 556e8000.dsp: warning: unknown sof_ext_man header type 5 size 0x20
[  157.866754] sof-audio-of 556e8000.dsp: warning: unknown sof_ext_man header type 3 size 0x30
[  157.875402] sof-audio-of 556e8000.dsp: Firmware info: version 1:8:0-69048
[  157.882233] sof-audio-of 556e8000.dsp: Firmware: ABI 3:19:0 Kernel ABI 3:17:0
[  157.889397] sof-audio-of 556e8000.dsp: warn: FW ABI is more recent than kernel
[  157.899479] sof-audio-of 556e8000.dsp: Firmware info: version 1:8:0-69048
[  157.906276] sof-audio-of 556e8000.dsp: Firmware: ABI 3:19:0 Kernel ABI 3:17:0
[  157.913412] sof-audio-of 556e8000.dsp: warn: FW ABI is more recent than kernel
[  157.935184] sof-audio-of 556e8000.dsp: Topology: ABI 3:18:1 Kernel ABI 3:17:0
[  157.942567] sof-audio-of 556e8000.dsp: warn: topology ABI is more recent than kernel
[  157.960284] sof-audio-of 556e8000.dsp: tplg: config SAI1 fmt 0x1 mclk 12288000 width 32 slots 2 mclk id 0
[  157.970725] sof-audio-of 556e8000.dsp: ASoC: Parent card not yet available, widget card binding deferred

root@imx8qmmek:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 [59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofwm8960audio [sof-wm8960-audio], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@imx8qmmek:~# 
```
Am I forgetting something?

marc-hb, 2021-07-08:
@iuliana-prodan first a disclaimer: I'm not claiming the DMA trace is ready for prime time, not even with #4452 . This being said, I do not observe any or the errors you posted.  I have "other problems" :-)

In general I think a good rule of thumb is to ignore all errors past the first one. So  I would first look into the sof_ext_man and ABI warnings - unless of course you know exactly what they mean.

Then the `error: ipc timed out for 0x90030000 size 80` seems interesting, it looks a bit like #4428. Strange.

marc-hb, 2021-07-08:
Forgot this sorry: we are seeing scheduling issues and IPC timeouts in Zephyr even without enabling the DMA trace. So it could also be just the messenger.

If you don't absolutely need it right now and I would recommend not trying it yet.

marc-hb, 2021-07-11:
>  I have "other problems" :-)

More specifically:
- the crash introduced by your PR #4217 
- the scheduler regression introduced by PR #4377 

After reverting these two locally I think I'm back on track.

marc-hb, 2021-07-16:
A few rough edges and not reviewed yet but PR #4503 works4me if you want to give it a try. Tested only with a revert of #4377 for now.

marc-hb, 2021-12-03:
Mostly works now

closed 2021-12-03 by marc-hb