thesofproject/sof #4421: [BUG] TGLU_RVP_SDW: DSP panic on suspend/resume

plbossart, 2021-06-28:
**Describe the bug**

https://sof-ci.01.org/linuxpr/PR3006/build5945/devicetest/?model=TGLU_RVP_SDW&testcase=check-suspend-resume-with-capture

**To Reproduce**
cmd: TPLG=sof-tgl-rt711-rt1308-4ch.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 1 -m capture

**Reproduction Rate**
Seen once so far

**Expected behavior**
No DSP panic

**Impact**
showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6b53609ca780 ASoC: SOF: pipelines: Harmonize all functions to use struct snd_sof_dev
     * SOF: bccecb1a dai: only process DAI config for comps on cores that enabled
  
2) Name of the topology file
    * Topology: sof-tgl-rt711-rt1308-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_RVP_SDW:

mengdonglin, 2021-06-29:
Issue observed in testing PR https://github.com/thesofproject/linux/pull/3006

plbossart, 2022-03-11:
https://github.com/thesofproject/linux/pull/3006 is merged and we haven't seen this problem, closing

closed 2022-03-11 by plbossart