thesofproject/sof #4422: [BUG]Dai failed to alloc dma buffer when running multiple-pipeline on JSL_RVP_NOCODEC

XiaoyunWu6666, 2021-06-29:
**Describe the bug**
dai failed to alloc dam buffer when running multiple-pipeline on   JSL_RVP_NOCODEC

**To Reproduce**
 TPLG=sof-jsl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 5b851f48
SOF Branch: main
SOF Commit: bccecb1a8eed
TPLG:sof-jsl-nocodec.tplg
Platform: JSL_RVP_NOCODEC

**Environment**
Start Time: 2021-06-27 21:21:11 UTC
End Time: 2021-06-28 02:15:54 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: 5b851f48
SOF Branch: main
SOF Commit: bccecb1a8eed 

**Screenshots or console output**
[console]
```
```
[dmesg]
```
```
[sof-logger]
```
```

lgirdwood, 2021-06-29:
Ok I think the memory errors today are all related, lets see the heap map and fix them.

XiaoyunWu6666, 2021-07-14:
dma buffer allocation issue seems to disappear from daily test result .
shall we close it ? and reopen once it happened again

plbossart, 2021-07-14:
The same error was seen in https://sof-ci.sh.intel.com/#/result/planresultdetail/5261?model=JSL_RVP_NOCODEC&testcase=multiple-pipeline-all

plbossart, 2021-07-16:
and it's not visible in test 5308. let's see if this continues....

lgirdwood, 2021-07-16:
@XiaoyunWu6666 can you retest, @keyonjie heap PR is merged so lots of the recent memory issues can be retested and closed if good.

XiaoyunWu6666, 2021-07-19:
> @XiaoyunWu6666 can you retest, @keyonjie heap PR is merged so lots of the recent memory issues can be retested and closed if good.

Sure. I will give the feedback soon

closed 2021-07-20 by mengdonglin