thesofproject/sof #4423: [BUG][TGLH] IPC error when BUFFER_NEW , while multiple-pipeline on TGLH_0A5E_SDW and TGLH_0A70_HDA

XiaoyunWu6666, 2021-06-29:
**Describe the bug**
in inner daily test 4928, model=TGLH_0A5E_SDW, testcase=multiple-pipeline-all
when running multiple pipelines to arecord
ipc error happened when BUFFER_NEW

**To Reproduce**
TPLG=sof-hda-generic-2ch.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20
If 'TPLG=sof-hda-generic-2ch.tplg ~/sof-test/test-case/multiple-pipeline.sh -f c -c 20' to run capture first then play, it went wrong also.

**Reproduction Rate**
100%
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 6b53609c
SOF Branch: main
SOF Commit: 6bbf6b0d3cfa
 Platform: TGLH_0A5E_SDW TGLH_0A70_HDA
**Screenshots or console output**
[dmesg]
```
[ 3725.445992] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[ 3725.445994] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PGA2.0 setup complete
[ 3725.445996] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[ 3725.446124] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x30200000 size 20
[ 3725.446127] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget BUF2.1
[ 3725.446130] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.446647] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.446650] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PCM0C freed
[ 3725.446652] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 3725.446770] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30210000: GLB_TPLG_MSG: BUFFER_FREE
[ 3725.446771] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF2.0 freed
[ 3725.446773] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.446865] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.446866] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PGA2.0 freed
[ 3725.446868] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 3725.447658] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 3725.447663] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.2.HDA1.IN freed
[ 3725.447664] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 1
[ 3725.447667] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[ 3725.447678] kernel:  HDA Analog: ASoC: soc_pcm_hw_params() failed (-12)
[ 3725.447693] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 3725.447872] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 3725.447875] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.448044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 3725.448045] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget HDA1.IN freed
[ 3725.448052] kernel:  HDA Analog: ASoC: dpcm_fe_dai_hw_params failed (-12)
```

lgirdwood, 2021-06-29:
@XiaoyunWu6666 I think this is out of memory too. What does the trace show for heap usage ?

XiaoyunWu6666, 2021-06-30:
@lgirdwood 
log from inner daily  4961  model=TGLH_0A70_HDA   testcase=multiple-pipeline-all
```
[      597386.017929] (        1090.052002) c0 memory                      src/lib/alloc.c:1073 INFO   used 0 free 64592
[      597408.309594] (          22.291666) c0 memory                      src/lib/alloc.c:1099 INFO heap: runtime status
[      597425.340844] (          17.031250) c0 memory                      src/lib/alloc.c:1071 INFO  heap: 0xbe080000 size 63488 blocks 7 caps 0x45
[      597442.163760] (          16.822916) c0 memory                      src/lib/alloc.c:1073 INFO   used 7680 free 55808
[      597458.934592] (          16.770832) c0 memory                      src/lib/alloc.c:1081 INFO   block 1 base 0xbe082000 size 128
[      597475.757508] (          16.822916) c0 memory                      src/lib/alloc.c:1084 INFO    count 64 free 43
[      597494.038758] (          18.281250) c0 memory                      src/lib/alloc.c:1081 INFO   block 2 base 0xbe084000 size 256
[      597512.997090] (          18.958332) c0 memory                      src/lib/alloc.c:1084 INFO    count 128 free 114
[      597531.017923] (          18.020832) c0 memory                      src/lib/alloc.c:1081 INFO   block 3 base 0xbe08c000 size 512
[      597548.257506] (          17.239582) c0 memory                      src/lib/alloc.c:1084 INFO    count 8 free 8
[      597564.715838] (          16.458332) c0 memory                      src/lib/alloc.c:1081 INFO   block 4 base 0xbe08d000 size 1024
[      597581.903338] (          17.187500) c0 memory                      src/lib/alloc.c:1084 INFO    count 4 free 4
[      597634.715835] (          52.812496) c0 memory                      src/lib/alloc.c:1102 INFO heap: runtime shared status
[      597651.538751] (          16.822916) c0 memory                      src/lib/alloc.c:1071 INFO  heap: 0x9e11d000 size 31744 blocks 5 caps 0x45
[      597668.361667] (          16.822916) c0 memory                      src/lib/alloc.c:1073 INFO   used 24704 free 7040
[      598013.465820] (         345.104156) c0 dma-trace                 src/trace/trace.c:135  INFO Suppressed 6 similar messages:   block %d base 0x%x size %d
[      598025.288737] (          11.822916) c0 dma-trace                 src/trace/trace.c:135  INFO Suppressed 6 similar messages:    count %d free %d
[      598038.413736] (          13.124999) c0 component             src/ipc/ipc3/helper.c:418  INFO comp new host <8b9d100c-6d78-418f-90a3-e0e805d0852b> type 1 id 2.9
[      598080.549151] (          42.135414) c0 dma                           src/lib/dma.c:121  INFO dma_get() ID 4 sref = 4 busy channels 3
[      598294.142893] (         213.593735) c0 buffer                src/ipc/ipc3/helper.c:882  INFO buffer new size 0x300 id 2.12 flags 0x0
[      598791.278290] (         497.135406) c0 dai          7.45           src/audio/dai.c:949  WARN dai_copy(): Copy_bytes 368 + avail bytes 0 < period bytes 384, possible glitch
[      599581.174091] (         789.895813) c0 component             src/ipc/ipc3/helper.c:418  INFO comp new pga  type 5 id 2.10
[      599622.476173] (          41.302082) c0 pga          2.10  ../audio/volume/volume.c:374  INFO vol->initial_ramp = 400, vol->ramp = 0, vol->min_value = 0, vol->max_value = 2072775
[      599857.684497] (         235.208328) c0 pga          2.10  ../audio/volume/volume.c:554  INFO volume_ctrl_set_cmd(), channel = 0, value = 2072775
[      599873.674080] (          15.989583) c0 pga          2.10  ../audio/volume/volume.c:554  INFO volume_ctrl_set_cmd(), channel = 1, value = 2072775
[      600187.059484] (         313.385406) c0 buffer                src/ipc/ipc3/helper.c:882  INFO buffer new size 0x300 id 2.13 flags 0x0
[      600280.757397] (          93.697914) c0 ipc                  src/ipc/ipc3/handler.c:1212 ERROR ipc: pipe 2 buffer 13 creation failed -12
[      601369.663604] (        1088.906250) c0 ipc                  src/ipc/ipc3/handler.c:1288 INFO ipc: comp 9 -> free
[      601468.205266] (          98.541664) c0 host         2.9           src/audio/host.c:577  INFO host_free()
```

lgirdwood, 2021-06-30:
@XiaoyunWu6666 thanks - that the info we need. @keyonjie can you look at this and it appears trace filtering has switched OFF some of the heap dump so we are missing info.

XiaoyunWu6666, 2021-07-14:
buffer_new issue seems to disappear from daily test result .
shall we close it ? and reopen once it happened again

lgirdwood, 2021-07-16:
Fixed by @keyonjie heap patch.

closed 2021-07-16 by lgirdwood

XiaoyunWu6666, 2021-07-19:
in inner daily 5308 , this bug did not happen