thesofproject/sof #4424: [BUG]TGLU_VOLT_SDW dai_config() error : No memory for dai_config

XiaoyunWu6666, 2021-06-29:
**Describe the bug**
TGLU_VOLT_SDW dai_config() error : No memory for dai_config occurred when running multiple pipeline

**To Reproduce**
TPLG=sof-tgl-sdw-max98373-rt5682.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20
If 'TPLG=sof-hda-generic-2ch.tplg ~/sof-test/test-case/multiple-pipeline.sh -f c -c 20' to run capture first then play, it went wrong also.
**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 6b53609c
SOF Branch: main
SOF Commit: 6bbf6b0d3cfa
Topology: sof-tgl-sdw-max98373-rt5682.tplg
 Platform: TGLU_VOLT_SDW

**Screenshots or console output**
[sof-logger]
```
[      889853.870890] (      889853.875000) c0 dai          7.35        src/ipc/ipc3/dai.c:253  ERROR dai_config(): No memory for dai_config.
[      889872.568806] (          18.697916) c0 ipc                      src/ipc/ipc3/dai.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed
[      891682.620818] (        1810.052002) c0 dai          8.41        src/ipc/ipc3/dai.c:253  ERROR dai_config(): No memory for dai_config.
[      891700.589567] (          17.968750) c0 ipc                      src/ipc/ipc3/dai.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed
[      895004.026936] (        3303.437256) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      895023.089435] (          19.062500) c0 host         4.19          src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      895040.433184] (          17.343750) c0 pipe         4.24  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 19
[      895059.547767] (          19.114582) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 4 comp 19 params failed -12
[      897018.870606] (        1959.322876) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      897037.724772] (          18.854166) c0 host         11.49         src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      897054.912271] (          17.187500) c0 pipe         11.52 ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 49
[      897074.391437] (          19.479166) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 11 comp 49 params failed -12
[      899017.308026] (        1942.916626) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[      899036.005942] (          18.697916) c0 host         2.6           src/audio/host.c:739  ERROR host_params(): failed to alloc dma buffer
[      899053.349691] (          17.343750) c0 pipe         2.11  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 6
[      899072.360107] (          19.010416) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 2 comp 6 params failed -12
```
[dmesg]
```
[ 4602.138185] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 4602.138261] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 4602.138774] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 4602.138791] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[ 4602.138913] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 4602.139162] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[ 4602.139182] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting DAI config for HDA1.OUT
[ 4602.139189] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp2 Pin: -12
[ 4602.139246] kernel:  iDisp2: ASoC: soc_pcm_hw_params() failed (-12)
[ 4602.139273] kernel:  HDMI 2: ASoC: dpcm_fe_dai_hw_params failed (-12)
[ 4602.139288] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 6 dir 0
[ 4602.139833] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 4602.139901] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 4602.139909] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 5 dir 0 cmd 1
[ 4602.140014] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[ 4602.140035] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed setting DAI config for HDA2.OUT
[ 4602.140039] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -12
[ 4602.140062] kernel:  iDisp3: ASoC: soc_pcm_hw_params() failed (-12)
[ 4602.140073] kernel:  HDMI 3: ASoC: dpcm_fe_dai_hw_params failed (-12)
```

slawblauciak, 2021-06-29:
Are dynamic pipelines enabled in those topologies?

XiaoyunWu6666, 2021-06-29:
> Are dynamic pipelines enabled in those topologies?

No , according to sof/tools/topology/topology1/CMakeLists.txt
```
"sof-tgl-max98373-rt5682\;sof-tgl-max98373-rt5682\;-DAMP_SSP=1" [line 142]
```
no DDYNAMIC=1 here @slawblauciak

lgirdwood, 2021-06-29:
@XiaoyunWu6666 can you provide the trace for heap after the topology has loaded. This is a big structure so it's probably taking a larger block.

RanderWang, 2021-06-30:
it is very likely caused by https://github.com/thesofproject/sof/commit/928dba0853efa4cd0d79aa66b1e9c221b8813a8f.

XiaoyunWu6666, 2021-06-30:
> it is very likely caused by [928dba0](https://github.com/thesofproject/sof/commit/928dba0853efa4cd0d79aa66b1e9c221b8813a8f).

thanks @RanderWang for your information .
I revert 928dba0 and got a PASS for this case

lgirdwood, 2021-06-30:
@RanderWang @XiaoyunWu6666 please provide a heap dump - we need to refine all heap pools to align with actual usage. @keyonjie is working on this.

RanderWang, 2021-07-01:
The error message indicated wrong usage of memory zone

src/lib/alloc.c:755  ERROR failed to alloc 0xd8 bytes zone 0x4 caps 0x1 flags 0x0
src/lib/alloc.c:596  ERROR alloc trace: unsupported mem zone
src/lib/alloc.c:768  ERROR failed to alloc 0xd8 bytes zone 0x4 caps 0x1 flags 0x0
src/lib/alloc.c:596  ERROR alloc trace: unsupported mem zone
src/ipc/ipc3/dai.c:253  ERROR dai_config(): No memory for dai_config.
src/ipc/ipc3/dai.c:184  ERROR ipc_comp_dai_config(): comp_dai_config() failed

[       62231.664194] (          16.875000) 1071   heap: 0xbe08f800 size 524288 blocks 1 caps 0x71
[       62248.695443] (          17.031250) 1073    used 56064 free 468224
[       62265.205859] (          16.510416) 1071   heap: 0xbe8003b0 size 64592 blocks 1 caps 0x69
[       62283.487108] (          18.281250) 1073    used 0 free 64592
[       62301.924608] (          18.437500) 1099  heap: runtime status
[       62317.862107] (          15.937499) 1071   heap: 0xbe080000 size 63488 blocks 7 caps 0x45
[       62334.580856] (          16.718750) 1073    used 11264 free 52224
[       62351.716272] (          17.135416) 1081    block 1 base 0xbe082000 size 128 
[       62368.643355] (          16.927082) 1084     count 64 free 32
[       62385.935021] (          17.291666) 1081    block 2 base 0xbe084000 size 256 
[       62404.841270] (          18.906250) 1084     count 128 free 109 
[       62421.924603] (          17.083332) 1081    block 3 base 0xbe08c000 size 512 
[       62439.216269] (          17.291666) 1084     count 8 free 8
[       62454.789185] (          15.572916) 1081    block 4 base 0xbe08d000 size 1024
[       62472.028768] (          17.239582) 1084     count 4 free 4
[       62522.445432] (          50.416664) 1102  heap: runtime shared status
[       62539.424598] (          16.979166) 1071   heap: 0x9e11d000 size 37888 blocks 5 caps 0x45
[       62556.664181] (          17.239582) 1073    used 28416 free 9472

XiaoyunWu6666, 2021-07-14:
dma buffer issue seems to disappear from daily test result .
shall we close it ? and reopen once it happened again

lgirdwood, 2021-07-16:
@XiaoyunWu6666 this and the other memory PRs should be tested and closed as @keyonjie fix has been merged.

closed 2021-07-16 by lgirdwood

plbossart, 2021-07-16:
@lgirdwood we haven't tested TGL_VOLT_SDW for a while due to boot issues... It's unclear how this platform works at the moment.

XiaoyunWu6666, 2021-07-19:
@lgirdwood @plbossart  
below are two fail cases in daily test on TGLU_VOLT_SDW, both about xrun
inner daily 5343?model=TGLU_VOLT_SDW&testcase=multiple-pipeline-capture-50
inner daily 5343?model=TGLU_VOLT_SDW&testcase=multiple-pause-resume-25

It seems to have nothing to do with 'failed to alloc dma buffer'