thesofproject/sof #4427: [BUG] JSL_RVP_NOCODEC check alsabat failed

XiaoyunWu6666, 2021-06-29:
**Describe the bug**
happened since inner daily test 4887  , while 4882 is still good
when check-alsabat-playback ,get warning and the case failed
```
2021-06-26 03:49:37 UTC [REMOTE_INFO] check the PCMs before alsabat test
2021-06-26 03:49:39 UTC [REMOTE_COMMAND] alsabat -Phw:sofnocodec,0 --standalone -n 240000 -r 48000 -c 2 -F 997
2021-06-26 03:49:40 UTC [REMOTE_COMMAND] alsabat -Chw:sofnocodec,0 -c 2 -r 48000 -F 997
WARNING: Signal too weak!
 WARNING: Found low peak 0.73 Hz, very close to DC
 FAIL: Peak freq too low 21.24 Hz
 FAIL: Peak freq too low 133.30 Hz
 FAIL: Peak freq too low 155.27 Hz
 FAIL: Peak freq too low 177.25 Hz
 FAIL: Peak freq too low 210.94 Hz
 FAIL: Peak freq too low 254.88 Hz
 FAIL: Peak freq too low 276.86 Hz
 FAIL: Peak freq too low 298.83 Hz
 FAIL: Peak freq too low 421.14 Hz
alsa-utils version 1.2.2

Entering capture thread (ALSA).
Get period size: 2048  buffer size: 16384
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 79.9; Percentage: [0]
Detected peak at 0.73 Hz of 19.55 dB
 Total 27.4 dB from 0.73 to 15.38 Hz
Detected peak at 21.24 Hz of 9.46 dB
 Total 27.4 dB from 21.24 to 21.24 Hz
Detected peak at 133.30 Hz of 11.90 dB
 Total 27.6 dB from 132.57 to 133.30 Hz
Detected peak at 155.27 Hz of 9.49 dB
 Total 27.7 dB from 155.27 to 155.27 Hz
Detected peak at 177.25 Hz of 8.95 dB
 Total 27.8 dB from 177.25 to 177.25 Hz
Detected peak at 210.94 Hz of 8.86 dB
 Total 27.8 dB from 210.94 to 210.94 Hz
Detected peak at 254.88 Hz of 10.29 dB
 Total 27.9 dB from 254.88 to 254.88 Hz
Detected peak at 276.86 Hz of 9.12 dB
 Total 27.9 dB from 276.86 to 276.86 Hz
Detected peak at 298.83 Hz of 10.10 dB
 Total 28.0 dB from 298.83 to 298.83 Hz
Detected peak at 421.14 Hz of 11.00 dB
 Total 28.1 dB from 421.14 to 421.14 Hz
Detected at least 10 signal(s) in total

Return value is -1003
```
This is a regression , I use the sof recipe of inner daily test 4882 and it got a PASS back

**To Reproduce**
TPLG=sof-jsl-nocodec.tplg ~/sof-test/test-case/check-alsabat.sh -p hw:sofnocodec,0 -c hw:sofnocodec,0 -C 2
**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 5b851f48
SOF Branch: main
SOF Commit: 2d70f100c7ad
 Topology: sof-jsl-nocodec.tplg
 Platform: jsl_rvp_nocodec

plbossart, 2021-06-29:
@XiaoyunWu6666 please double-check that all mixer values are set to 0dB and all switches are on. I tested on Up2-nocodec and alsabat will only work if the mixer settings are correct. We did change the topology so the mixer values are likely incorrect.

XiaoyunWu6666, 2021-06-30:
> @XiaoyunWu6666 please double-check that all mixer values are set to 0dB and all switches are on. I tested on Up2-nocodec and alsabat will only work if the mixer settings are correct. We did change the topology so the mixer values are likely incorrect.

Yes alsabat only works well if the mixer settings are correct . If using the recipe on daily 4482 I can find some setting that fixs this case .However with 4887 , modifying the alsamixer seeting doesn't help to make things better.
If I set all the volume maximum , i got an 
```
Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 3228.7; Percentage: [9]
```
If I set all mixer values 0db
then I got 
```
Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 3228.7; Percentage: [0]
```

Is there any suspicious commit? Or should I do a bisect?

XiaoyunWu6666, 2021-06-30:
replacing only the tplg file works.

XiaoyunWu6666, 2021-06-30:
➜  sof git:(fdd8ec03b) ✗ git bisect good fdd8ec03b
b5b0b2c4f23cd86b245096395e963a9af5f4fa86 is the first bad commit
commit b5b0b2c4f23cd86b245096395e963a9af5f4fa86
Author: Pierre-Louis Bossart <pierre-louis.bossart@linux.intel.com>
Date:   Fri Jun 18 14:44:53 2021 -0500

    topology: use same file for all cavs 1.8+ nocodec topologies

    The beginning of the end of insanity?

    Signed-off-by: Pierre-Louis Bossart <pierre-louis.bossart@linux.intel.com>

 tools/topology/topology1/CMakeLists.txt      |  14 +-
 tools/topology/topology1/sof-cavs-nocodec.m4 | 289 +++++++++++++++++++++++++++
 tools/topology/topology1/sof-cnl-nocodec.m4  | 289 ---------------------------
 tools/topology/topology1/sof-icl-nocodec.m4  | 176 ----------------
 tools/topology/topology1/sof-tgl-nocodec.m4  | 186 -----------------
 5 files changed, 296 insertions(+), 658 deletions(-)
 create mode 100644 tools/topology/topology1/sof-cavs-nocodec.m4
 delete mode 100644 tools/topology/topology1/sof-cnl-nocodec.m4
 delete mode 100644 tools/topology/topology1/sof-icl-nocodec.m4
 delete mode 100644 tools/topology/topology1/sof-tgl-nocodec.m4

lgirdwood, 2021-06-30:
@XiaoyunWu6666 fwiw, I've also seen similar issues with background noise on the DUTs. Have you tried switching off the DUTs and test PC for an hour and retesting ?

plbossart, 2021-06-30:
@lgirdwood in nocodec topologies, we use the loopback mode, there is not dependency on external components/noise.

@XiaoyunWu6666 the same topology works on APL and TGL, and locally on CNL. I don't think you root-caused the issue.

plbossart, 2021-06-30:
I tried directly on the device and the recorded wave file is clearly complete garbage.

[image]

if this is a topology-only change it's probably in the SSP configuration then.

plbossart, 2021-06-30:
The only thing I found in the topology was that going back to s24le seems to work...

@XiaoyunWu6666 can you retry with https://github.com/thesofproject/sof/pull/4438. Make sure that you have all mixers to 0dB and switches on - asound.state attached below

[jsl-nocodec-asound.state.txt](https://github.com/thesofproject/sof/files/6743536/jsl-nocodec-asound.state.txt)

@lgirdwood @keyonjie any idea why the SSP seems to behave differently on ICL/JSL?

XiaoyunWu6666, 2021-07-01:
> @XiaoyunWu6666 can you retry with #4438. Make sure that you have all mixers to 0dB and switches on - asound.state attached below

yes 4438 worked

XiaoyunWu6666, 2021-07-02:
same issue happen on TGLU_RVP_NOCODEC in daily 5028 too

XiaoyunWu6666, 2021-07-05:
> same issue happen on TGLU_RVP_NOCODEC and CML_RVP_NOCODEC in daily 5028 too

Shall I keep them in one issue or open a new one ? I will try whether #4438 can fix CML_RVP_NOCODEC and TGLU_RVP_NOCODEC or not later

plbossart, 2021-07-06:
@XiaoyunWu6666 I think we are still having alsa mixer issues....

plbossart, 2021-07-06:
> @XiaoyunWu6666 I think we are still having alsa mixer issues....

Actually it's more like *ONLY* APL_NOCODEC can work with s32le, I just tried with CML_NOCODEC and using the same fix as on JSL s24le works. no idea why. see https://github.com/thesofproject/sof/pull/4456

plbossart, 2021-07-07:
All alsa-bat tests pass now with PR #4456, let's close this.

closed 2021-07-07 by plbossart