thesofproject/sof #4433: [BUG]Dai failed to alloc dma buffer when running multiple-pipeline on TGLH_0A5E_SDW

XiaoyunWu6666, 2021-06-30:
**Describe the bug**
dai failed to alloc dam buffer when running multiple-pipeline on  TGLH_0A5E_SDW

**To Reproduce**
 TPLG=sof-jsl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20

**Reproduction Rate**
100%

**Environment**

Kernel Branch: topic/sof-dev
Kernel Commit: 6b53609c
SOF Branch: main
SOF Commit: 
6bbf6b0d3cfa

**Screenshots or console output**
[etrace]
```
[  7570440502.979805] (           0.000000) c0 dma-trace             src/trace/dma-trace.c:343  ERROR FW ABI 0x3013000 DBG ABI 0x5003000 tag v1.8-rc1-267-g6bbf6b0d3cfa src hash 0x72fae93b (ldc hash 0x72fae93b)
[     1349267.810968] (     1349267.750000) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[     1349284.790134] (          16.979166) c0 dai          2.10           src/audio/dai.c:543  ERROR dai_params(): failed to alloc dma buffer
[     1349303.227634] (          18.437500) c0 pipe         2.11  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 6
[     1349322.342216] (          19.114582) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 2 comp 6 params failed -12
[     1352857.550409] (        3535.208252) c0 buffer                   src/audio/buffer.c:44   ERROR buffer_alloc(): could not alloc structure
[     1352874.008742] (          16.458332) c0 dai          5.29           src/audio/dai.c:543  ERROR dai_params(): failed to alloc dma buffer
[     1352894.321241] (          20.312500) c0 pipe         5.30  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 23
[     1352913.279573] (          18.958332) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 5 comp 23 params failed -12
```

XiaoyunWu6666, 2021-06-30:
may be duplicated with #4411

keyonjie, 2021-07-06:
We should not see this anymore since commit 928dba0 was been reverted in PR https://github.com/thesofproject/sof/pull/4439.

But anyway, I am working on the PR https://github.com/thesofproject/sof/pull/4445 to make more use of our HP SRAM banks, which will fix this issue even with requiring more runtime_shared buffers.

XiaoyunWu6666, 2021-12-13:
Close this since it haven't happened in CI daily tests 
Fix： PR #4439.

closed 2021-12-13 by XiaoyunWu6666