thesofproject/sof #4434: [BUG]dai_report_xrun:overrun due to no space available on TGLU_VOLT_SDW

XiaoyunWu6666, 2021-06-30:
**Describe the bug**
 when dw_dma_get_data_size.  xrun was detected and then dai report xrun with 'overrun due to no space'

**To Reproduce**
TPLG=sof-tgl-sdw-max98373-rt5682.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25
**Reproduction Rate**
100%
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: dd6665e0
SOF Branch: main
SOF Commit: eb7add5135fd

**Screenshots or console output**
[sof-logger]
```
[    11881985.204936] (    11881985.000000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    11882003.069518] (          17.864582) c0 dai          4.23           src/audio/dai.c:889  ERROR dai_report_xrun(): overrun due to no space available
[    11882020.934101] (          17.864582) c0 dai          4.23           src/audio/dai.c:679  ERROR comp_overrun(): sink->free = 128, copy_bytes = 0
[    11882124.996597] (         104.062492) c0 pipe         4.24  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 23, dir = 0
[    11882144.163263] (          19.166666) c0 pipe         4.24  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[    11902019.110390] (       19874.947266) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    11902036.089556] (          16.979166) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[    11902053.485388] (          17.395832) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 0, copy_bytes = 0
[    11902187.079133] (         133.593750) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[    11902206.402049] (          19.322916) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[    13005986.045689] (     1103779.625000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    13006003.493605] (          17.447916) c0 dai          4.23           src/audio/dai.c:889  ERROR dai_report_xrun(): overrun due to no space available
[    13006020.733187] (          17.239582) c0 dai          4.23           src/audio/dai.c:679  ERROR comp_overrun(): sink->free = 128, copy_bytes = 0
[    13006121.149850] (         100.416664) c0 pipe         4.24  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 23, dir = 0
[    13006139.951932] (          18.802082) c0 pipe         4.24  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[    13094017.500524] (       87877.546875) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    13094034.792190] (          17.291666) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[    13094052.031772] (          17.239582) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 0, copy_bytes = 0
[    13094188.750517] (         136.718750) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[    13094207.917183] (          19.166666) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[    14239980.319571] (     1145772.375000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    14239997.611237] (          17.291666) c0 dai          4.23           src/audio/dai.c:889  ERROR dai_report_xrun(): overrun due to no space available
[    14240014.746653] (          17.135416) c0 dai          4.23           src/audio/dai.c:679  ERROR comp_overrun(): sink->free = 128, copy_bytes = 0
[    14240114.798732] (         100.052078) c0 pipe         4.24  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 23, dir = 0
[    14240133.861231] (          19.062500) c0 pipe         4.24  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[    14326016.670319] (       85882.812500) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[    14326034.118235] (          17.447916) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[    14326051.566151] (          17.447916) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 0, copy_bytes = 0
[    14326181.461979] (         129.895828) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[    14326200.941145] (          19.479166) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! 
```

plbossart, 2021-06-30:
@lgirdwood Daily test 4928 on 6/29 seems to work, Daily test 4961 on 6/30 does not. 

We haven't merged anything at the kernel level that is related to SoundWire or streaming, more precisely we merged @RanderWang's fix that made this problem go away.

Firmware changes?

ranj063, 2021-07-01:
> @lgirdwood Daily test 4928 on 6/29 seems to work, Daily test 4961 on 6/30 does not.
> 
> We haven't merged anything at the kernel level that is related to SoundWire or streaming, more precisely we merged @RanderWang's fix that made this problem go away.
> 
> Firmware changes?

@plbossart this one is not SDW I think. I saw this today with the TGL nocodec tplg. Haven't been able to understand what's going on. And it is also on the same test multiple-pause-resume.

This could be a genuine issue but I won't rule out the test itself being the culprit. I need to go through the script to figure out.

plbossart, 2021-07-08:
@ranj063 I don't think this has anything to do with TGL-nocodec. All the issues I see are between the Speaker and 'Amplifier Reference' devices

````
2021-07-07 22:42:35 UTC [REMOTE_INFO] Speaker to command: aplay -D hw:0,2 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
spawn aplay -D hw:0,2 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
2021-07-07 22:42:35 UTC [REMOTE_INFO] Amplifier Reference to command: arecord -D hw:0,3 -r 48000 -c 2 -f S16_LE -vv -i /dev/null -q
````

And given the topology, I think it's a case of dependencies between connected streams that's not well handled.
[image]

plbossart, 2021-07-08:
And I can prove my theory with two simple commands

in terminal 1, run
````
root@jf-tglu-volt-sdw-1:~# aplay -Dhw:0,2 -c2 -r48000 -fdat /dev/zero -i
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
  aplay: pcm_write:2086: write error: Input/output error     
````

and in terminal two run

````
root@jf-tglu-volt-sdw-1:~# arecord -Dhw:0,3 -c2 -r48000 -fdat test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
arecord: pcm_read:2178: read error: Input/output error
````

After a 10s of monkey testing with the space bar, both streams reached an xrun.

of course when I tried to reproduce the issue I couldn't, so it's a nasty race condition somewhere.

ranj063, 2021-07-08:
@plbossart oh this has to be mux/demux acting up most likely.

plbossart, 2021-07-08:
@ranj063 we could test this theory with a modified topology where BUFF3.0, SMARTAMP3.0 and BUF3.2, i.e. completely independent pipelines. If we still see a problem it would point to a SoundWire enable/disable issue (that would be surprising though since we don't have this problem with other SoundWire topologies, this is the only case where we have a demux and joined pipelines).

plbossart, 2021-07-08:
I think the rules need to be
a) any component with more than one source (and a single sink) can only pause when all its sources are paused. (that applies to mixer, mux and smart-amp)
b) any component with more than one sink (and a single source) can only pause when all its sinks are paused (that applies to demux)

ranj063, 2021-07-09:
> a) any component with more than one source (and a single sink) can only pause when all its sources are paused. (that applies to mixer, mux and smart-amp)

@plbossart is this correct to assume that we pause smart-amp only when all sources are paused? In theory, we should be using the smart-amp in such a way that when the playback stream in paused, the reference capture stream is also paused via UCM perhaps. But in reality, the way we test the smart_amp is by treating the playback and capture streams as separate streams. Im wondering if we should add the UCM changes too so we test it the way we it will be used.

plbossart, 2021-07-09:
@ranj063 I am not sure if we need to special-case the smart-amp component. We've already added logic in the mixer code to deal with one source paused/others active, I think it's only a matter of making sure the smart-amp component follows the same logic.

lgirdwood, 2021-07-16:
@XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?

XiaoyunWu6666, 2021-07-19:
> @XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?

TGLU_RVP_SDW boot_failed  in inner daily 5308 , let me double check it

XiaoyunWu6666, 2021-07-19:
@lgirdwood @plbossart 
inner daily 5343 model=TGLU_VOLT_SDW testcase=multiple-pause-resume-25
inner daily 5343 model=TGLU_VOLT_SDW testcase=multiple-pipeline-capture-50
this issue still happened.
[etrace]
```
[      574920.862571] (      574920.875000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[      574939.143821] (          18.281250) c0 dai          4.23           src/audio/dai.c:889  ERROR dai_report_xrun(): overrun due to no space available
[      574956.747987] (          17.604166) c0 dai          4.23           src/audio/dai.c:679  ERROR comp_overrun(): sink->free = 0, copy_bytes = 0
[      575059.456316] (         102.708328) c0 pipe         4.24  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 23, dir = 0
[      575078.518815] (          19.062500) c0 pipe         4.24  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[      596941.017946] (       21862.500000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[      596958.517946] (          17.500000) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[      596975.653362] (          17.135416) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 0, copy_bytes = 0
[      597104.507523] (         128.854156) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[      597123.205439] (          18.697916) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[     1795019.147422] (     1197896.000000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[     1795036.387005] (          17.239582) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[     1795053.157838] (          16.770832) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 768, copy_bytes = 0
[     1795177.116166] (         123.958328) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[     1795196.074499] (          18.958332) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[     3046945.295592] (     1251749.250000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[     3046963.003925] (          17.708332) c0 dai          3.17           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[     3046980.191424] (          17.187500) c0 dai          3.17           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 17, source->avail = 768, copy_bytes = 0
[     3047105.087252] (         124.895828) c0 pipe         3.18  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 17, dir = 1
[     3047123.785168] (          18.697916) c0 pipe         3.18  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
```

keyonjie, 2021-07-19:
> @XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?

@lgirdwood @XiaoyunWu6666 this is not memory issue, it is xrun issue so my heap refinement will not help.

lgirdwood, 2021-07-19:
> > @XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?
> 
> @lgirdwood @XiaoyunWu6666 this is not memory issue, it is xrun issue so my heap refinement will not help.

@keyonjie ok, can you change the subject then to match the bug.

keyonjie, 2021-07-19:
> > > @XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?
> > 
> > 
> > @lgirdwood @XiaoyunWu6666 this is not memory issue, it is xrun issue so my heap refinement will not help.
> 
> @keyonjie ok, can you change the subject then to match the bug.

sure, done.

XiaoyunWu6666, 2021-07-20:
> > @XiaoyunWu6666 are these memory bugs now still all open after #4445 merged ?
> 
> @lgirdwood @XiaoyunWu6666 this is not memory issue, it is xrun issue so my heap refinement will not help.

Yes , thanks @keyonjie 
As what we discussed before , do you think current daily test is not 'hard' enough to reproduce xrun issue? for example in today's daily test TGLU_RVP_NOCODEC_MULTICORE is with all green . However , this issue is still hiding below.

keqiaozhang, 2022-01-10:
Recently, we added the ADLP Chrome device ADLP_BRYA_SDW to CI, this device has some codec as TGLU_VOLT_SDW and this issue also can be reproduced on ADLP platform, the error logs are almost the same. please refer to inner test ID:9237.

marc-hb, 2022-02-22:
@kfrydryx, @abonislawski  any update? This is one of the few failures still impacting daily tests.

mengdonglin, 2022-02-28:
This issue won't be fixed. When smart amplifier is present, the playback and capture (feedback) pipelines are not independent. The playback pipeline is data provider and feedback pipeline is consumer. We cannot do random pause/release on these 2 pipelines in parallel.

closed 2022-02-28 by mengdonglin