thesofproject/sof #4435: [BUG] Glitches in test 17_01_TestMultiCoreMultiKernelSsp24000Hz24b32b8ch

kkarask, 2021-06-30:
**Describe the bug**
Glitches was found in test 17_01_TestMultiCoreMultiKernelSsp24000Hz24b32b8ch
Playback (core 0) and capture (core 1) pipelines connected to SSP with external loopback.

**Topology**
```
 pipe_plb core 0
                    +--------------------+
    +----------+    | +------+   +-----+ |   +----+
    | Host plb |----->|Buffer|-->| Dai |---->|SSP0|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap core 1           |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +----+
    | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
    +----------+    | +------+   +-----+ |   +----+
                    +--------------------+
```
**To Reproduce**
Run test 17_01_TestMultiCoreMultiKernelSsp24000Hz24b32b8ch with Diagnostic driver

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: bccecb1a8eed84a0737fdf2878204adfb6cf845d
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
[image]
[TGL_B0_ 17_01_TestMultiCoreMultiKernelSsp24000Hz24b32b8ch.zip](https://github.com/thesofproject/sof/files/6739598/TGL_B0_.17_01_TestMultiCoreMultiKernelSsp24000Hz24b32b8ch.zip)

lgirdwood, 2021-06-30:
@kkarask it looks like a lot of these are duplicates in that the test and topology are similar with similar results. Can we label this and other similar glitches as duplicates of the original simple glitch. Thanks

wszypelt, 2021-07-28:
Glitches are visible in test 17_00_TestMultiCoreSameKernelSsp16000Hz32b32b4ch  The topology is the same, but pipe_plb and pipe_cap  is on the same core(core = 1)
ICL RVP platform (near the start of the capture:)
[image]