thesofproject/sof #4437: [BUG] Glitches on ASRC component

kkarask, 2021-06-30:
**Describe the bug**
Glitches was found in test 19_00_TestAsrcPlayback8000Hz32b32b2ch
Playback pipeline contains ASRC. 

**Topology**
```
  +---------------------------------+
    +------+    | +---+  +----+  +---+  +-------+ |
    | Host |----->|Buf|->|Asrc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +----+  +---+  +-------+ | |
                +---------------------------------+ |
                                                    |
                +------------------+                |
    +------+    | +---+  +-------+ |                |
    | Host |<-----|Buf|<-|SSP Dai|<-----------------+
    +------+    | +---+  +-------+ |
                +------------------+
```

**To Reproduce**
Run test 19_00_TestAsrcPlayback8000Hz32b32b2ch with Diagnostic driver

**Reproduction Rate**
Sporadic.

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: eb7add5135fda83c779b936b76e57f54fd293a8d

2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
[image]

[image]

[TGL_19_00_TestAsrcPlayback8000Hz32b32b2ch.zip](https://github.com/thesofproject/sof/files/6740425/TGL_19_00_TestAsrcPlayback8000Hz32b32b2ch.zip)

kkarask, 2022-01-20:
Glitch was found in test 19_00_TestAsrcPlayback8000Hz32b32b2ch on ICL RVP platform too.
Hash:  1165dae2b586bfac8ba2140f52a4e46fa32201a9

[image]

Logs:
[ICL_19_00_TestAsrcPlayback8000Hz32b32b2ch.zip](https://github.com/thesofproject/sof/files/7904601/ICL_19_00_TestAsrcPlayback8000Hz32b32b2ch.zip)

lgirdwood, 2022-01-21:
@singalsu any comments ? Any idea if we have enough MCPS for each config ?

antonirx, 2023-01-05:
unsupported platform
on TGL issue does not occur
Tested FW hash: 4943aaa

closed 2023-01-05 by wszypelt