thesofproject/sof #4458: [BUG]DSP panic when simultaneous playback -capture on ADLP_RVP_NOCODEC

XiaoyunWu6666, 2021-07-07:
**Describe the bug**
DSP panic when simultaneous playback capture on ADLP_RVP_NOCODEC in daily-ADL 5117

**To Reproduce**
TPLG=sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/simultaneous-playback-capture.sh -l 50

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 6bd3089a
SOF Branch: main
SOF Commit: a7604433c2a2

**Screenshots or console output**

```shell
[ 5204.620589] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 5204.620943] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 5204.627920] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[ 5204.627925] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 5204.627931] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5204.628026] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[ 5204.628028] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[ 5204.628030] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[ 5204.628033] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[ 5204.628035] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
```
Let me open another new issue for ADLP_RVP_NOCODEC

XiaoyunWu6666, 2021-07-07:
does it have something to do with #4328

mengdonglin, 2021-07-13:
This issue isn't reproduced in daily test report 5215. I guess it was fixed by PR #4462.

XiaoyunWu6666, 2021-08-03:
> This issue isn't reproduced in daily test report 5215 and 5207. I guess it was fixed by PR #4462. @XiaoyunWu6666 Can you help double check and see if we can close this issue?

haven't happened in daily tests for quite long

lgirdwood, 2021-08-03:
Lets monitor and close if not reproducing over next few days.

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang