thesofproject/sof #4459: [BUG]PCM_PARAMS timed out on TGLH_RVP_NOCODEC_ZEPHYR and APL_UP2_NOCODEC_ZEPHYR

XiaoyunWu6666, 2021-07-07:
**Describe the bug**
happen in inner daily 5118
PCM_PARAMS timed out on  TGLH_RVP_NOCODEC_ZEPHYR and APL_UP2_NOCODEC_ZEPHYR
when aplaying with -Dhw:0,0

**To Reproduce**
TPLG=sof-apl-nocodec.tplg ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F
OR
aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 19 /dev/zero -v -q

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 6bd3089a 
SOF Branch: main
SOF Commit: a7604433c2a2
Topology :sof-tgl-h-nocodec.tplg
Platform: TGLH_RVP_NOCODEC_ZEPHYR

**Screenshots or console output**
[console]
```
ubuntu@sh-apl-up2-nocodec-zephyr-01:~$ aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 19 /dev/zero -v -q
aplay: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE

```
[dmesg]
```shell
[24139.438375] sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[24139.941561] sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x60010000 size 108
[24139.941595] sof-audio-pci-intel-apl 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[24139.941608] sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[24139.941663] sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[24139.941680] sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[24139.941693] sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[24139.941707] sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[24139.941720] sof-audio-pci-intel-apl 0000:00:0e.0: error: waking up any trace sleepers
[24139.941734] sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 1
[24139.941747] sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -110
[24139.941771]  Port0: ASoC: soc_pcm_hw_params() failed (-110)
[24139.941795]  Port0: ASoC: dpcm_fe_dai_hw_params failed (-110)

```

marc-hb, 2021-07-10:
It bisected this, see #4472

In the Zephyr logs this shows as

```
[00022612] <err> os:  ** FATAL EXCEPTION
[00022612] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF d{slot 33, seq=34} ata error)
[00022612] <err> os:  **  PC 0xbe00b7ed VADDR 0x9e08bbc4
[00022612] <err> os:  **  PS 0x60a20
[00022612] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 {slot 37, seq=38} WOE:1 OWB:10 CALLINC:2)
[00022612] <err> os:  **  A0 0xbe00ba7e  SP 0xbe0498e0  A2 0{slot 39, seq=40} x1fff  A3 0x9e08bbc4
[00022612] <err> os:  **  A4 0x4  A5 0xc1  A6 0x1b20  A7 0xb{slot 41, seq=42} e049b64
[00022612] <err> os:  **  A8 0x7fff  A9 0x7fff A10 0xe A11 0{slot 43, seq=44} x9e0732c2
[00022612] <err> os:  ** A12 0x2 A13 0x182 A14 0x180 A15 0x6{slot 45, seq=46} 0
[00022612] <err> os:  ** LBEG 0xbe01739c LEND 0xbe0173a6 LCO{slot 47, seq=48} UNT (nil)
[00022612] <err> os:  ** SAR 0x19
[00022612] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception{slot 50, seq=51}  on CPU 0
[00022612] <err> os: Current thread: 0x9e075ca8 (unknown)
[00022614] <err> os: Halting system
```

lyakh, 2021-07-14:
@XiaoyunWu6666 @mengdonglin good to close this? A fix in in upstream #4479

XiaoyunWu6666, 2021-07-15:
> @XiaoyunWu6666 @mengdonglin good to close this? A fix in in upstream #4479

good to me .
As daily test shows , ipc timed out on TGLH_RVP_NOCODEC_ZEPHYR and APL_UP2_NOCODEC_ZEPHYR when multiple-pipeline/ pause-release / suspend-resume , do not happen when simply aplay .

marc-hb, 2021-07-15:
Thanks @XiaoyunWu6666 , closing

closed 2021-07-15 by marc-hb