thesofproject/sof #4465: [BUG][GLK]Xrun and IO error when multiple-pause-resume

XiaoyunWu6666, 2021-07-08:
**Describe the bug**
happened in inner daily 5153
model=GLK_BOB_DA7219
testcase=multiple-pause-resume-25
xrun is observed and  io error happened

**To Reproduce**
 TPLG=sof-glk-da7219.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 143fbb70 storage: 143fbb70
SOF Branch: main
SOF Commit: 690481ae6990 storage: 690481ae6990

**Screenshots or console output**
[console]
```
 arecord: pcm_read:2153: read error: Input/output error
```
[etrace]
```
[   159576109.909015] (    71144656.000000) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   159848116.096123] (      272006.187500) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   160131119.157794] (      283003.062500) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   160424116.073235] (      292996.906250) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   167759074.167187] (     7334958.000000) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   168053118.582586] (      294044.406250) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   168288118.833664] (      235000.250000) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   180255222.472718] (    11967104.000000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   180878751.770858] (      623529.312500) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33864 mask 63 val 0 us 937
[   189231679.563942] (     8352928.000000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   197586563.034032] (     8354883.500000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   205924563.744376] (     8338000.500000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   212239109.378876] (     6314545.500000) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   212240105.993419] (         996.614563) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[   214146636.855161] (     1906530.875000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   222554162.250242] (     8407525.000000) c0 wait                         src/lib/wait.c:46   ERROR poll timeout reg 33352 mask 63 val 0 us 937
[   223653038.091993] (     1098875.875000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[   223653044.706576] (           6.614583) c0 dai          3.19           src/audio/dai.c:889  ERROR dai_report_xrun(): overrun due to no space available
[   223653049.966993] (           5.260417) c0 dai          3.19           src/audio/dai.c:679  ERROR comp_overrun(): sink->free = 768, copy_bytes = 0
[   223653091.529491] (          41.562500) c0 pipe         3.20  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 19, dir = 0
[   223653097.258658] (           5.729167) c0 pipe         3.20  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[   223848091.990493] (      194994.734375) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[   223848098.865492] (           6.875000) c0 dai          2.12           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[   223848104.125909] (           5.260417) c0 dai          2.12           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 12, source->avail = 220, copy_bytes = 0
[   223848150.115490] (          45.989582) c0 pipe         2.13  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 12, dir = 1
[   223848155.844657] (           5.729167) c0 pipe         2.13  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[   224455091.966373] (      606936.125000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[   224455098.685122] (           6.718750) c0 dai          2.12           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[   224455103.789289] (           5.104167) c0 dai          2.12           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 12, source->avail = 412, copy_bytes = 0
[   224455142.643454] (          38.854164) c0 pipe         2.13  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 12, dir = 1
[   224455148.424704] (           5.781250) c0 pipe         2.13  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[   225062091.161003] (      606942.750000) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[   225062098.036002] (           6.875000) c0 dai          2.12           src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[   225062103.192252] (           5.156250) c0 dai          2.12           src/audio/dai.c:661  ERROR comp_underrun(): dev->comp.id = 12, source->avail = 604, copy_bytes = 0
[   225062141.629751] (          38.437500) c0 pipe         2.13  ....../pipeline-stream.c:168  ERROR pipeline_copy(): ret = -61, start->comp.id = 12, dir = 1
[   225062147.254750] (           5.625000) c0 pipe         2.13  ..../pipeline-schedule.c:63   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
```

plbossart, 2021-07-08:
Probably the same issue as with NOCODEC topologies, where we also use the early bclk enablement.

PR https://github.com/thesofproject/sof/pull/4462 should hopefully fix this.

mengdonglin, 2021-07-13:
@XiaoyunWu6666 Can we still reproduce this issue? PR #4462 has been merged.

XiaoyunWu6666, 2021-07-13:
> @XiaoyunWu6666 Can we close this issue? PR #4462 has been merged. We haven't observed this issue in last 4 daily reports 5206-5244.

yes ,thanks

closed 2021-07-13 by XiaoyunWu6666