thesofproject/sof #4470: [BUG] 24_01_TestBluetoothParallelOffloadSsp Exception SofIpcErrorResponseException occurred

wszypelt, 2021-07-09:
**Describe the bug**
Bug was found in:
24_01_TestBluetoothParallelOffloadSsp16b16b2ch
24_01_TestBluetoothParallelOffloadSsp24b32b4ch
24_01_TestBluetoothParallelOffloadSsp32b32b8ch

the content of the error:
ERROR:Exception SofIpcErrorResponseException occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':730 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

Last working build: SOF_FW_MASTER_01.01.00.2464 

**Topology**
```
    3x
                        pipe_plb
                        +--------------------+
        +----------+    | +------+   +-----+ |   +----+
        | Host plb |----->|Buffer|-->| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+    |  |
                                                  |  |
                        pipe_cap                  |  |
                        +--------------------+    |  |
        +----------+    | +------+   +-----+ |   +----+
        | Host cap |<-----|Buffer|<--| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+

```
Description:
Three set of pipelines, each configured for different sampling rate, all connected to the same SSP, consecutive
interface configuration and playback/capture is performed on those pipelines.

**To Reproduce**
Run tests with diagnostic driver
24_01_TestBluetoothParallelOffloadSsp16b16b2ch
24_01_TestBluetoothParallelOffloadSsp24b32b4ch
24_01_TestBluetoothParallelOffloadSsp32b32b8ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: be3dff2780e10862ec0693879625e70390046a36
2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP, WHL RVP, TGL RVP, APL RVP
3) Test environment:
BRANCH: master HASH: e8fa676f694a584ea7c0cce74e6ed24a7801d749

**Screenshots or console output**
```
[     4636406.901042] (        4242.552246) c0 ipc                  src/ipc/ipc3/handler.c:536  ipc: dai 1.0 -> config 
[     4636416.588542] (           9.687500) c0 dai                           src/lib/dai.c:164  dai_get type 1 index 0 new sref 7
[     4636426.067708] (           9.479167) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:274  ssp_set_config(): Already configured. Ignore config
[     4636434.947917] (           8.880208) c0 dai                           src/lib/dai.c:188  dai_put type 1 index 0 new sref 6
[     4636443.880208] (           8.932292) c0 ipc                      src/ipc/ipc3/dai.c:159  ipc_comp_dai_config() dai type = 1 index = 0
[     4636454.348958] (          10.468750) c0 dai          1.1         src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 0 dd 0x9e122000
[     4636469.244792] (          14.895833) c0 dai          2.5         src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 0 dd 0x9e122100
[     4636484.114583] (          14.869792) c0 dai          3.9         src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 0 dd 0x9e122200
[     4636497.239583] (          13.125000) c0 dai          4.13        src/ipc/ipc3/dai.c:223  dai_config() dai type = 1 index = 0 dd 0x9e122300
[     4645076.484375] (        8579.245117) c0 dma-trace                 src/trace/trace.c:135  Suppressed 2 similar messages: dai_config() dai type = %d index = %d dd %p
[     4645084.531250] (           8.046875) c0 pipe         3.11  ....../pipeline-params.c:215  pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 16000
[     4645093.177083] (           8.645833) c0 pipe         3.11  ....../pipeline-params.c:219  pipe params stream_tag 1 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[     4645136.223958] (          43.046875) c0 dai          3.9         src/ipc/ipc3/dai.c:84   dai_data_config() dai type = 1 index = 0 dd 0x9e122200
[     4645146.276042] (          10.052083) c0 dai          3.9            src/audio/dai.c:297  ERROR dai_verify_params(): pcm rate parameter 16000 does not match hardware rate 8000
[     4645155.130208] (           8.854167) c0 dai          3.9            src/audio/dai.c:459  ERROR dai_params(): pcm params verification failed.
[     4645163.906250] (           8.776042) c0 pipe         3.11  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -22, host->comp.id = 8
[     4645172.786458] (           8.880208) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 3 comp 8 params failed -22
[     4645182.447917] (           9.661458) c0 pipe         3.11  ......./pipeline-graph.c:341  pipe reset
[     4645219.010417] (          36.562500) c0 dai          3.9            src/audio/dai.c:687  dai_reset()
Skipped 0 bytes after the last statement. Wrap possible, check the start of the output for later logs.
```
**Logs**
[2021_07_09_00_26_07.zip](https://github.com/thesofproject/sof/files/6791252/2021_07_09_00_26_07.zip)

slawblauciak, 2021-07-13:
From what I understood this is a regression.
@wszypelt do we know when it has last worked before?

wszypelt, 2021-11-22:
Issue still occure

wszypelt, 2022-05-06:
It looks like the problem is no longer present, I'm closing

closed 2022-05-06 by wszypelt