thesofproject/sof #4480: [BUG] TGLU_RVP_NOCODEC DSP panic while suspend-resume test

fredoh9, 2021-07-12:
**Describe the bug**
DSP panic logs seems different from the others DSP panic, and after seeing DSP panic, following FW status and IPC are working fine.

```
[12428.728341] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[12428.731876] kernel: serial 00:03: activated
[12428.737923] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
[12428.738009] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[12428.738012] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[12428.738030] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[12428.738041] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x25101c2 0x0
[12428.959487] kernel: atkbd serio0: Failed to deactivate keyboard on isa0060/serio0
[12428.995064] kernel: nvme nvme0: 8/0/0 default/read/poll queues
[12429.030091] kernel: ata1: SATA link down (SStatus 4 SControl 300)
[12429.038399] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5 timedout
[12429.038419] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
```

Found in internal weekly stress test, test result 5213.
model=TGLU_RVP_NOCODEC&testcase=check-suspend-resume-100

**To Reproduce**
TPLG=sof-tgl-nocodec.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 100

**Reproduction Rate**
not sure but not 100% need high iteration.
In this test, it is found in 71/100 loop.

**Expected behavior**
suspend/resume test return success

**Environment**

- Kernel Branch: topic/sof-dev
- Kernel Commit: 62368fe1
- SOF Branch: main
- SOF Commit: 5374da8c8917
- tplg: sof-tplg-nocodec.tplg

Attached full dmesg from the test.
[dmesg_tgl_nocodec_dsp_panic.zip](https://github.com/thesofproject/sof/files/6803816/dmesg_tgl_nocodec_dsp_panic.zip)

fredoh9, 2021-07-12:
check-suspend-resume-with-capture-25 in same weekly test found same DSP panic.

plbossart, 2021-07-12:
@fredoh9 are you saying that the same commits pass or fail depending on how long the tests are?

plbossart, 2021-07-12:
Answering to my own question

PASS: https://sof-ci.sh.intel.com/#/result/planresultdetail/5216?model=TGLU_RVP_NOCODEC&testcase=check-suspend-resume-50

TPLG=sof-tgl-nocodec.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 50

Start Time: 2021-07-11 21:20:54 UTC
End Time: 2021-07-11 23:45:30 UTC

Kernel Branch: topic/sof-dev
Kernel Commit: 
62368fe1
SOF Branch: main
SOF Commit: 
5374da8c8917

FAIL: https://sof-ci.sh.intel.com/#/result/planresultdetail/5213?model=TGLU_RVP_NOCODEC&testcase=check-suspend-resume-100

TPLG=sof-tgl-nocodec.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 100

Start Time: 2021-07-10 18:23:05 UTC
End Time: 2021-07-11 00:18:36 UTC

Kernel Branch: topic/sof-dev
Kernel Commit: 
62368fe1
SOF Branch: main
SOF Commit: 
5374da8c8917

Same exact commits, same platform, different duration. Gah.

ranj063, 2021-07-12:
@plbossart @fredoh9 I think this issue is related to https://github.com/thesofproject/sof/issues/4419

There's a temporary workaround for it in https://github.com/thesofproject/sof/pull/4466

plbossart, 2021-07-12:
@ranj063 we are not changing the firmware as described in #4419. What makes you think it might be related?

ranj063, 2021-07-12:
> @ranj063 we are not changing the firmware as described in #4419. What makes you think it might be related?

The other issue can be reproduced even without changing the FW @plbossart. @slawblauciak confirmed it in one of the comments

slawblauciak, 2021-07-13:
As mentioned before, can you try this PR and see if it helps? https://github.com/thesofproject/sof/pull/4466

ranj063, 2021-07-13:
@fredoh9 would it be possible to check if #4466 helps with this issue?

keqiaozhang, 2021-09-15:
Same issue observed on ADLP-RVP-NOCODEC:
https://sof-ci.sh.intel.com/#/result/planresultdetail/6549?model=ADLP_RVP_NOCODEC&testcase=check-suspend-resume-with-capture-100

ranj063, 2021-09-15:
@keqiaozhang are you testing with the same commit in SOF as the original issue? or are you testing with latest main?

lgirdwood, 2021-09-20:
@keqiaozhang are you able to do a FW stack trace when it next occurs. @keyonjie knows how to do this if you have any questions.

XiaoyunWu6666, 2021-09-29:
The DSP panic reproduced again on TGLU_RVP_NOCODEC in inner daily 6914 (sorry I forgot to do a FW stack trace dump , but tried to reproduce it manually ... So the dmesg just went away)

Add 'options snd_sof sof_debug=0x41'  in /etc/modprobe.d on the DUT
Let's see what this debug option will bring to us next time the DSP panic happened in daily test
Let's see