thesofproject/sof #4481: [BUG]IO error and DSP panic when check-capture-50round on ADLP_RVP_NOCODEC/TGLU_RVP_NOCODEC_MULTICORE

XiaoyunWu6666, 2021-07-13:
**Describe the bug**
when check capture continuously on ADLP_RVP_NOCODEC/TGLU_RVP_NOCODEC_MULTICORE , IO error happened with dsp panic.
inner daily  5244  model=TGLU_RVP_NOCODEC_MULTICORE&testcase  check-capture-50rounds

**To Reproduce**
TPLG=sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-capture.sh -d 1 -l 1 -r 50

**Reproduction Rate**
not sure but can increase the number after '-r ' to increase the reproduction rate

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: e12a743b
SOF Branch: main
SOF Commit: 
5374da8c8917
 Topology: sof-tgl-nocodec-ci.tplg
PlatformADLP_RVP_NOCODEC/TGLU_RVP_NOCODEC_MULTICORE

**Screenshots or console output**
[console]
```
2021-07-12 22:01:37 UTC N ===== Testing: (Round: 2/50) (PCM: smart-nocodec [hw:0,2]) (Loop: 1/1) =====
2021-07-12 22:01:37 UTC N no file prefix, use /dev/null as dummy capture output
2021-07-12 22:01:37 UTC N arecord   -Dhw:0,2 -r 48000 -c 4 -f S16_LE -d 1 /dev/null -v -q
Hardware PCM card 0 'sof-nocodec' device 2 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 4
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 8192
  period_size  : 2048
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2048
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8192
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)
arecord: pcm_read:2153: read error: Input/output error
```
[dmesg]
```shell
[ 2041.367066] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 2
[ 2041.367085] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 2041.367615] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 2041.367625] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790908
[ 2041.367677] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2041.368110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2041.368342] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 2 dir 1 cmd 1
[ 2041.368872] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x2024001e successful
[ 2041.368880] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 2041.369361] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 2041.377085] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error : DSP panic!
[ 2041.377101] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 2041.377109] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 2041.377224] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: runtime exception
[ 2041.377229] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace point 00004000
[ 2041.377234] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: panic at :0
[ 2041.377239] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[ 2041.377243] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[ 2041.377248] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000001c EXCVADDR 0xc0000004 PS       0x00060b25 SAR     0x00000000
[ 2041.377253] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe02de97 EPC2     0xbe02d40a EPC3     0x00000000 EPC4    0x00000000
[ 2041.377258] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 2041.377263] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 2041.377268] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 2041.377272] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0xbe130cb0
[ 2041.377278] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cb0: 00000000 00000000 00000000 00000000
[ 2041.377284] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cb4: 7e050e45 be130d30 00078600 00000001
[ 2041.377289] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cb8: 5bc48d00 79c91bbd 000c0800 00000000
[ 2041.377294] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cbc: 0dead000 00000000 0322c028 ffff9d08
[ 2041.377298] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cc0: c083c04c ffffffff 52599ec0 ffff9d08
[ 2041.377303] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cc4: 936cfc92 ffffffff 0ec04b00 ffff9d08
[ 2041.377307] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130cc8: 0ec04b00 ffff9d08 05e897c0 ffff9d08
[ 2041.377312] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xbe130ccc: 00000000 00000000 05e897c0 ffff9d08
[ 2041.377316] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: waking up any trace sleepers
[ 2041.377383] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 2041.625785] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 2041.625807] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 2041.872569] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 2 dir 1 cmd 0
[ 2041.872590] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 2041.882804] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 2041.882825] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 2041.882838] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -62
[ 2041.882847] kernel:  smart-nocodec: ASoC: trigger FE cmd: 0 failed: -62
[ 2041.882896] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 2 dir 1
[ 2041.882906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 2041.893104] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60030000 size 12
[ 2041.893142] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -62
[ 2041.893276] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2041.903542] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[ 2041.903558] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed resetting DAI config for SSP2.IN

```

[sof-logger]
```
[   400001709.261610] (     1626868.875000) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
[   400022395.614955] (       20686.353516) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
[   400043273.634959] (       20878.019531) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
[   402433589.477476] (     2390315.750000) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
[   402454184.528741] (       20595.050781) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1
[   402474853.746670] (       20669.218750) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 1

Found valid LDC address after skipping 20 bytes (one line uses 20 + 0 to 16 bytes)

		 --- negative DELTA = -3510499.392 us: wrap, IPC_TRACE, other? ---

[  6285857680.066207] (           0.000000) c2 idc                ......./intel/cavs/idc.c:58   INFO idc_irq_handler(), IPC_IDCTFC_BUSY
[          21.979166] (          21.979166) c2 idc                ......./intel/cavs/idc.c:165  INFO idc_do_cmd()
[          47.083331] (          25.104166) c2 ipc                  src/ipc/ipc3/handler.c:536  INFO ipc: dai 1.2 -> config 

Found valid LDC address after skipping 12 bytes (one line uses 20 + 0 to 16 bytes)
```

XiaoyunWu6666, 2021-08-03:
did not see this in CI daily tests , was there any possible solutions that has solved it?

Shall we close it or just lower the priority

lgirdwood, 2021-08-03:
Lets monitor and close if not reproducing over next few days.

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang