thesofproject/sof #4487: [FEATURE] Update and remove passthrough definitions

cujomalainey, 2021-07-13:
some components have outdated manual copy code, we need to clean these up with the new `audio_stream_copy` function.

Current spotted ones are:
* [Crossover](https://github.com/thesofproject/sof/blob/72121fb06c0f059cf0977d02596adb9cc0e6b12c/src/audio/crossover/crossover_generic.c#L101) This works because the definition of [`audio_stream_copy`](https://github.com/thesofproject/sof/blob/72121fb06c0f059cf0977d02596adb9cc0e6b12c/src/include/sof/audio/audio_stream.h#L555) does not include updating the buffer r/w
* [IGO NR module](https://github.com/thesofproject/sof/blob/0c21218b379d033d82616a651a7e702583762709/src/audio/igo_nr/igo_nr.c#L85) not entirely necessary, just would allow a bypass of the whole interleave/deinterleave logic

I will need to see if there are any others but a quick search for "passthrough" yielded the above results.

lgirdwood, 2023-05-10:
@andrula-song @singalsu I assume all modules now using optimal copy/processing API/loops now. Can we close ?

singalsu, 2023-05-15:
@lgirdwood The crossover and igo_nr are still using the read/write frag function in a MCPS  expensive way. We will convert both to module adapter in this Q2 that will address this issue too.

alex-cri, 2023-05-30:
Moving to v2.7 for the remaining items to be addressed.

lgirdwood, 2023-07-04:
@singalsu @andrula-song any update, I guess most modules now optimized, can we close ?

andrula-song, 2023-07-05:
checked the code, igo_nr still not converted to module_adapter interface. I am afraid we can not close it now.

kv2019i, 2023-11-29:
Thanks @andrula-song . For IGO_NR we have a draft for module conversion, but that won't make it to v2.8, so let me that that PR for v2.9 and push feature as well. Draft at https://github.com/thesofproject/sof/pull/8301

kv2019i, 2024-03-04:
@andrula-song @singalsu IGO_NR is now converted to module interface, #8301 is merged as well. Can we close?

singalsu, 2024-03-13:
@kv2019i  IGO_NR is now OK, but crossover isn't. The module conversion is done but there was no need to touch the processing core functions. It would be a separate work item from us.

lgirdwood, 2024-04-17:
@singalsu whats remaining effort for crossover ?

kv2019i, 2024-06-05:
v2.10 cut @singalsu @lgirdwood , crossover still open, moving to 2.11.

kv2019i, 2024-12-13:
Feature cutoff for v2.12 today, what's the status with cross-over?

johnylin76, 2024-12-16:
sorry for the late reply, uploaded #9726 for review

closed 2025-01-03 by kv2019i