thesofproject/sof #4488: [BUG] Reset Handler Cleanup

cujomalainey, 2021-07-14:
Follow up from [here](https://github.com/thesofproject/sof/pull/3789/files#diff-c3b15aa3cf6ee70bf6aab304248b7e223c4876a91b558b9ad8cde290a6604c52R326)

Now that we have well defined behaviour for the reset handler we need to followup on cleaning up state in reset. Initial scan shows some components still in violation

* Codec adapter has number of variables set in `prepare` and `params` step that are not zeroed out - @mrajwa 
* Crossover doesn't reset stream data or processing function pointer - @afq984 
* Dcblock similar to above for crossover  - @afq984 
* DRC `cd->source_format` is not cleaned up - @johnylin76 
* EQ FIR `source_format` isn't cleaned up but doesn't appear to be used? - @singalsu 
* EQ IIR same var again, could be a local var? Sink format as well - @singalsu 
* IGO fails to clean up all `cd` data stored in `params` step - @fuyuntsuo 
* Multiband DRC same as DRC  - @johnylin76 
* Mux processing funcs not cleared - @akloniex 
* Selector no state cleaned up - @keyonjie 
* Smart Amp no state cleaned up - @ryans-lee 
* SRC same as EQ IIR - @singalsu 
* TDFB in/out buffer clear only appears to happen when new config appears on prepare, not on pipeline reset? - @singalsu 
* Volume does not clear out old state set in prepare for processing - @singalsu 
* Host appears to store a number of state variables in hd->config that are never cleared and many not used - @lgirdwood 
* KPB number of variables set on prepare that are not cleared in reset  - @mrajwa 
* mixer fails to clear mixer function state also has a magic number to stop pipeline walk? - @lgirdwood 

Sorry for the large blast radius for messages  just hoping to get some coherency for all components so we can stop propagating errors to new members joining the project through copy paste issues :)

singalsu, 2021-07-19:
Thanks @cujomalainey for nice summary! I'm targeting make a fix PR before my vacation next week for components assigned to me.

lgirdwood, 2021-07-23:
@cujomalainey @singalsu I've added check boxes to allow us to follow whats done. @singalsu please update.

johnylin76, 2021-07-27:
Updated check boxes for drc and multiband_drc since #4546 is merged.

cujomalainey, 2022-04-12:
Need to check RTNR to see if it needs fixing

lgirdwood, 2022-04-20:
@kv2019i some mixer, @andrula-song some processing components here.

cujomalainey, 2022-04-22:
Also AEC needs this

lgirdwood, 2022-04-22:
@eddy1021 @ranj063 @singalsu fyi - updates on the list.

lgirdwood, 2022-04-22:
@jsarha fyi - one for you too.

jsarha, 2022-04-22:
> @jsarha fyi - one for you too.

Gathering context ATM, to get on track what is needed and where...

cujomalainey, 2022-04-25:
@eddy1021 leave the codec / module adapter to @ranj063 as she is doing a lot of work there right now.

jsarha, 2022-05-04:
> Host appears to store a number of state variables in hd->config that are never cleared and many not used - @jsarha 

Looks to me that somebody beat me to it:
https://github.com/thesofproject/sof/commit/1c136b1d2f6a7f55bd69c6580ffc979a9880f447

@lgirdwood was there something more to this, or should I just tap my ✅ ?

eddy1021, 2022-05-09:
Mark `dcblock`, `selector`, `smart amp`, and `kpb` as done.
`crossover` is blocked by pulls #5729

lgirdwood, 2022-05-13:
> Mark `dcblock`, `selector`, `smart amp`, and `kpb` as done. `crossover` is blocked by pulls #5729

#5729 merged.
@ranj063 any comment on the codec adapter part (in the BUG description) ?

closed 2022-08-23 by cujomalainey