thesofproject/sof #4491: [BUG] SOF linker error, .bss cannot fit into sof_fw section

aiChaoSONG, 2021-07-15:
The APL FW cannot be built with XCC because of a linker error, the issue is also reproduced on CI.
```
/home/chao/xtensa/XtDevTools/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-ld: sof section `.bss' will not fit in region `sof_fw'
make[3]: *** [CMakeFiles/sof.dir/build.make:1672: sof] Error 2
make[2]: *** [CMakeFiles/Makefile2:1415: CMakeFiles/sof.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2240: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
make: *** [Makefile:736: bin] Error 2

```

aiChaoSONG, 2021-07-15:
revert 38f18bd082747c7fc0859987c9a034c663118266 fix the issue

slawblauciak, 2021-07-15:
P1 as it blocks SOF CI

kkarask, 2021-07-15:
It affects CI SOF regression tests. Builds are failing due to that issue.

lgirdwood, 2021-07-15:
@aiChaoSONG PR revert merged. Lets re-test.
@keyonjie fyi, will need a fix on APL.

slawblauciak, 2021-07-20:
@aiChaoSONG good to close now?

aiChaoSONG, 2021-07-20:
keyon provided new fix, 424beb0f8ff9548efeef8f2bc81baf70e0415e40, close.

closed 2021-07-20 by aiChaoSONG