thesofproject/sof #4492: [BUG] IPC timed out when multiple-pause-resume on APL_UP2_NOCODEC_ZEPHYR

XiaoyunWu6666, 2021-07-15:
**Describe the bug**
ipc timed out when multiple-pipeline on APL_UP2_NOCODEC_ZEPHYR

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25

**Environment**
Start Time: 2021-07-15 02:10:50 UTC
End Time: 2021-07-15 04:47:20 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: 9b277b80
SOF Branch: main
SOF Commit: 0fa9f7d89202

**Screenshots or console output**
[console]
```
=== PAUSE ===                                                            
(10/25) pcm'DMIC16kHz' cmd'arecord' id'1': Wait for 179 ms before pause
                                                                           
 
(10/25) pcm'DMIC16kHz' cmd'arecord' id'1': Wait for 182 ms before resume
arecord: pcm_read:2153: read error: Input/output error
                                                                           

########  +                                        | 19% 
#         +                                        | 19%
=== PAUSE ===                                                 
```

no dmesg log

lgirdwood, 2021-07-16:
@XiaoyunWu6666 @mengdonglin it looks like these Zephyr related issues from yesterday are all probably related.

kv2019i, 2021-08-24:
Potential bugfix: https://github.com/thesofproject/sof/pull/4673

XiaoyunWu6666, 2021-08-30:
> @XiaoyunWu6666 thanks, lets give it a few more days then close. xruns expected atm due to use of uncache memory.

hi @lgirdwood , in inner daily 6175, ipc timed out happened when multiple-pause-resume again.
```
[ 3602.715232] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 3602.815078] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: pcm: trigger stream 1 dir 0 cmd 3
[ 3603.217943] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx timed out for 0x60060000 (msg/reply size: 12/12)
[ 3603.218132] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: preventing DSP entering D3 state to preserve context
[ 3603.218140] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[ 3603.218155] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
[ 3603.218168] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3603.218181] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 3603.218193] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[ 3603.218202] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[ 3603.218214] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[ 3603.218268] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[ 3603.218279] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
[ 3603.218321] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: FW Poll Status: reg[0xa0]=0x20240000 successful
[ 3603.218334] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at soc_component_trigger on 0000:00:0e.0: -110
[ 3603.218350] kernel:  DMIC: ASoC: trigger FE cmd: 3 failed: -110
```

XiaoyunWu6666, 2021-09-07:
in inner daily 6362
IPC timed out when multiple-pause-resume on APL_UP2_NOCODEC_ZEPHYR still happened , when TRIG_RELEASE

closed 2021-11-01 by XiaoyunWu6666

XiaoyunWu6666, 2021-11-01:
close #4742 and #4492 , haven't been seen in daily tests for quite long