thesofproject/sof #4506: [BUG]APL_UP2_NOCODEC_ZEPHYR dsp not suspended after 15s when check-kmod-load-unload

XiaoyunWu6666, 2021-07-20:
**Describe the bug**
APL_UP2_NOCODEC_ZEPHYR
dsp not suspended after 15s when check-kmod-load-unload 

**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/check-kmod-load-unload.sh -l 25

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 8877846e storage: 8877846e
SOF Branch: main
SOF Commit: f96e25ffcc8d storage: f96e25ffcc8d
* Topology:sof-apl-nocodec.tplg
Platform: APL_UP2_NOCODEC_ZEPHYR 

**Screenshots or console output**
[console]
```shell
2021-07-19 22:55:39 UTC [REMOTE_INFO] ===== Starting iteration 1 of 25 =====
2021-07-19 22:55:39 UTC [REMOTE_INFO] wait dsp power status to become suspended
2021-07-19 22:56:04 UTC [REMOTE_ERROR] dsp is not suspended after 15s, end test
2021-07-19 22:56:04 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2021-07-19 22:56:04 UTC [REMOTE_ERROR]  die()  @  /home/ubuntu/sof-test/test-case/../case-lib/lib.sh
2021-07-19 22:56:04 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-kmod-load-unload.sh:69
2021-07-19 22:56:05 UTC [REMOTE_INFO] Test Result: FAIL
```
no dmesg info or sof-logger info available

lgirdwood, 2021-07-21:
Thanks @XiaoyunWu6666 , this issue is a work item on @kv2019i list. We are missing some parts of the IP shutdown sequence as they are being transferred into Zephyr SOC support. @dcpleung fyi.

kv2019i, 2021-08-24:
@XiaoyunWu6666 This should be fixed in latest main, can you check?

XiaoyunWu6666, 2021-08-25:
> @XiaoyunWu6666 This should be fixed in latest main, can you check?

I'm afraid not . in today's inner daily 6047 , it appeared again on APL_UP2_NOCODEC_ZEPHYR in case check-kmod-load-unload

XiaoyunWu6666, 2021-08-31:
from inner daily 6200 , after excluding affects from previous testcases , check-runtime-pm-status and kmod-load-unload worked well without 'dsp not suspended'

I'll close this issue if next daily works well too , thanks

XiaoyunWu6666, 2021-09-07:
in inner daily 6362 this is good , let's close it

closed 2021-09-07 by XiaoyunWu6666