thesofproject/sof #4508: [BUG]firmware boot failure after resume ,on ADLS_RVP_HDA and IO error happened

XiaoyunWu6666, 2021-07-20:
**Describe the bug**
firmware boot failure after resume ,on ADLS_RVP_HDA and IO error happened
inner daily 5357?model=ADLS_RVP_HDA&testcase=check-runtime-pm-status-15
[ 3174.181975] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: firmware boot failure
[ 3174.181997] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3174.182044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 3174.182115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x4170 0x0 0x0 0x0 0x2550114 0x0
[ 3174.182121] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3174.182126] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg ~/sof-test/test-case/check-runtime-pm-status.sh -l 15

**Reproduction Rate**
not sure
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 8877846e
SOF Branch: main
SOF Commit: f96e25ffcc8d
 Topology:sof-hda-generic-2ch.tplg 
 Platform: ADLS_RVP_HDA

**Screenshots or console output**
[console]
```
2021-07-19 22:19:47 UTC [REMOTE_INFO] ===== Iteration 11 of 15 for HDA Analog =====
2021-07-19 22:19:47 UTC [REMOTE_COMMAND] arecord -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/null -q
arecord: main:830: audio open error: Input/output error
/home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh: line 98: kill: (54649) - No such process
2021-07-19 22:19:49 UTC [REMOTE_ERROR] arecord process for pcm HDA Analog is not alive
2021-07-19 22:19:49 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2021-07-19 22:19:49 UTC [REMOTE_ERROR]  die()  @  /home/ubuntu/sof-test/test-case/../case-lib/lib.sh
2021-07-19 22:19:49 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh:101
```
[dmesg]
```
[ 3172.170547] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download successful, booting...
[ 3174.181975] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: firmware boot failure
[ 3174.181997] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3174.182044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 3174.182115] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x4170 0x0 0x0 0x0 0x2550114 0x0
[ 3174.182121] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3174.182126] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[ 3174.182147] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -5
[ 3174.182175] kernel:  Analog Playback and Capture: soc_pcm_open() failed (-5)
[ 3174.182183] kernel:  HDA Analog: ASoC: dpcm_be_dai_startup() failed at Analog Playback and Capture (-5)
[ 3174.182193] kernel:  HDA Analog: dpcm_fe_dai_startup() failed (-5)
[ 3174.722549] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3174.722613] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3174.723318] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3174.723324] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3175.802625] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 3175.802639] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 3175.802695] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3175.802721] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3175.803149] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3175.803152] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 3175.899047] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 3175.899063] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
```

mengdonglin, 2021-07-20:
@slawblauciak thinks this may have relation to #4480

kv2019i, 2021-07-20:
So I guess the link to issue #4480 is patch https://github.com/thesofproject/sof/pull/4466 . Multiple boots have succeeded before the failure, so this could be related.

lgirdwood, 2021-07-21:
Are we booting from IMR of using code loader here ?

XiaoyunWu6666, 2021-08-03:
did not see this in CI daily tests , was there any possible solutions that has solved it?
I lower the priority , shall we close it?

lgirdwood, 2021-08-03:
@XiaoyunWu6666 lets monitor this over next few days and we can close if it's no longer reproducing.

keqiaozhang, 2021-08-31:
This issue cannot be reproduced after upgrading to ES silicon.
500 iteration warm reboot test passed.
489/500 iterations suspend/resume test passed, the failure is caused by the irrelevant onboard network issue:
```
[64104.017868] kernel: e1000e 0000:00:1f.6 eno0: Hardware Error
```
So this issue can be closed.

closed 2021-08-31 by keqiaozhang