thesofproject/sof #4519: [BUG] glitch in 14_04_TestDemuxSynchronizedCapture48000Hz32b32b4ch

wszypelt, 2021-07-21:
**Describe the bug**
Test 14_04_TestDemuxSynchronizedCapture48000Hz32b32b4ch
Glitch was found near the start of the stream.

**Topology**
```
Topology:

        pipe_plb
    +-------------------------------+
    | +------+   +------+   +-----+ |                                                 +------+
    | | Host |-->| Buff |-->| Dai |-------------------------------------------------->| SSPx |
    | +------+   +------+   +-----+ |                                                 +------+
    +-------------------------------+                                                    |
                                                                                         |
                                                                                         |
        pipe_cap                                                                         |
    +----------------------------------------------------------------------------+       |
    | +-------+   +------+   +-----+   +------+   +-------+   +------+   +-----+ |    +------+
    | | Host1 |<--| Buff |<--| Vol |<--| Buff |<--| DeMux |<--| Buff |<--| Dai |<-----| SSPx |
    | +-------+   +------+   +-----+   +------+   +-------+   +------+   +-----+ |    +------+
    +-------------------------------------------------|--------------------------+
                                                      |
        pipe_cap                                      |
    +-------------------------------------------+     |
    | +-------+   +------+   +-----+   +------+ |     |
    | | Host2 |<--| Buff |<--| Vol |<--| Buff |<------+
    | +-------+   +------+   +-----+   +------+ |
    +-------------------------------------------+

```
"Demuxing" one playback pipeline using Demux component in capture pipeline into two capture pipelines, where
Demux component is included in Host-Dai pipeline with Volume component.
loopback signal on external SSP connection. Record signals on capture pipelines with audio verification.

**To Reproduce**
Run test with Diagnostic driver 14_04_TestDemuxSynchronizedCapture48000Hz32b32b4ch
With options --playback-iterations=75

**Reproduction Rate**
Sporadic. 1/75

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: BRANCH: main HASH: 99ebc324f4ef7cd3c4869761e38a0f32ed809302

2) Name of the platform(s) on which the bug is observed.
    * Platform: ICL RVP
3) Test environment:
BRANCH: master HASH: 52309bc5c72c600bf7cfd86e75ab45ecbf075feb

**Screenshots or console output**
[image]

**Logs**
[2021_07_20_01_49_04.zip](https://github.com/thesofproject/sof/files/6853413/2021_07_20_01_49_04.zip)

kkarask, 2021-08-20:
Glitches are visible in both captures with above topology on WHL RVP Platform with reproduction rate 1/75.

Verify captured stream 1:
[image]

Verify captured stream 2:
[image]

Logs:
[WHL_14_04_TestDemuxSynchronizedCapture48000Hz32b32b4ch.zip](https://github.com/thesofproject/sof/files/7020767/WHL_14_04_TestDemuxSynchronizedCapture48000Hz32b32b4ch.zip)

antonirx, 2023-01-05:
unsupported platform
on TGL issue does not occur
Tested FW hash: 4943aaa

closed 2023-01-05 by wszypelt