thesofproject/sof #4522: [BUG] Component tuning and audio test scripts do not work with topology2

singalsu, 2021-07-21:
**Describe the bug**

The Matlab language scripts in tools/tune and tools/test/audio directories do not work with topology2.

**To Reproduce**
Create configuration blob for component or run audio_process_test().

**Reproduction Rate**

**Expected behavior**
Similar as with topology1

**Impact**
Showstopper for doing new component setup blobs with topology2. Only minimal configuration blobs
set for builds needs has been ported.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

singalsu, 2022-05-13:
There's work in progress PR in #5603.  It depends on #5306 also.

singalsu, 2023-02-22:
IIR and FIR changes are merged. TDFB, and MFCC are remaining.

kv2019i, 2023-03-06:
@singalsu Still on track for V2.5 -- we aim to branch this week?

singalsu, 2023-03-06:
TDFB and MFCC won't make it to v2.5. We don't have in IPC4 the controls framework for TDFB (enum, switch). They are missing also from module adapter, so quite a lot work. In MFCC I have not yet figured out use of compressed stream.

lgirdwood, 2023-05-10:
@singalsu on track for v2.6 ? or should be moved to v2.7 ?

singalsu, 2023-05-15:
@lgirdwood Thanks, I moved to v2.7. We still have many components without IPC4 support, so even if changing the tools and blobs could happen before start of June, we can't test them. One more month looks possible.

lgirdwood, 2023-07-04:
@singalsu any update ? how are we trending ?

singalsu, 2023-07-25:
> @singalsu any update ? how are we trending ?

I've been doing these updates the same time with module API conversions and creating topologies to test them. DC blocker is still without a PR or draft PR. I plan  to do it after the crossover and multi-band drc conversions. It's fortunately much smaller effort than these two. One month ETA for all work merged?

alex-cri, 2023-08-21:
@singalsu , how have things progressed? Would like to move to 2.8 if remaining effort will take longer than a week.

alex-cri, 2023-09-06:
@singalsu , is there an update?

singalsu, 2023-09-06:
> @singalsu , is there an update?

I think all non-deprecated (*) components have a module adapter and ipc4 PR done. But some are still not yet merged. We miss tplg2 support PRs still for some. The enum control that worked with tplg1 is not yet solved for tplg2. It's however used only by one component (beamformer) that has not been included to signed FW builds.

So all the work (tplg2 support) won't be completed in week so we can move on to 2.8.

(*) We will deprecate now tone generator and selector - turn kconfig off and have warning if enabled.

singalsu, 2023-10-16:
All components those have setup script for topology1 now have setup scripts for topology2. This can be closed now!

closed 2023-10-16 by singalsu