thesofproject/sof #4526: [BUG][APL]dai_sg_alloc ERROR when multiple-pipeline on APL_UP2_NOCODEC

XiaoyunWu6666, 2021-07-22:
**Describe the bug**
inner daily 5401?model=APL_UP2_NOCODEC&testcase=multiple-pipeline-all
when multiple-pipeline ,dms_sg_alloc failed with err -12 in dai_capture_params. causing  ipc error when PCM_PARMS 

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: d54b7803
SOF Branch: main
SOF Commit: 1f4395f8057e
Topology: sof-apl-nocodec.tplg
)Platform: APL_UP2_NOCODEC

**Screenshots or console output**
[sof-logger]
```
[      302221.185907] (           0.000000) c0 dai          6.34           src/audio/dai.c:429  ERROR dai_capture_params(): dma_sg_alloc() for period_count 4 period_bytes 384 failed with err = -12
[           6.302083] (           6.302083) c0 pipe         6.35  ....../pipeline-params.c:235  ERROR pipeline_params(): ret = -12, host->comp.id = 30
[          12.447916] (           6.145833) c0 ipc                  src/ipc/ipc3/handler.c:271  ERROR ipc: pipe 6 comp 30 params failed -12
```

[dmesg]
```
[ 5291.077680] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 5291.078667] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[ 5291.078817] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hw params ipc failed for stream 6
[ 5291.078825] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:0e.0: -12
[ 5291.078863] kernel:  Port5: ASoC: soc_pcm_hw_params() failed (-12)
[ 5291.078895] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5291.079651] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5291.079664] kernel:  Port5: ASoC: dpcm_fe_dai_hw_params failed (-12)
```
[console]
```
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
```

ranj063, 2021-07-22:
@XiaoyunWu6666 could you please check if https://github.com/thesofproject/sof/pull/4521/ can fix this?

marc-hb, 2021-07-22:
Before testing a fix: can this be reproduced easily?

XiaoyunWu6666, 2021-07-22:
> Before testing a fix: can this be reproduced easily?

yes, it can

ranj063, 2021-07-22:
> Before testing a fix: can this be reproduced easily?

@marc-hb I added a PR to add 2 more PCMs to all nocodec topologies yesterday for deep buffer support. Im not surprised this showed up. If Keyon's PR doesnt help, we might need to trim down the pipelines for APL nocodec.

XiaoyunWu6666, 2021-07-22:
> @XiaoyunWu6666 could you please check if #4521 can fix this?

Yes let me check

keyonjie, 2021-07-22:
the PR#4521 is not quite ready yet, let's try it later.

marc-hb, 2021-07-22:
> @marc-hb I added a PR to add 2 more PCMs to all nocodec topologies yesterday for deep buffer support. Im not surprised this showed up. 

Thanks, mentioning commit 6516a8c89ba0 would have helped.

plbossart, 2021-07-22:
I just don't get how we end-up with such regressions.

Ranjani's change to use the deep-buffer was tested with PR #4516 - no problems detected.
https://sof-ci.01.org/sofpr/PR4516/build9704/devicetest/

The mixer change was also tested https://sof-ci.01.org/sofpr/PR4517/build9703/devicetest/

????

plbossart, 2021-07-22:
Bisecting results:

````
c5f50f533ff876600bc0899822e9763c7ab75cae is the first bad commit
commit c5f50f533ff876600bc0899822e9763c7ab75cae
Author: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
Date:   Mon Apr 12 17:58:44 2021 +0300

    Drivers: Intel: DMIC: Separate decimation filters setup computation
    
    This patch moves the run-time decimation factors search and filter
    coefficients scaling code to separate module as preparation to add
    more functionality to the driver. There are no changes to
    functionality.
    
    Signed-off-by: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>

 src/drivers/intel/dmic/CMakeLists.txt  |    2 +-
 src/drivers/intel/dmic/dmic.c          | 1071 +-------------------------------
 src/drivers/intel/dmic/dmic_computed.c | 1002 ++++++++++++++++++++++++++++++
 src/include/sof/drivers/dmic.h         |   80 +++
 zephyr/CMakeLists.txt                  |    4 +
 5 files changed, 1100 insertions(+), 1059 deletions(-)
 create mode 100644 src/drivers/intel/dmic/dmic_computed.c

````

plbossart, 2021-07-22:
https://github.com/thesofproject/sof/pull/4532 reverts all the recent commits impacting APL from the first bad commit above.

ranj063, 2021-07-22:
> Ranjani's change to use the deep-buffer was tested with PR #4516 - no problems detected.

we dont run multiple-pipeline-all test in the device test for PRs. thats why

XiaoyunWu6666, 2022-01-24:
haven't seen it in CI for months

closed 2022-01-24 by XiaoyunWu6666