thesofproject/sof #4538: [BUG]DSP panic when suspend-resume on cml-sku0983-sdw

XiaoyunWu6666, 2021-07-23:
**Describe the bug**
in inner daily  /5449?model=CML_SKU0983_SDW&testcase=check-suspend-resume-50
DSP panic when suspend-resume 

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 50
**Reproduction Rate**
 not 100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: d54b7803 storage: d54b7803-2
SOF Branch: main
SOF Commit: 52bc39fe73f9 storage: 52bc39fe73f9
 Topology: /lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg 
 Platform:  cml-sku0983-sdw

**Screenshots or console output**
[console]
```
2021-07-22 23:04:39 UTC [REMOTE_INFO] ===== Round(47/50) =====
2021-07-22 23:04:39 UTC [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Thu Jul 22 23:04:45 2021
2021-07-22 23:04:46 UTC [REMOTE_COMMAND] sleep for 5
2021-07-22 23:04:51 UTC [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1626995074"
2021-07-22 23:04:51 UTC [ERROR] Caught kernel log error
===========================>>
[ 5561.663293] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 5561.663306] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5561.663320] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 5561.663392] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 5561.663398] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
<<===========================

```

keyonjie, 2021-08-16:
@ujfalusi looks we have very limited dmesg log when DSP panic happen now?
```
[ 5561.663293] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 5561.663306] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5561.663320] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 5561.663392] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 5561.663398] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

ujfalusi, 2021-08-16:
@keyonjie, by default we are not flooding the kernel log with repeated dumps when the firmware crashes, we print the dsp dump only once and we also print the ipc dump once (separately counted).
All optional dumps and the dump flood can be re-enabled via module parameter `sof_debug` with bit 4 is set.

The dumps have not been changed, we print the same information as before but we have markers around to make them easier to spot.

lgirdwood, 2021-08-16:
@ujfalusi does the FW stack get dumped by default (it should not be an option) ?

ujfalusi, 2021-08-16:
The content of the dump has not been changed and there is no option to change it. The only change is that by default we print the dump (DSP / IPC) once.
The (HDA) registers will be dumped if the fw_state is SOF_FW_BOOT_COMPLETE, if the dump is called during any other state then the registers are not printed and they were not printed in the past either.

XiaoyunWu6666, 2021-08-20:
happen in inner daily 5971 on model CML_SKU0983_SDW in testcase check-suspend-resume-50

keqiaozhang, 2021-08-23:
@fredoh9 , can you help to follow up this issue? since we only have one device of this model in JF and it's hard for us reproduce the suspend/resume issue via ssh.

lgirdwood, 2021-08-28:
> I can reproduce this on CML_SKU0983_SDW. I think if we use iteration of 50, high chance to reproduce.
> 
> @lgirdwood Where is the crash dump script? 

sof-coredump-reader.py is in sof/tools/coredumper/

> And I can try to run the script manually then I can try to integrate to SOF CI test. 

Great :)

> Do you recommend sof_debug to set bit 4 also?

What will this do ?

YongAnLu00, 2021-08-30:
happen in inner daily 6175 on model ADLP_RVP_NOCODEC in testcase check-suspend-resume-50

lgirdwood, 2021-08-30:
@fredoh9 
>> Do you recommend sof_debug to set bit 4 also?
> What will this do ?

Yes - just read above about this - yes please do make sure this is enabled by default in the kernel for CI.

@XiaoyunWu6666 @YongAnLu00 can we make sure bit 4 is enabled by default for daily test too. Thanks

keqiaozhang, 2021-09-06:
> @XiaoyunWu6666 @YongAnLu00 can we make sure bit 4 is enabled by default for daily test too. Thanks

@lgirdwood do you mean to enable the dynamic pipeline test on all of our CI devices?

lgirdwood, 2021-09-06:
> > @XiaoyunWu6666 @YongAnLu00 can we make sure bit 4 is enabled by default for daily test too. Thanks
> 
> @lgirdwood do you mean to enable the dynamic pipeline test on all of our CI devices?

No sorry,  I mean enable the kernel option to display FW crash stack and registers on all CI builds (it should be enabled by default, but Im not seeing it on CI or daily  test results pasted here). @keyonjie knows all the options so could locally confirm if any option is needed.

keqiaozhang, 2021-09-07:
@lgirdwood  so you mean to enable the bit 1(SOF_DBG_RETAIN_CTX)?
```
/* debug flags */
#define SOF_DBG_ENABLE_TRACE    BIT(0)
#define SOF_DBG_RETAIN_CTX      BIT(1)  /* prevent DSP D3 on FW exception */
#define SOF_DBG_VERIFY_TPLG     BIT(2) /* verify topology during load */
#define SOF_DBG_DYNAMIC_PIPELINES_OVERRIDE      BIT(3) /* 0: use topology token
                                                        * 1: override topology
                                                        */
#define SOF_DBG_DYNAMIC_PIPELINES_ENABLE        BIT(4) /* 0: use static pipelines
                                                        * 1: use dynamic pipelines
                                                        */
#define SOF_DBG_DISABLE_MULTICORE               BIT(5) /* schedule all pipelines/widgets
                                                        * on primary core
                                                        */
#define SOF_DBG_PRINT_ALL_DUMPS         BIT(6) /* Print all ipc and dsp dumps */
```

lgirdwood, 2021-09-07:
@keqiaozhang thanks, got you now. Bit 6 should be enabled as FW registers were missing from some logs.
@ujfalusi if we have an IPC timeout and there are NO FW registers can we print something in dmesg to that effect. i.e. ```no FW context to dump.  DSP hang ?```

ujfalusi, 2021-09-07:
@lgirdwood, certainly. I have other debug related changes here and there, I will collect them to a separate PR.

fredoh9, 2021-09-07:
I overlooked the codes. Yes, looks like sof_debug parameter to have bit 6(SOF_DBG_PRINT_ALL_DUMPS) and also 'flags' need to set bit 4(SOF_DBG_DUMP_OPTIONAL), below two locations?

https://github.com/thesofproject/linux/blob/6a95ae1a6a50ef29d01434775b66546ab85a2971/sound/soc/sof/debug.c#L866

https://github.com/thesofproject/linux/blob/6a95ae1a6a50ef29d01434775b66546ab85a2971/sound/soc/sof/ops.c#L160

```
#define SOF_DBG_DUMP_REGS		BIT(0)
#define SOF_DBG_DUMP_MBOX		BIT(1)
#define SOF_DBG_DUMP_TEXT		BIT(2)
#define SOF_DBG_DUMP_PCI		BIT(3)
#define SOF_DBG_DUMP_OPTIONAL		BIT(4) /* only dump if SOF_DBG_PRINT_ALL_DUMPS is set */
```

ujfalusi, 2021-09-10:
@fredoh9, the `SOF_DBG_PRINT_ALL_DUMPS` ( BIT(6) ) is for the `sof_debug` module parameter.
The `SOF_DBG_DUMP_OPTIONAL` is used internally by `snd_sof_dsp_dbg_dump()` to suppress some of the noise when the kernel is not debugged.
By setting the SOF_DBG_PRINT_ALL_DUMPS bit in sof_debug we will print the dumps marked in code with SOF_DBG_DUMP_OPTIONAL _and_ we will print _all_ dumps, never silencing them as we do otherwise.

ujfalusi, 2021-09-10:
Regarding to the error print of
```
[ 5561.663293] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 5561.663306] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5561.663320] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 5561.663392] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 5561.663398] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
```
It is printed in response to SOF_IPC_PANIC_MAGIC, the code is 'HDA_DSP_ROM_FW_ENTERED' and `hda_dsp_dump()` decided to print only the extended rom status because the sdev->fw_state was _not_ SOF_FW_BOOT_COMPLETE.

This is most likely happen if the firmware crashes before it had a chance to send the SOF_IPC_FW_READY message?

lgirdwood, 2021-09-20:
> It is printed in response to SOF_IPC_PANIC_MAGIC, the code is 'HDA_DSP_ROM_FW_ENTERED' and `hda_dsp_dump()` decided to print only the extended rom status because the sdev->fw_state was _not_ [...]

@ujfalusi This is a bug - kernel should print out any FW crash even before FW boot complete message. Can you fix ?

ujfalusi, 2021-09-20:
@lgirdwood , sure, I can take a look. Fwiw we never printed more information before the firmware boot has been completed.
Even before https://github.com/thesofproject/sof/commit/6ca5cecbd1c17 more information was only printed if `sdev->boot_complete` was *true*

I'm not sure yet, but before the boot is completed we might have invalid garbage we would be presenting to user? That could be misleading for the debugging.

lgirdwood, 2021-09-21:
@ujfalusi I dont think we will get garbage, the FW has to actively send the exception IPC so relevant data should be present. This can happen long before we send the boot complete message.

plbossart, 2021-09-21:
FWIW I reached this DSP panic on CML_SKU09C6_SDW after 150 iterations while trying to see if PR #4785 solved issue https://github.com/thesofproject/sof/issues/4779

ujfalusi, 2021-09-22:
@lgirdwood , https://github.com/thesofproject/linux/pull/3170 should improve the quality and usefulness of our dumps.

1994lwz, 2021-09-28:
This issue is reproduce on the cml-sku0955-hda in the daily 6885.

keyonjie, 2021-10-11:
The FW find there is no EDF task to run and asserts here:

```
	/* having next task is mandatory */
	assert(task_next);
```

keyonjie, 2021-10-11:
the "trace point: 0x000041c0" means we failed in ipc_init().

XiaoyunWu6666, 2021-10-12:
> @XiaoyunWu6666, the DSP panic is always shown. Before my PR we would not have a register and backtrace as the fw_state is not yet `SOF_FW_BOOT_COMPLETE`. With the ` SOF_DBG_PRINT_ALL_DUMPS` you will make sure that all other DSP/IPC dumps will be shown from different paths, like IPC timeouts or boot failures before the firmware is set to run.
> 
> I think it is still good to keep it set, but I don't expect too useful information coming from it, mostly extended noise.

Thanks for providing details. Will keep it set :)

XiaoyunWu6666, 2021-10-12:
happen on CML_SKU0983_SDW in inner daily 7284 again

plbossart, 2021-10-13:
Also seen in my private tests for deep-buffer on CML_SKU0983_SDW
https://sof-ci.sh.intel.com/#/result/planresultdetail/7312?model=CML_SKU0983_SDW&testcase=check-suspend-resume-50

ranj063, 2021-10-14:
@lgirdwood @plbossart I have a theory: We never "complete" the IPC task when the CTX_SAVE is sent and instead just power off the memory and return the IPC response. I'm not certain what happens when we power off the SRAM ie if it is really powered off or it is in self-refresh and somehow when we resume and initialize the IPC EDF task again, the task list is non-empty from the previous boot. I confirmed my theory with this change in the FW: 
```
diff --git a/rimage b/rimage
index 896009740..9a26e4558 160000
--- a/rimage
+++ b/rimage
@@ -1 +1 @@
-Subproject commit 89600974030fcbca903894c327c3c64afcede14d
+Subproject commit 9a26e4558094f19f3f7becd89eb9e8a9a9dd82b9
diff --git a/src/include/sof/trace/trace.h b/src/include/sof/trace/trace.h
index 543f735ee..aefbaacda 100644
--- a/src/include/sof/trace/trace.h
+++ b/src/include/sof/trace/trace.h
@@ -72,6 +72,7 @@ struct tr_ctx;
 #define TRACE_BOOT_PLATFORM_SSP_FREQ   (TRACE_BOOT_PLATFORM + 0x1A0)
 #define TRACE_BOOT_PLATFORM_DMA                (TRACE_BOOT_PLATFORM + 0x1B0)
 #define TRACE_BOOT_PLATFORM_IPC                (TRACE_BOOT_PLATFORM + 0x1C0)
+#define TRACE_BOOT_PLATFORM_IPC_NOTEMPTY (TRACE_BOOT_PLATFORM + 0x1C1)
 #define TRACE_BOOT_PLATFORM_IDC                (TRACE_BOOT_PLATFORM + 0x1D0)
 #define TRACE_BOOT_PLATFORM_DAI                (TRACE_BOOT_PLATFORM + 0x1E0)
 #define TRACE_BOOT_PLATFORM_SSP                (TRACE_BOOT_PLATFORM + 0x1F0)
diff --git a/src/schedule/edf_schedule.c b/src/schedule/edf_schedule.c
index a0a3b46eb..3815cade3 100644
--- a/src/schedule/edf_schedule.c
+++ b/src/schedule/edf_schedule.c
@@ -143,6 +143,9 @@ int schedule_task_init_edf(struct task *task, const struct sof_uuid_entry *uid,
        if (ret < 0)
                return ret;
 
+       if (!list_is_empty(&task->list))
+               trace_point(TRACE_BOOT_PLATFORM_IPC_NOTEMPTY);
+
        if (edf_sch_get_pdata(task))
                return -EEXIST;
 
```

What I see in the log when the error happens is that the task list for the IPC EDF task is non-empty right away during initialization.

[12093.612868] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[12093.612880] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[12093.612884] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[12093.612899] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[12093.613190] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: reason: assertion failed (0xb)
[12093.613196] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: trace point: 0x000041c1
[12093.613202] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: panic at src/schedule/edf_schedule.c:98
[12093.613208] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: DSP Firmware Oops
[12093.613212] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060025 SAR     0x00000000
[12093.613220] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe02be0d EPC3     0x00000000 EPC4    0x00000000
[12093.613228] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[12093.613234] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: EPS2     0x00060c20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[12093.613240] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[12093.613246] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: stack dump from 0xbe086140
[12093.613254] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086140: 00000000 00000000 00000000 00000000
[12093.613261] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086150: 00000000 00000000 00000000 00000000
[12093.613267] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086160: be086140 00000000 00000000 00000000
[12093.613273] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086170: 00000000 00000000 00000000 00000000
[12093.613279] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086180: 00000000 00000000 00000000 00000000
[12093.613284] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe086190: 00000000 00000000 2f637273 65686373
[12093.613290] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe0861a0: 656c7564 6664652f 6863735f 6c756465
[12093.613297] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: 0xbe0861b0: 00632e65 00000000 00000062 ffffffff
[12093.613320] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[12093.912859] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[12093.912866] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[12093.912877] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[12093.912910] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: extended rom status:  0x5 0xdead00b 0x41c1 0x0 0x0 0x0 0x1811102 0x0

Now given this, there are 2 things we can possibly do: 
1. during edf task init, we could call list_init for the task list to ensure it is infact empty.
2. On the other hand, I feel like the right thing to do it do a graceul shutdown during the CTX_SAVE IPC to make sure that the IPC task is complete and there are no new tasks in the task list. We could achieve this by possibly moving the call to platform_pm_runtime_power_off() from ipc_platform_do_cmd() to task_main_primary_core().

keyonjie, 2021-10-14:
@ranj063 if the task->list is not initialized, it will not be empty. We can simply add initialization e.g.
```
diff --git a/src/ipc/ipc-common.c b/src/ipc/ipc-common.c
index 3f4ce758e629..392401e9ff5c 100644
--- a/src/ipc/ipc-common.c
+++ b/src/ipc/ipc-common.c
@@ -242,6 +242,7 @@ int ipc_init(struct sof *sof)
        spinlock_init(&sof->ipc->lock);
        list_init(&sof->ipc->msg_list);
        list_init(&sof->ipc->comp_list);
+       list_init(&sof->ipc->ipc_task.list);
 
        return platform_ipc_init(sof->ipc);
 }
```

But why do you think the uninitialization of this list will lead to EDF task list empty, The ->next and ->prev will be configured in schedule_edf_task() anyway.

ranj063, 2021-10-14:
> so, the list here is just not initialized, it is not like the "not empty" as what we usual think.

I see, I think I made a mistake. I should really be checking for the edf_pdata->list.

ranj063, 2021-10-14:
Basically, whats surprising is that the EDF task gets executed without a single IPC being sent from the host

keyonjie, 2021-10-15:
Looks we see this in today's daily test also.
@XiaoyunWu6666 in case you can reproduce it manually, can you help to try if https://github.com/thesofproject/sof/pull/4874 help?

XiaoyunWu6666, 2021-10-15:
in today's inner daily test , this happened again .
@keyonjie  I will try your PR ASAP

ranj063, 2021-10-15:
> hi @keyonjie I tried #4874 , the error did not happen with 100 rounds .But this issue is not 100% manually reproducible .

@keyonjie #4874 fixes the symptom not the root cause.

miRoox, 2021-11-22:
cannot be reproduced, close.

lyakh, 2021-11-23:
I think the problem is not about whether or not the task list is empty, but that an EDF software interrupt triggers after resume. IIUC, EDF scheduling is done from software interrupts. As long as an EDF interrupt triggers, `edf_scheduler_run()` will run. And if the task list is empty, or if tasks on the list aren't suitable to be run, we get an `assert()`. So the real question seems to be why we're getting a software EDF interrupt after resume? Has it been scheduled before suspend or after resume or is it spurious (unlikely)? Can it be, that it was scheduled before suspend and somehow survived the suspend? If it was correctly scheduled before suspend but just couldn't be processed, it should be safe to drop it after resume. Can we clear the interrupt in `scheduler_restore_edf()`?

lyakh, 2021-11-24:
An important question: do we know on which core the panic happens? I assume this isn't happening when restoring a secondary core after a D0ix?
After looking at the code some more, I don't see a good way to clear the EDF (software) interrupt. I think `interrupt_mask()` would would clear it, as described in an Architecture Specification document, so we could add a
```
interrupt_mask();
interrupt_unmask();
```
pair of calls to `scheduler_init_edf()` anywhere before the call to `interrupt_enable()`, but I'm not 100% sure this would work. E.g. as in #5026 . If that doesn't work, I'd propose to remove the `assert()` call at least temporarily to win us more time to investigate this properly.

mengdonglin, 2021-11-29:
This issue can be reproduced on an ADL-P device with https://github.com/thesofproject/sof/releases/tag/v1.9.2, but cannot reproduce with latest main branch

miRoox, 2021-12-06:
It was fixed by #4874, the patch was also cherry-picked to [stable-v1.9](https://github.com/thesofproject/sof/commit/a978086db17171ae097c0f2b4a07f9a1a2beaa7d) and [stable-v2.0](https://github.com/thesofproject/sof/commit/6672f251449aeb262420da6be8f6d6e211d17417).

closed 2021-12-06 by miRoox