thesofproject/sof #4539: [BUG][Zephyr][JSL] boot fails on cavs20 platforms

kv2019i, 2021-07-23:
**Describe the bug**
SOF-Zephyr boot fails on cavs20 platforms like JSL.

**To Reproduce**
Build SOF-on-Zephyr with icl target, boot on DUT.

**Reproduction Rate**
100% (tested on two separate systems)

**Expected behavior**
Boot successful.

**Impact**

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.13.0-rc5-daily-nocodec-20210616-0 (works with XTOS SOF build)
    * SOF: 4c376680f55bf9264afec7b229a259852920bb6f
    * Zephyr: 7a2b248b9804cc59a62b3b7dcdf00f9475ef64e1
2) Name of the topology file
    * Topology: sof-jsl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: JSL RVP

**Screenshots or console output**
[  550.037336] sof-audio-pci-intel-icl 0000:00:1f.3: Firmware download successful, booting...
[  552.043751] sof-audio-pci-intel-icl 0000:00:1f.3: error: firmware boot failure
[  552.043797] sof-audio-pci-intel-icl 0000:00:1f.3: status: fw entered - code 00000005
[  552.043865] sof-audio-pci-intel-icl 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x2100186 0x0
[  552.043875] sof-audio-pci-intel-icl 0000:00:1f.3: error: failed to boot DSP firmware -5
[  552.044410] sof-audio-pci-intel-icl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010303 successful

lgirdwood, 2022-02-15:
@kv2019i @lyakh who has JSL or cavs20 platform.

kv2019i, 2022-02-15:
@lgirdwood  I think only me. There's UP board with JSL, making this is a bit hard target to support for community release.

kv2019i, 2022-02-18:
Did a quick test with latest SOF&Zephyr and boot is still failing. FW download is successful (auth is ok), but boot fails to a timeout. Something goes wrong in the early Zephyr boot process.

kv2019i, 2022-03-09:
Reproduce the issue with plain Zephyr:
```
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1006701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Trimming 16 bytes of extended manifest
INFO:cavs-fw:Resetting HDA device
INFO:cavs-fw:Powering down DSP cores
INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
INFO:cavs-fw:Mapped 2M huge page at 0x162e00000 to contain 65536 bytes of firmware
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:Starting DSP, ADSPCS = 0x303
INFO:cavs-fw:Wait for ROM startup
INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
WARNING:cavs-fw:Load failed?  FW_STATUS = 0x0
INFO:cavs-fw:cAVS firmware load complete
```

lgirdwood, 2022-03-09:
@kv2019i is trace point enabled in Zephyr or can mailbox be used to write IDs before each init code action (like tracepoint).

lgirdwood, 2022-03-09:
@andyross fyi - fixes incoming for HPSRAM.

lgirdwood, 2022-03-10:
@kv2019i how should we deploy this for rc1 ?

kv2019i, 2022-03-14:
Fixed now in main, closing this.

closed 2022-03-14 by kv2019i