thesofproject/sof #4542: [BUG][ZEPHYR][HDA] hda-generic topologies don't work

lyakh, 2021-07-23:
**Describe the bug**
Tested a sof-hda-generic topology on Mantis / CNL and it exploded with HDA link DMA timeouts. Tested both dynamic and static topology, single- and multi-core. The nocodec topology works with Zephyr on the same hardware and the same kernel.

**To Reproduce**
load Zephyr-built firmware.

**Reproduction Rate**
100%

plbossart, 2021-08-05:
@lyakh I would assume the same problem happens on Up2 without any ACPI initrd override - plain HDMI playback would also fail?

kv2019i, 2021-08-10:
I've not seen issues with HDMI/DP tests, so this sounds a bit odd. I'll take a look.

kv2019i, 2021-08-13:
Ran a full test on TGL HDA machine and no issues. So either this specific to CNL/WHL, or it was fixed with recent commits. Assigning back to @lyakh to revalidate on Mantis.

lgirdwood, 2021-11-11:
@kv2019i @lyakh can we close ?

marc-hb, 2021-11-11:
I've been using `sof-hda-generic-4ch.tplg` with Zephyr on my WHL Up Xtreme and I don't think I have seen anything "explode" but what does that mean? Can we have some basic reproduction steps and error message?

keqiaozhang, 2021-11-12:
We haven't seen this issue on TGL UP Xtreme.

lyakh, 2021-11-12:
> @kv2019i @lyakh can we close ?

@lgirdwood just double-checked: I'm still getting a flood of xruns, followed by a DSP panic in `host_copy_normal()`:
```
[00005946] <err> os:  ** FATAL EXCEPTION
[00005946] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[00005946] <err> os:  **  PC 0xbe0228b2 VADDR (nil)
[00005946] <err> os:  **  PS 0x60020
[00005946] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[00005946] <err> os:  **  A0 0xbe035e5a  SP 0xbe04d630  A2 (nil)  A3 0x9e060e00
[00005946] <err> os:  **  A4 0xbe060ec0  A5 0x1  A6 0x1  A7 0x1
[00005946] <err> os:  **  A8 0xbe04d780  A9 0xbe04d760 A10 (nil) A11 (nil)
[00005946] <err> os:  ** A12 0xbe04d7a0 A13 0x1 A14 0x60c25 A15 0xbe04d774
[00005946] <err> os:  ** LBEG 0xbe012d69 LEND 0xbe012d6f LCOUNT (nil)
[00005946] <err> os:  ** SAR 0x20
[00005946] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00005946] <err> os: Current thread: 0x9e057dc0 (unknown)
[00005948] <err> os: Halting system
```
```
be0228a4 <host_copy_normal>:
be0228a4:       008136                  entry   a1, 64
be0228a7:       1d2242                  l32i    a4, a2, 116
be0228aa:       203220                  or      a3, a2, a2
be0228ad:       0124a2                  l32i    a10, a4, 4
be0228b0:       020c                    movi.n  a2, 0
be0228b2:       0a68                    l32i.n  a6, a10, 0
be0228b4:       5129                    s32i.n  a2, a1, 20
be0228b6:       c668                    l32i.n  a6, a6, 48
be0228b8:       4129                    s32i.n  a2, a1, 16
be0228ba:       0e2662                  l32i    a6, a6, 56
be0228bd:       10c1c2                  addi    a12, a1, 16
be0228c0:       14c1b2                  addi    a11, a1, 20
be0228c3:       172452                  l32i    a5, a4, 92
be0228c6:       112472                  l32i    a7, a4, 68
be0228c9:       0006e0                  callx8  a6
```

lyakh, 2021-11-12:
> @lyakh I would assume the same problem happens on Up2 without any ACPI initrd override - plain HDMI playback would also fail?

@plbossart sorry for not replying earlier. You mean also using the hda-generic topology on UP2? Sorry, I still haven't set up an UP2 in that mode. I have a second one, so I can use one with a hat and another one with HDA, just need to do that

lgirdwood, 2021-11-15:
@lyakh any solution for the oops, looks like a NULL ptr or an assert().

lyakh, 2022-02-15:
WHL daily tests with hda-generic run successfully in CI. Also just tried to reproduced on Mantis and the bug couldn't be reproduced. Closing

closed 2022-02-15 by lyakh