thesofproject/sof #4552: [BUG]ipc timed out when GLB_TRACE_MSG after PIPE3 COMPLETE on TGLU_RVP_NOCODEC_MULTICORE

XiaoyunWu6666, 2021-07-27:
**Describe the bug**
inner daily 5515?model=TGLU_RVP_NOCODEC_MULTICORE&testcase=check-runtime-pm-status-15
but cannot reproduce it yet

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-runtime-pm-status.sh -l 15
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 2113dc7f
SOF Branch: main
SOF Commit: 57ee04f2d931
 Topology: sof-tgl-nocodec-ci.tplg 
Platform:TGLU_RVP_NOCODEC_MULTICORE

**Screenshots or console output**
[dmesg]
```
[ 5555.224294] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 5555.723906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5555.723954] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5555.732973] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[ 5555.732989] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5555.732994] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5555.733008] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 5555.733015] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5555.733030] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5555.733035] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5555.733040] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5555.733054] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5555.733348] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5555.733354] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5555.733368] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.2.SSP0.IN id 13
[ 5555.733380] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 5555.733420] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 5556.223900] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5556.223945] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5556.236955] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[ 5556.236977] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.1.SSP0.OUT id 7
[ 5556.236990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[ 5556.739955] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[ 5556.739990] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[ 5557.243911] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x40020000 size 76
[ 5557.243929] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ctx_restore ipc error during resume -110
[ 5557.243940] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -110
[ 5557.243950] kernel:  NoCodec-1: soc_pcm_open() failed (-110)
[ 5557.243956] kernel:  Port1: ASoC: dpcm_be_dai_startup() failed at NoCodec-1 (-110)
[ 5557.243964] kernel:  Port1: dpcm_fe_dai_startup() failed (-110)
```

keyonjie, 2021-07-27:
looks the pipeline complete IPC timeout for pipeline 3 on core 3, @lyakh is it possible related with your commit "ipc: let secondary cores acknowledge IPC message"?

lgirdwood, 2021-07-27:
@XiaoyunWu6666 how reproducible is this ?
@keyonjie A lot of the multicore issues are related to cache coherency and shared object access races. @ranj063 work should be merged first before we start deep diving other multicore issues (since we dont wont to duplicate effort).

XiaoyunWu6666, 2021-08-02:
inner daily 5635?model=TGLU_RVP_NOCODEC_MULTICORE&testcase=check-kmod-load-unload-after-playback-5 has similar logs
[dmesg]
```
[   53.404531] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.3.SSP1.OUT id 19
[   53.404532] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   53.904112] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   53.904156] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   53.906138] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[   53.906152] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[   53.906156] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[   53.906173] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   53.906179] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   53.906188] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   53.906194] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[   53.906199] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   53.906213] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[   53.906498] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   53.906504] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   53.906512] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.2.SSP0.IN id 13
[   53.906523] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   54.404070] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   54.404110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   54.408818] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[   54.408839] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: complete pipeline PIPELINE.1.SSP0.OUT id 7
[   54.408851] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   54.911428] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30130000 size 12
[   54.911460] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[   55.414693] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x40020000 size 76
[   55.414714] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ctx_restore ipc error during resume -110
[   55.414754] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   55.917741] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   55.917759] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: -110 failed to prepare DSP for device removal
[   55.917983] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[   55.925199] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[   55.925222] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[   55.925229] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1

```

XiaoyunWu6666, 2021-08-02:
inner daily  5635?model=TGLU_RVP_NOCODEC_MULTICORE&testcase=check-runtime-pm-status-15 has similar logs .
When setting up new components ,ipc timed out happened
[dmesg]
```
[ 5355.456138] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5355.456161] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.456193] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5355.456215] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5355.456444] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.456458] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.956077] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5355.956110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5355.963999] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x50030000 size 124
[ 5355.964016] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5355.964021] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5355.964037] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 5355.964044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5355.964059] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5355.964066] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5355.964070] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5355.964083] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5355.964386] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5355.964391] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5355.964404] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: set/get ctrl ipc comp 27
[ 5355.964411] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed kcontrol value set for widget: 27
[ 5355.964416] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail to set up kcontrols for widget SMART_AMP5.0
[ 5355.964421] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore kcontrols for widget SMART_AMP5.0
[ 5355.964433] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 5355.964464] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 5356.456041] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5356.456082] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5356.468221] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30020000 size 12
[ 5356.468242] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free widget SMART_AMP5.0
[ 5356.468248] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore pipeline after resume -110
[ 5356.468259] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -110
[ 5356.468271] kernel:  NoCodec-0: soc_pcm_open() failed (-110)
[ 5356.468278] kernel:  Port0: ASoC: dpcm_be_dai_startup() failed at NoCodec-0 (-110)
[ 5356.468287] kernel:  Port0: dpcm_fe_dai_startup() failed (-110)
[ 5358.130979] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131003] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131204] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.357704] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 5359.357717] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 5359.357826] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.357873] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.358446] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.358451] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.616193] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 5359.616202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 5359.616246] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.616272] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.616599] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.616601] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data

```

lgirdwood, 2021-08-03:
Lets wait until @ranj063 multicore patches are merged as this will need retested.

XiaoyunWu6666, 2021-08-04:
in inner daily 5685 , this does not appear (but let's keep observing it in following daily tests)
FYI , another issue happened on TGLU_RVP_NOCODEC_MULTICORE, ipc timed out when SET_DATA
inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-50 AND
inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-with-capture-5

lgirdwood, 2021-08-04:
> another issue happened on TGLU_RVP_NOCODEC_MULTICORE, ipc timed out when SET_DATA
> inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-50 AND
> inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-with-capture-5

Please log this as a separate issue.

XiaoyunWu6666, 2021-08-05:
> > another issue happened on TGLU_RVP_NOCODEC_MULTICORE, ipc timed out when SET_DATA
> > inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-50 AND
> > inner 5685?model=TGLU_RVP_NOCODEC_CI&testcase=check-suspend-resume-with-capture-5
> 
> Please log this as a separate issue.

Yes , there are several separate issues for TGLU_RVP_NOCODEC_CI : #4586 #4553 #4603  . The two mentioned above is in #4586 

FYI , in today's inner daily 5709 , #4552 did not happen .

XiaoyunWu6666, 2021-08-09:
two duplicated issues : #4597  #4586

XiaoyunWu6666, 2021-08-19:
ipc timed out also happened with system suspend resume 
"failed to restore kcontrols for widget SMART_AMP5.0"
see inner daily 5946 5931

XiaoyunWu6666, 2021-08-30:
in inner daily 6175 testcase=multiple-pipeline-capture-50
COMP_CONNECTipc error happened when runtime resume

lgirdwood, 2021-08-30:
> in inner daily 6175 testcase=multiple-pipeline-capture-50
> COMP_CONNECT ipc error happened when runtime resume

Thanks @XiaoyunWu6666 are the logs on daily test for the IPC timeouts - is the plan for daily test to also post logs here too ?

XiaoyunWu6666, 2021-08-31:
Sorry temporarily there isn't any plan of posting logs automatically , but I can post vital logs every time I update comments in issues

XiaoyunWu6666, 2021-09-14:
Also happened in inner daily  6562
on model ADLP_RVP_NOCODEC_CI
in  testcase check-suspend-resume-50 , ipc timed out when system resume in COMP_NEW phase

lgirdwood, 2021-09-15:
@XiaoyunWu6666 is this still multicore, you changed the issue title ?

XiaoyunWu6666, 2021-09-16:
> @XiaoyunWu6666 is this still multicore, you changed the issue title ?

hi @lgirdwood  , this is still multicore . I changed the issue title just to make it consistent with model names in CI (TGLU_RVP_NOCODEC_CI uses multicore topology).
But if it causes any confuse , I can modify it back.

keyonjie, 2021-12-06:
suppose this should already be fixed with the latest main or stable-v2.0 branch?

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang