thesofproject/sof #4553: [BUG]ipc timed out when CTX_SAVE on TGLU_RVP_NOCODEC_MULTICORE

XiaoyunWu6666, 2021-07-27:
**Describe the bug**
inner daily 5515?model=TGLU_RVP_NOCODEC_MULTICORE&testcase=check-suspend-resume-50
ipc timed out when CTX_SAVE
**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-suspend-resume.sh -l 50
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 2113dc7f
SOF Branch: main
SOF Commit: 57ee04f2d931
 Topology: sof-tgl-nocodec-ci.tplg 
Platform:TGLU_RVP_NOCODEC_MULTICORE

**Screenshots or console output**
[dmesg]
```
[ 5570.205582] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x20140000 successful
[ 5570.205607] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 5570.707207] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[ 5570.707212] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[ 5570.707255] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[ 5570.707799] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[ 5570.707803] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 5570.707893] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[ 5570.709008] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Current DSP power state: D3
[ 5570.779659] kernel: ACPI: EC: interrupt blocked
[ 5575.517086] kernel: ACPI: EC: interrupt unblocked
```

XiaoyunWu6666, 2021-07-27:
I run this directly after rebooting and got a FAILED at the 259/500 rounds (really hard to reproduce)
```
[ 4631.281214] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 4631.281219] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 4631.369878] usb 3-4: cannot get connectors status: req = 0x81, wValue = 0x700, wIndex = 0xa00, type = 0
[ 4631.475478] nvme nvme0: 8/0/0 default/read/poll queues
[ 4631.706178] atkbd serio0: Failed to deactivate keyboard on isa0060/serio0
[ 4631.781291] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4631.781344] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4631.784906] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x50030000 size 124
[ 4631.784915] sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 4631.784918] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 4631.784931] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 4631.784935] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 4631.784958] sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 4631.784962] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 4631.784964] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 4631.784983] sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 4631.785319] sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 4631.785323] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 4631.785328] sof-audio-pci-intel-tgl 0000:00:1f.3: error: set/get ctrl ipc comp 27
[ 4631.785331] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed kcontrol value set for widget: 27
[ 4631.785334] sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail to set up kcontrols for widget SMART_AMP5.0
[ 4631.785337] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore kcontrols for widget SMART_AMP5.0
[ 4631.785345] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 4631.891926] asix 3-8.1:1.0 enx000ec667532b: Link is Down
[ 4632.096667] atkbd serio0: Failed to enable keyboard on isa0060/serio0
[ 4632.281238] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 4632.281284] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 4632.288881] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30020000 size 12
[ 4632.288894] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free widget SMART_AMP5.0
[ 4632.288898] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore pipeline after resume -110
[ 4632.288901] PM: dpm_run_callback(): pci_pm_resume+0x0/0x80 returns -110
[ 4632.288915] sof-audio-pci-intel-tgl 0000:00:1f.3: PM: failed to resume async: error -110
```

lyakh, 2021-07-27:
@XiaoyunWu6666 can we figure out what core tge failing IPC was sent to? We either need a longer kernel log before the failure or a firmware trace, ideally both :-) Could you attach files instead of pasting?

lgirdwood, 2021-07-27:
Long term we should be able to IPC flood ping to the other cores.

XiaoyunWu6666, 2021-07-28:
@lyakh I tried that on ubuntu@sh-tglu-rvp-nocodec-multicore-01 again and it failed at the 219 rounds
```
2021-07-28 07:25:39 UTC [INFO] ===== Round(220/500) =====
2021-07-28 07:25:39 UTC [COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: wakeup from "mem" using /dev/rtc0 at Wed Jul 28 07:25:45 2021
2021-07-28 07:25:46 UTC [COMMAND] sleep for 5
2021-07-28 07:25:51 UTC [INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1627457134"
2021-07-28 07:25:52 UTC [ERROR] Caught kernel log error
===========================>>
[ 3057.509782] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out fize 108
[ 3057.509796] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump-----
[ 3057.509810] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts ctl 0xc0000001 rirb 00
[ 3057.509815] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0s 0x00000000
[ 3057.509837] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x0tus 0x00000000 mask 0x00000003
[ 3057.509840] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump---
[ 3057.509843] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump-----
[ 3057.509862] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - c
[ 3057.510190] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected faulace 0x00004000
[ 3057.510194] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump---
[ 3057.510198] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: set/get ctrl ip
[ 3057.510202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed kcontrolwidget: 15
[ 3057.510206] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail to set up idget PGA3.0
[ 3057.510209] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restor widget PGA3.0
[ 3058.013732] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out fize 12
[ 3058.013745] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free
[ 3058.013749] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restoer resume -110
[ 3058.517741] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out fize 76
<<===========================

```
[dmesg.log.txt](https://github.com/thesofproject/sof/files/6891782/dmesg.log.txt)

keyonjie, 2021-08-03:
maybe @ranj063 's work on dynamic pipeline + multi-core can help?
https://github.com/thesofproject/linux/pull/3019

lgirdwood, 2021-08-03:
@mengdonglin @keyonjie I'm going to ignore multicore issues atm until @ranj063 has merged her fixes. No point in duplicating the effort atm.

keyonjie, 2021-12-06:
suppose this should already be fixed with the latest main or stable-v2.0 branch?

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang