thesofproject/sof #4560: [BUG]underrun due to no data available when multiple-pipeline on CML_HEL_RT5682

XiaoyunWu6666, 2021-07-28:
**Describe the bug**
in inner daily  5539?model=CML_HEL_RT5682&testcase=multiple-pipeline-all
underrun happened when multiple pipeline and xrun recovery failed
can reproduce it
**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt1011-rt5682.tplg ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 85bfa34a
SOF Branch: main
SOF Commit: 9a7a5ce17dcb
 Topology:sof-cml-rt1011-rt5682.tplg 
Platform:CML_HEL_RT5682

**Screenshots or console output**
[etrace]
```
[      412591.494022] (      412591.500000) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      412670.660685] (          79.166664) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      412758.473182] (          87.812500) c0 dma-trace              src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
[      412772.223181] (          13.749999) c0 dma-trace                   src/audio/dai.c:886  ERROR dai_report_xrun(): underrun due to no data available
[      412786.025264] (          13.802083) c0 dma-trace                   src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 4, source->avail = 192, copy_bytes = 0
[      412799.566930] (          13.541666) c0 dma-trace          ....../pipeline-stream.c:307  ERROR pipeline_trigger_run(): ret = -22, host->comp.id = 0, cmd = 8
[      412812.900263] (          13.333333) c0 dma-trace          ......../pipeline-xrun.c:159  ERROR pipeline_xrun(): Pipelines notification about XRUN failed, ret = -22
[      412842.796095] (          29.895832) c0 dma-trace          ....../pipeline-stream.c:194  ERROR pipeline_copy(): ret = -61, start->comp.id = 4, dir = 1
[      412855.921095] (          13.124999) c0 dma-trace          ..../pipeline-schedule.c:95   ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
[      412930.191925] (          74.270828) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      413212.691914] (         282.500000) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      413291.025244] (          78.333328) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      413367.275241] (          76.250000) c0 dma-trace          ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
```

ranj063, 2021-07-29:
the problem was introduced by https://github.com/thesofproject/sof/commit/9a7a5ce17dcb 

@lyakh FYI

lyakh, 2021-07-29:
@XiaoyunWu6666 can I get a full dmesg, please?

plbossart, 2021-07-29:
https://sof-ci.sh.intel.com/#/result/planresultdetail/5568?model=CML_HEL_RT5682&testcase=multiple-pipeline-all

[dmesg.txt](https://github.com/thesofproject/sof/files/6901155/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/6901163/slogger.txt)

lgirdwood, 2021-07-29:
@lyakh from the logs above pasted by @plbossart it looks like we get stuck somewhere for ~10ms after starting SSP1
```
[      134886.192557] (          33.958332) c0 ll-schedule        ./schedule/ll_schedule.c:222  INFO perf ll_work peak plat 6489 cpu 108072
[      135939.109182] (        1052.916626) c0 dma-trace                 src/trace/trace.c:137  INFO Suppressed 1 similar messages: perf comp_copy peak plat %d cpu %d
[      135950.619598] (          11.510416) c0 dai          1.4   ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 291 cpu 4848
[      135975.463347] (          24.843748) c0 host         5.22  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 229 cpu 3800
[      136026.921678] (          51.458332) c0 pga          5.23  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 768 cpu 12816
[      136066.452927] (          39.531250) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 218 cpu 3616
[      136142.598757] (          76.145828) c0 ll-schedule        ./schedule/ll_schedule.c:222  INFO perf ll_work peak plat 6613 cpu 110192
[      137129.838301] (         987.239563) c0 dma-trace                 src/trace/trace.c:137  INFO Suppressed 1 similar messages: perf comp_copy peak plat %d cpu %d
[      137141.192467] (          11.354166) c0 host         1.0   ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 225 cpu 3732
[      138458.171581] (        1316.979126) c0 host         5.22  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 229 cpu 3804
[      138542.598661] (          84.427078) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 219 cpu 3636
[      139779.786112] (        1237.187500) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 224 cpu 3712
[      143383.588052] (        3603.802002) c0 host         1.0   ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 231 cpu 3828
[      144707.858833] (        1324.270752) c0 host         5.22  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 232 cpu 3848
[      146030.202531] (        1322.343750) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 227 cpu 3764
[      170984.889039] (       24954.685547) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 228 cpu 3776
[      193382.127732] (       22397.238281) c0 host         1.0   ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 231 cpu 3832
[      208485.304216] (       15103.176758) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 228 cpu 3780
[      283484.832485] (       74999.531250) c0 host         6.28  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 228 cpu 3784
[      332183.997217] (       48699.164062) c0 host         5.22  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 232 cpu 3852
[      404424.098513] (       72240.101562) c0 pipe         7.39  ....../pipeline-stream.c:252  INFO pipe trigger cmd 1
[      404438.890179] (          14.791666) c0 ll-schedule        ./schedule/ll_schedule.c:390  INFO task add 0xbe080400 pipe-task 
[      404449.983929] (          11.093750) c0 ll-schedule        ./schedule/ll_schedule.c:394  INFO task params pri 0 flags 0 start 0 period 1000
[      404466.702678] (          16.718750) c0 ll-schedule        ./schedule/ll_schedule.c:306  INFO new added task->start 10366793 at 10365625
[      404477.692261] (          10.989583) c0 ll-schedule        ./schedule/ll_schedule.c:309  INFO num_tasks 7 total_num_tasks 7
[      404792.535998] (         314.843750) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:1059 INFO ssp_trigger() cmd 1
[      404804.983914] (          12.447916) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:876  INFO ssp_pre_start()
[      404817.223497] (          12.239583) c0 mn                 ./drivers/intel/ssp/mn.c:180  INFO MCLK 24000000, source = 0
[      404828.525580] (          11.302083) c0 mn                 ./drivers/intel/ssp/mn.c:258  INFO mclk_rate 24000000, mclk_source_clock 0
[      404839.358913] (          10.833333) c0 mn                 ./drivers/intel/ssp/mn.c:220  INFO mclk_id 0 mdivr_val 1
[      404853.056829] (          13.697916) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:215  INFO ssp_bclk_prepare_enable(): sscr0 = 0xc3d00437
[      404865.348495] (          12.291666) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:970  INFO ssp_start(): SSE set for SSP1
[      404876.598495] (          11.250000) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:974  INFO ssp_start()
[      417456.181328] (       12579.583008) c0 ll-schedule        ./schedule/ll_schedule.c:222  INFO perf ll_work peak plat 247760 cpu 4129748 <<<<<<<<<<<<<<<<<<<<< HERE
[      417561.650074] (         105.468742) c0 sa                          src/lib/agent.c:65   INFO perf sys_load peak plat 249564 cpu 4145159
[      417572.743824] (          11.093750) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 249564
[      417629.566738] (          56.822914) c0 hda-dma            ..../intel/hda/hda-dma.c:861  ERROR hda_dma_link_check_xrun(): underrun detected
[      417654.618821] (          25.052082) c0 dai          4.20  ....../pipeline-stream.c:205  INFO perf comp_copy peak plat 541 cpu 9012
[      417705.868819] (          51.249996) c0 dw-dma                 src/drivers/dw/dma.c:1069 INFO dw_dma_free_data_size() size is 0!
[      417721.545901] (          15.677083) c0 dw-dma                 src/drivers/dw/dma.c:1102 ERROR dw_dma_get_data_size(): xrun detected
```
This also causes the agent to be late, so it seems we are spinning somewhere with IRQs OFF.

plbossart, 2021-07-29:
One possible reason why we only see this problem on Helios is the topology:

[image]

The code now explicitly does

````
list_for_item(tlist, &ctx->pipelines) {
			p = container_of(tlist, struct pipeline, list);
			if (pipeline_is_timer_driven(p)) {
				/* Use the first of connected pipelines to trigger */
				if (cmd >= 0) {
					p->trigger.cmd = cmd;
					p->trigger.host = ppl_data->start;
					cmd = -EINVAL;
				}
			} else {
				p->xrun_bytes = 0;
				p->status = COMP_STATE_ACTIVE;
			}
````

Is this a valid assumption when there are split/joins in pipelines?

lyakh, 2021-07-30:
@plbossart the loop topology is a great tip, thanks! Wondering what other devices have similar ones and particularly which of them I have available for testing...

bardliao, 2021-07-30:
I don't know if this helps. I print ` p->trigger.cmd` in `pipeline_task`
```patch
--- a/src/audio/pipeline/pipeline-schedule.c
+++ b/src/audio/pipeline/pipeline-schedule.c
@@ -55,6 +55,7 @@ static enum task_state pipeline_task(void *arg)
                        return SOF_TASK_STATE_COMPLETED;
        }

+       pipe_warn(p, "pipeline_task(): bard: p->trigger.cmd %d", p->trigger.cmd);
        if (p->trigger.cmd >= 0) {
                /* First pipeline task run for either START or RELEASE */
                struct sof_ipc_reply reply = {
```
And it seems no `WARN pipeline_task(): bard: p->trigger.cmd 1` before `ERROR hda_dma_link_check_xrun(): underrun detected`
[trace.txt](https://github.com/thesofproject/sof/files/6905806/trace.txt)

plbossart, 2021-07-30:
Indeed it seems this loop needs to be handled, but it's unlikely to be the cause of this issue: we don't actually test the 16kHz input !!

This is a gap in our coverage, so we need to change the tests and be ready for new bugs to pop-up...

lyakh, 2021-07-30:
@plbossart @lgirdwood I think I have a trace. What that my patch does is move "trigger START" to task context. What that means, is that when the host sends a "trigger START" the firmware will get the IPC, and schedule its processing in the pipeline's LL scheduling task. Next scheduling tick the scheduler begins scheduling tasks. I.e. it runs a loop over its list of scheduled tasks. One of those tasks is executing the "trigger START" from the last IPC. So far so good. But I've just caught a case where this START processing for pipeline 7 (SSP1 out) took more than 10ms on Helios. Everybody dies (TM)

lyakh, 2021-07-30:
we have a problem with Helios speakers:
```
[    33707182.879347] (          59.791664) c0 dai          1.4            src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 96, DAI took 1852
[    33739584.544726] (          11.145833) c0 dai          5.26           src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 59, DAI took 24
[    36617182.711631] (          59.791664) c0 dai          1.4            src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 97, DAI took 1850
[    36634584.429689] (          11.197916) c0 dai          6.32           src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 59, DAI took 24
[    39490932.909938] (          59.791664) c0 dai          1.4            src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 96, DAI took 1861
[    39813500.397120] (       12562.030273) c0 dai          7.38           src/audio/dai.c:752  INFO dai_comp_trigger_internal(), DMA took 94, DAI took 241977
```
i.e. all other DAIs initialise in under 100us, and the speaker DAI takes more than 10ms

lgirdwood, 2021-07-30:
@lyakh lets resolve the 10ms SSP start on Monday and then we can reapply this one alongside the pause/stop update.
I've no idea why SSP start would take 10ms to start.....

lyakh, 2021-08-02:
> @lyakh lets resolve the 10ms SSP start on Monday and then we can reapply this one alongside the pause/stop update.
> I've no idea why SSP start would take 10ms to start.....

@lgirdwood I think I know why, because of this line
```
define(`SET_SSP1_CONFIG_DATA', `SSP, 1, 24, 0, 0, 10')',
`
```
in platform/intel/cml.m4 which sets `SOF_TKN_INTEL_SSP_BCLK_DELAY` for SSP1 to 10 which then is used in
```
		wait_delay(clock_ms_to_ticks(PLATFORM_DEFAULT_CLOCK,
					     ssp->params.bclk_delay));
```
in ssp.c. Any idea why that is needed?

bardliao, 2021-08-02:
> @lgirdwood I think I know why, because of this line
>
> ```
> define(`SET_SSP1_CONFIG_DATA', `SSP, 1, 24, 0, 0, 10')',
> `
> ```
>
> in platform/intel/cml.m4 which sets `SOF_TKN_INTEL_SSP_BCLK_DELAY` for SSP1 [...]

Some codecs need BCLK enabled while configuring their registers.  With SET_SSP1_CONFIG_DATA sets BCLK will start a provided delay before SSP data start and the codecs can configure their registers in the time. Using SSP_CC_BCLK_ES (BCLK early start) may also work now. See 
https://github.com/thesofproject/sof/pull/4391/commits/32d04e9a47a19131ee627cac6f9905d16acfaaf1

lgirdwood, 2021-08-02:
Ok, this is broken. @lyakh so this will need reflected in the LL PR. i.e. we will need to
1. Check for any early DAI enablement. We could have an new API for this e.g ```early_trigger()``` and call this at LL tick 0. (this starts the SSP BCLK)
2. LL tick 10 then calls trigger()
3. LL tick 11 then does the first copy.

This means we can remove the busy wait in ssp.c

plbossart, 2021-08-02:
Not following @lgirdwood 

Yes there is a 10ms delay that looks like a copy-paste mistake, it's not really necessary here.

But is it wrong? Is there any assumption that the trigger completes immediately, and if yes what did change in @lyakh's update?
We have a couple of wait_delay() calls in the SSP, and it'd be useful to clarify the rules/expectations on what a wait is tolerated.

XiaoyunWu6666, 2021-08-03:
Close it now since relevant feature was reverted in #4578

lyakh, 2021-08-03:
> Not following @lgirdwood
> 
> Yes there is a 10ms delay that looks like a copy-paste mistake, it's not really necessary here.
> 
> But is it wrong? Is there any assumption that the trigger completes immediately, and if yes what did change in @lyakh's update?
> We have a couple of wait_delay() calls in the SSP, and it'd be useful to clarify the rules/expectations on when a wait is tolerated.

@plbossart my PR moved the trigger:START processing from the IPC thread context to the pipeline task context. And when you busy-loop for 10ms in a LL task you block all other LL tasks for that time.

lyakh, 2021-08-03:
> Ok, this is broken. @lyakh so this will need reflected in the LL PR. i.e. we will need to
>
>     1. Check for any early DAI enablement. We could have an new API for this e.g `early_trigger()` and call [...]

@lgirdwood yes, this is a possibility. And you're proposing to keep the order of component triggering otherwise the same? Maybe it would be more like a late trigger, than an early trigger. I.e. in the loop where all START triggers are called, a component can return a status, that tells the caller, that the component needs a delay. The loop would then break. 10 ticks later the `trigger_late()` callback from the delayed component would be called, after which the `pipeline_for_each_comp()` loop would resume from the next component? Hopefully no components need more than one delay...

lgirdwood, 2021-08-03:
> Not following @lgirdwood
> 
> Yes there is a 10ms delay that looks like a copy-paste mistake, it's not really necessary here.
> 
> But is it wrong? Is there any assumption that the trigger completes immediately, and if yes what did change in @lyakh's update?

So the delay is fine as it's needed by the codec, but its how we are delaying that is an issue. @lyakh will update FW so we can do any codec required delay and add this to pipeline state machine i.e. we will probably have a 2 internal TRIGGER_START commands that can be used internally (no change for host).

> We have a couple of wait_delay() calls in the SSP, and it'd be useful to clarify the rules/expectations on when a wait is tolerated.

I think a rough rule of thumb is probably less than 100uS in total for DAI state transitions (since copy() is not performed on the same LL tick)

lyakh, 2021-09-06:
@XiaoyunWu6666 @lgirdwood @plbossart Can this be closed now that #4559 has been merged?

XiaoyunWu6666, 2021-09-06:
@lyakh hi , for me it's good to close since it haven't happened in daily tests for almost more than two weeks

keyonjie, 2021-09-06:
I am still seeing this with my debugging today, but not sure if it is introduced by my local change. 

Can we leave this open for some more days?

XiaoyunWu6666, 2022-01-24:
haven't seen it in CI for over 3 months

closed 2022-01-24 by XiaoyunWu6666