thesofproject/sof #4563: [BUG] Glitches in the whole file when pipelines with multicore connected to HDA

kkarask, 2021-07-28:
**Describe the bug**
The whole file with glitches was found in test 17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch

**Topology**
Topology: playback (core 0) and capture (core 1) pipelines connected to HDA with link loopback

```
                    pipe_plb core 0
                    +--------------------+
    +----------+    | +------+   +-----+ |   +-----+
    | Host plb |----->|Buffer|-->| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +-----+
                    +--------------------+    |  |
                                              |  |
                    pipe_cap core 1           |  |
                    +--------------------+    |  |
    +----------+    | +------+   +-----+ |   +-----+
    | Host cap |<-----|Buffer|<--| Dai |---->|codec|
    +----------+    | +------+   +-----+ |   +-----+
                    +--------------------+
```

**To Reproduce**
Run test 17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch with Diagnostic driver with parameter:
--playback_iterations=75

**Reproduction Rate**
1/75

**Environment**
1) Firmware branch name and commit
    * SOF: Branch: main, Hash: 9648fb65a5b8b86707e0a160efa992f3f3c97f59
    
2) Name of the platform(s) on which the bug is observed.
    * Platform: WHL RVP

**Screenshots or console output**
[image]

[WHL_17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch.zip](https://github.com/thesofproject/sof/files/6891551/WHL_17_03_TestMultiCoreMultiKernelHda48000hz16b16b4ch.zip)

lgirdwood, 2021-08-03:
@kkarask lets re-test this after @ranj063 multicore fixes are merged as she has found and fixed a few isues that will cause the above glitches.

wszypelt, 2021-08-11:
I have re-tested over 800 iterations. All iterations pass.

closed 2021-08-11 by wszypelt