thesofproject/sof #4568: [BUG] [TGL] PCM 99 capture fails with the IGO integrated firmware with Google Assistant "OK Google" feature

RDharageswari, 2021-07-28:
**Describe the bug**
Turn on Google Assistant "OK Google" feature,
arecord -Dhw:0,99 try.wav -c4 -r48000 -fs32_le -d60 
localhost ~/dhara # Recording WAVE 'try.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1405: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 128
CHANNELS: 4
RATE: 48000
PERIOD_TIME: (21333 21334)
PERIOD_SIZE: 1024
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (85333 85334)
BUFFER_SIZE: 4096
BUFFER_BYTES: 65536
TICK_TIME: 0
While the capture works after I turn off "OK Google" feature.

**Reproduction Rate**
always

**Expected behavior**
no hw params failure

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
blocking

tgl-13 branch with IGO patches & IGO binary

dmesg logs:
   57.017745] x86/PAT: flashrom:4591 conflicting memory types fe000000-100000000 uncached-minus<->uncached
[   57.020237] x86/PAT: reserve_memtype failed [mem 0xfe000000-0xffffffff], track uncached-minus, req uncached-minus
[   71.223828] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   71.224352] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 3
[   71.224867] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -12
[   71.225532]  DMIC: ASoC: hw_params FE failed -12
localhost ~ #

[fw_logs.txt](https://github.com/thesofproject/sof/files/6896843/fw_logs.txt)

keyonjie, 2021-07-29:
looks the buffer heap is used up here, it is understandable as both KPB history buffer and this IGO are requires of quite big buffer, need to debug and fine tune this part:
```
[    37924481.461767] (          26.145832) c1 buffer                   src/audio/buffer.c:60   ERROR buffer_alloc(): could not alloc size = 49152 bytes of type = 32
```

fuyuntsuo, 2021-07-30:
Add more information: `igo` is using the Buffer Zone as its working buffer (<40KB IIRC. code [here](https://github.com/thesofproject/sof/blob/53718867df8501f745795e674ca92cf1c16443e5/src/audio/igo_nr/igo_nr.c#L308)) but `kpb` is using Run-time Zone (49KB in the log. code [here])

Are the sizes of these two zone supposed to be determined in linking time instead of run-time?

keyonjie, 2021-07-30:
> Add more information:
> `igo` is using the Buffer Zone as its working buffer (<40KB IIRC. code [here](https://github.com/thesofproject/sof/blob/53718867df8501f745795e674ca92cf1c16443e5/src/audio/igo_nr/igo_nr.c#L308)), but
> `kpb` is using the Run-time Zone (49KB in the log. code [here](https://github.com/thesofproject/sof/blob/53718867df8501f745795e674ca92cf1c16443e5/src/audio/kpb.c#L150))
> 
> The size of these two zones are supposed to be determined in linking time. These two components are not sharing the same heap.

the 'kpb' history buffer could requires up to e.g. 384KB history buffer at buffer zone.

keyonjie, 2021-07-30:
@fuyuntsuo @RDharageswari please help to check if the PR https://github.com/thesofproject/sof/pull/4575 help for this issue.

fuyuntsuo, 2021-07-30:
@keyonjie 
I've applied the PR to private repo `tgl-013-drop-stable` but it fails during compiling due to memory map change. 
[log.txt](https://github.com/thesofproject/sof/files/6908523/log.txt)

Actually I have tried to build the IGO with Github Main branch but fails neither. Seems like after the commit dcd4f34c63de0e4e56228591007d796c2764bf2b we don't have enough memory for IGO any more. 
[dcd4f3.txt](https://github.com/thesofproject/sof/files/6908617/dcd4f3.txt) 
[0001-Disable-GWD-DSM-and-BT.zip](https://github.com/thesofproject/sof/files/6908620/0001-Disable-GWD-DSM-and-BT.zip)

yongzhi1, 2021-07-30:
Hi @keyonjie, thanks to PR #4575, the reported issue should have been resolved. That said, I have not done extensive testing, only verified the reported scenario, ie. WoV + DMIC recording.

lgirdwood, 2021-07-30:
@yongzhi1 @RDharageswari who will be able to do full test (and when) ? @mwasko fyi.
@keyonjie @slawblauciak good efforts in debugging this issue !

cujomalainey, 2021-09-29:
Ping on this bug, has this issue been fixed?

bkokoszx, 2021-10-01:
@sathya-nujella 
I think we can close it now?

cujomalainey, 2021-10-01:
No activity for 3m, I am going to close this, feel free to reopen if i am incorrect.

closed 2021-10-01 by cujomalainey