thesofproject/sof #4579: [BUG]ipc timed out when check-signal-stop-start-playback on APL_UP2_NOCODEC_ZEPHYR

XiaoyunWu6666, 2021-08-02:
**Describe the bug**
in inner daily 5635?model=APL_UP2_NOCODEC_ZEPHYR&testcase=check-signal-stop-start-playback-50
after run stop/start test on PCM:Port5 Deep Buffer,playback, ipc timed out 
cannot reproduce though (with -c 500)
**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg ~/sof-test/test-case/check-signal-stop-start.sh -m playback -c 50
**Reproduction Rate**
not sure

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 32657609
SOF Branch: main
SOF Commit: d38c3c391291
 Topology: /lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg 
 Platform:  APL_UP2_NOCODEC_ZEPHYR

**Screenshots or console output**
[console]
```
2021-08-01 22:15:05 UTC [REMOTE_INFO] Testing: run stop/start test on PCM:Port5 Deep Buffer,playback. Interval time: 0.5
2021-08-01 22:15:05 UTC [REMOTE_COMMAND] aplay -Dhw:0,3 -r 48000 -c 2 -f S16_LE /dev/zero -q &
……
2021-08-01 22:15:55 UTC [REMOTE_INFO] Stop/start count: 47
underrun!!! (at least 225.761 ms long)
2021-08-01 22:15:56 UTC [REMOTE_INFO] Stop/start count: 48
underrun!!! (at least 225.135 ms long)
2021-08-01 22:15:57 UTC [REMOTE_INFO] Stop/start count: 49
underrun!!! (at least 228.072 ms long)
2021-08-01 22:15:58 UTC [REMOTE_INFO] Stop/start count: 50
underrun!!! (at least 225.431 ms long)
2021-08-01 22:15:59 UTC [REMOTE_COMMAND] kill process: kill -9 17668
declare -- cmd="journalctl_cmd --since=@1627855936"
2021-08-01 22:16:00 UTC [ERROR] Caught kernel log error
===========================>>
[ 2940.036244] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: ipc timed out for 0x40010000 size 76
[ 2940.036435] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump start ]------------
[ 2940.036448] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[ 2940.036459] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 2940.036470] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 2940.036479] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ IPC dump end ]------------
[ 2940.036487] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump start ]------------
[ 2940.036497] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: status: fw entered - code 00000005
[ 2940.036547] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[ 2940.036556] kernel: sof-audio-pci-intel-apl 0000:00:0e.0: ------------[ DSP dump end ]------------
<<===========================
```
[dmesg]
[dmesg.txt](https://github.com/thesofproject/sof/files/6914566/dmesg.txt)

kv2019i, 2021-08-10:
Potential bugfix https://github.com/thesofproject/sof/pull/4618

lgirdwood, 2021-08-17:
@XiaoyunWu6666 can we close ?

YongAnLu00, 2021-08-18:
This test case failed in our daily test because some previous tests had failed and the hardware was in an abnormal state. I ran this test case independently with 600 stop-start iterations, and no error was found. So maybe we can close it.

closed 2021-08-18 by YongAnLu00