thesofproject/sof #4583: @plbossart I think I found the issue. I based my implementation on existing broken code.

dbaluta, 2021-08-02:
@plbossart I think I found the issue. I based my implementation on existing broken code.

So, for example the idion:

```
+ifdef(`CHANNELS_MIN',`',
+`define(CHANNELS_MIN, `2')')

```

is wrong. Because if CHANNELS_MIN is not define it will set it to 2 (CHANNELS_MIN = 2) but the side effect is that from now on the CHANNELS_MIN will stay defined. 

By luck this didn't cause any trouble by now. I think I have found the correct in my patch but we should also fix the rest of the patches. Will se if I have time for this in the following days until then I will log a bug.

_Originally posted by @dbaluta in https://github.com/thesofproject/sof/issues/4582#issuecomment-891220315_

lgirdwood, 2021-08-03:
@dbaluta can this be closed now PR is merged ?

dbaluta, 2021-08-03:
@mengdonglin this PR is for all other instances of CHANNELS_MIN. It is a bug that it isn't detected because of some luck. I will try to take care of this by EOW.

lgirdwood, 2021-08-16:
@dbaluta can we close ?

closed 2021-08-17 by dbaluta