thesofproject/sof #4586: [BUG]ipc timed out when SET_DATA on TGLU_RVP_NOCODEC_MULTICORE

XiaoyunWu6666, 2021-08-03:
**Describe the bug**
Found in inner daily on model TGLU_RVP_NOCODEC_MULTICORE in testcase check-runtime-pm-status-15.
According to console, when check runtime pm status with 'aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q' (Iteration 7/15) , tried to stop the aplay (on pipeline#1) but connection timed out.
According to dmesg,ipc timed out when kernel sending SET DATA to firmware

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-runtime-pm-status.sh -l 15

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 32657609
SOF Branch: main
SOF Commit: d38c3c391291
Topology: sof-tgl-nocodec-ci.tplg (pipeline  3,4 on core 3 ; pipeline 8 on core 2)
Platform:TGLU_RVP_NOCODEC_CI

**Screenshots or console output**
[console]
```
2021-08-01 22:51:36 UTC N ===== Iteration 6 of 15 for Port0 =====
2021-08-01 22:51:36 UTC N aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q
2021-08-01 22:51:38 UTC N runtime status: active
2021-08-01 22:51:38 UTC N kill process: kill -9 60700
2021-08-01 22:51:38 UTC N aplay killed
2021-08-01 22:51:38 UTC N wait dsp power status to become suspended
2021-08-01 22:51:41 UTC N dsp suspended in 3s
2021-08-01 22:51:41 UTC N runtime status: suspended
declare -- cmd="journalctl_cmd --since=@1627858291"
2021-08-01 22:51:42 UTC N ===== Iteration 7 of 15 for Port0 =====
2021-08-01 22:51:42 UTC N aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q
error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)
aplay: main:830: audio open error: Connection timed out
/home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh: line 98: kill: (60889) - No such process
2021-08-01 22:51:44 UTC [REMOTE_ERROR] aplay process for pcm Port0 is not alive
```
[dmesg]
```
[ 5355.454480] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget SSP1.IN setup complete
[ 5355.454488] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[ 5355.455269] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[ 5355.455279] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.4.SSP1.IN setup complete
[ 5355.455289] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 5355.455660] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 5355.455671] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5355.455879] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5355.455883] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PCM2P setup complete
[ 5355.455897] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5355.456138] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5355.456161] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.456193] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5355.456215] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5355.456444] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.456458] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x50030000: GLB_COMP_MSG: SET_DATA
[ 5355.956077] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5355.956110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5355.963999] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x50030000 size 124
[ 5355.964016] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5355.964021] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5355.964037] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 5355.964044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5355.964059] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 5355.964066] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5355.964070] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5355.964083] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: status: fw entered - code 00000005
[ 5355.964386] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5355.964391] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5355.964404] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: set/get ctrl ipc comp 27
[ 5355.964411] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed kcontrol value set for widget: 27
[ 5355.964416] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail to set up kcontrols for widget SMART_AMP5.0
[ 5355.964421] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore kcontrols for widget SMART_AMP5.0
[ 5355.964433] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 5355.964464] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: trace IO error
[ 5356.456041] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5356.456082] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5356.468221] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc timed out for 0x30020000 size 12
[ 5356.468242] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to free widget SMART_AMP5.0
[ 5356.468248] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to restore pipeline after resume -110
[ 5356.468259] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -110
[ 5356.468271] kernel:  NoCodec-0: soc_pcm_open() failed (-110)
[ 5356.468278] kernel:  Port0: ASoC: dpcm_be_dai_startup() failed at NoCodec-0 (-110)
[ 5356.468287] kernel:  Port0: dpcm_fe_dai_startup() failed (-110)
[ 5358.130979] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131003] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5358.131204] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.357704] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 5359.357717] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
[ 5359.357826] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.357873] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.358446] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.358451] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Copying cached debugfs data
[ 5359.616193] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: enabling device failed: -22
[ 5359.616202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: fail in sof_ipc_trace_update_filter -22
```

XiaoyunWu6666, 2021-08-09:
duplicate as #4552 ,all happened when runtime resume

keyonjie, 2021-12-06:
suppose this should already be fixed with the latest main or stable-v2.0 branch?

keqiaozhang, 2021-12-13:
Confirmed that this issue is fixed with latest main and stable v2.0 branch.

closed 2021-12-13 by keqiaozhang