thesofproject/sof #4595: [BUG]DSP panic happened after sleep finished and boot DSP firmware, on CML_SKU0983_SDW

XiaoyunWu6666, 2021-08-04:
**Describe the bug**
When running arecord with -Dhw:0,1 in the background ,DSP panic happened after sleep finished  and boot DSP firmware, on CML_SKU0983_SDW
Found in inner daily 5685 , on model CML_SKU0983_SDW, in testcase check-suspend-resume-with-capture-5.
Haven’t reproduced it yet.

**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m capture
**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 16d5c452
SOF Branch: main
SOF Commit: 1f641729cd38
TPLG: TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg
PLATFORM: CML_SKU0983_SDW

**Screenshots or console output**
[console]
when run ' arecord -Dhw:0,1 -r 48000 -c 2 -f S16_LE /dev/null -q' in the background
```
2021-08-03 23:17:00 UTC Sub-Test: [REMOTE_INFO] ===== Round(4/5) =====
2021-08-03 23:17:00 UTC Sub-Test: [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Aug  3 23:17:06 2021
2021-08-03 23:17:07 UTC Sub-Test: [REMOTE_COMMAND] sleep for 5
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1628032615"
2021-08-03 23:17:12 UTC [ERROR] Caught kernel log error
===========================>>
[ 6340.383451] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 6340.383465] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6340.383478] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 6340.383499] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 6340.383505] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
<<===========================
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_ERROR] Caught error in kernel log
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_ERROR]  die()  @  /home/ubuntu/sof-test/test-case/../case-lib/lib.sh
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-suspend-resume.sh:91
2021-08-03 23:17:12 UTC Sub-Test: [REMOTE_INFO] Test Result: FAIL!
2021-08-03 23:17:12 UTC [REMOTE_ERROR] suspend resume failed
```
[dmesg]
```
[ 6340.369411] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: loading firmware
[ 6340.369430] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[ 6340.369455] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[ 6340.369506] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[ 6340.369514] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[ 6340.369519] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: booting DSP firmware
[ 6340.369848] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[ 6340.370091] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[ 6340.370117] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dsp_stream_setup_bdl: period_bytes:0x0
[ 6340.370122] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: hda_dsp_stream_setup_bdl: periods:1
[ 6340.370302] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[ 6340.370327] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 6340.370411] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f0f successful
[ 6340.370523] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f00 successful
[ 6340.370548] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = f
[ 6340.371039] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask f
[ 6340.371181] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0000 successful
[ 6340.371205] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = 1
[ 6340.371256] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 6340.383270] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
[ 6340.383304] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0e0e successful
[ 6340.383326] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010e0e successful
[ 6340.383334] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask e
[ 6340.383451] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 6340.383457] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 6340.383465] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6340.383478] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 6340.383499] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 6340.383505] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 6340.683394] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5 timedout
[ 6340.683430] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010f0f successful
[ 6340.683459] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
```
[sof-logger]
[slogger.txt](https://github.com/thesofproject/sof/files/6928429/slogger.txt)

lgirdwood, 2021-08-04:
FW panic() or assert() here,  looks like its maybe during boot according to dmeg (cant see topology or streams), fw log only shows the previous boot AFAICT.

plbossart, 2021-08-04:
Test 5585 included the kernel changes for multi-core and dynamic pipelines, but IIRC this was tested by @bardliao and @ranj063

ranj063, 2021-08-04:
http://sof-ci.sh.intel.com/#/result/planresultdetail/5449 shows a similar failure with the suspend/resume test. I think this issue is a duplicate of https://github.com/thesofproject/sof/issues/4538

plbossart, 2021-08-04:
Can't reproduce this, even with 50 iterations:

````TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 50 -m capture
````

This also doesn't seem like a critical issue, this happened in iteration 0 but iteration 1 succeeded.
I am also not clear how a dsp panic can happen at such an ealy stage.

````
[ 6340.370302] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[ 6340.370327] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 6340.370411] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f0f successful
[ 6340.370523] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f00 successful
[ 6340.370548] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = f
[ 6340.371039] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask f
[ 6340.371181] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0000 successful
[ 6340.371205] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = 1
[ 6340.371256] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 6340.383270] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
[ 6340.383304] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0e0e successful
[ 6340.383326] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010e0e successful
[ 6340.383334] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask e
[ 6340.383451] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error : DSP panic!
[ 6340.383457] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 6340.383465] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6340.383478] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[ 6340.383499] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0xdead00b 0x41c0 0x0 0x0 0x0 0x1811102 0x0
[ 6340.383505] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 6340.683394] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5 timedout
[ 6340.683430] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010f0f successful
[ 6340.683459] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[ 6340.683468] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 6340.683475] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Attempting iteration 1 of Core En/ROM load...
[ 6340.683485] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 6340.683503] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f0f successful
[ 6340.683521] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0f00 successful
[ 6340.683527] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = f
[ 6340.683535] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask f
[ 6340.683555] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0000 successful
[ 6340.683561] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: unstall/run core: core_mask = 1
[ 6340.683569] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 6340.684367] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
[ 6340.684405] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f0e0e successful
[ 6340.684428] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010e0e successful
[ 6340.684437] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask e
[ 6340.760299] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5000001 successful
[ 6340.774692] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5 successful
[ 6340.774722] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x140000 successful
[ 6340.774738] kernel: sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware download successful, booting...

````

plbossart, 2021-08-04:
Agree @ranj063 we've seen this before.

I also vaguely recall a comment, maybe from @lyakh, in another issue where it wasn't clear why a dsp panic was thrown in early stages.

ranj063, 2021-08-04:
a panic can show up only as a result of an IPC RX but at this stage IPC interrupts shouldnt even be enabled. quite strange!

ranj063, 2021-08-04:
oh actually I think the panic shows up as soon as the IPC interrupts are enabled during boot. still quite strange. We should either ignore this or check for fw boot state to determine if the panic is valid or not.

lgirdwood, 2021-08-04:
> I am also not clear how a dsp panic can happen at such an ealy stage.

Its only fundamental init related stuff that can panic here, most likely some IP programming, but I cant see anything in dmesg as I think debug has been disabled (as panic dumps the stack etc)

plbossart, 2021-08-06:
I reproduced this today after 270 iterations...

[panic.tar.gz](https://github.com/thesofproject/sof/files/6947315/panic.tar.gz)

lgirdwood, 2021-09-20:
@plbossart I've asked @ujfalusi to make sure we dump FW stack and populate exception debugFW here (which driver is not doing atm as FW has not sent boot message). This should shed some light on issue as the stack trace can be run through gdb.

closed 2022-01-24 by XiaoyunWu6666