thesofproject/sof #4596: [BUG]ADLS_RVP_HDA TIMEOUT when check-kmod-load-reload

XiaoyunWu6666, 2021-08-04:
**Describe the bug**
This happened after ADL DUTs added back to inner daily tests , and check-kmod-load-reload-25 tested on ADLS_RVP_HDA.

TIMEOUT happened :
- before dsp suspended (inner daily 5657)
- OR  when check firmware load information after kmod load (inner daily 5685)
- OR  when waiting dsp to become suspended(inner daily 5595)

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg ~/sof-test/test-case/check-kmod-load-unload.sh -l 25
**Reproduction Rate**
100% almost

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 9474f300
SOF Branch: main
SOF Commit: 53718867df85
Topology: /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg
 Platform: ADLS_RVP_HDA

**Screenshots or console output**
[console]
TIMEOUT happened :
- before dsp suspended (inner daily 5657)
```
2021-08-02 22:19:14 UTC [REMOTE_INFO] ==== firmware boot complete: 7 of 25 ====
(and then TIMEOUT , no console logs )
```
- OR  when check firmware load information after kmod load (inner daily 5685)

```
……(above is los of sof_remove and sof_insert)
MODPROBE	snd_sof_pci_intel_apl		 
MODPROBE	snd_sof_pci_intel_cnl		 
MODPROBE	snd_sof_pci_intel_icl		 
MODPROBE	snd_sof_pci_intel_tgl		 
MODPROBE	snd_sof_acpi		 
MODPROBE	snd_sof_pci		 
MODPROBE	snd_usb_audio		 
2021-08-03 23:12:03 UTC [REMOTE_INFO] checking for general errors after kmod insert with sof-kernel-log-check tool
declare -- cmd="journalctl_cmd --since=@1628032317"
2021-08-03 23:12:04 UTC [REMOTE_INFO] checking if firmware is loaded successfully
2021-08-03 23:12:04 UTC Sub-Test: [REMOTE_INFO] Checking SOF Firmware load info in kernel log
(and then TIMEOUT , no console logs )
```

- OR  when waiting dsp to become suspended(inner daily 5595)
```
2021-07-29 22:19:39 UTC [REMOTE_INFO] ===== Starting iteration 13 of 25 =====
2021-07-29 22:19:39 UTC [REMOTE_INFO] wait dsp power status to become suspended
(and then TIMEOUT , no console logs )
```

XiaoyunWu6666, 2021-08-05:
I manually tried kmod-load-reload on ADLS_RVP_HDA and found checking firmware load information after kmod-load-reload took rather long time.
```
2021-08-05 03:33:44 UTC [INFO] checking for general errors after kmod insert with sof-kernel-log-check tool
2021 08 05 Thursday 03:33:44 UTC
declare -- cmd="journalctl_cmd --since=@1628134419"
2021 08 05 Thursday 03:33:45 UTC
2021-08-05 03:33:45 UTC [INFO] checking if firmware is loaded successfully
2021-08-05 03:33:45 UTC Sub-Test: [INFO] Checking SOF Firmware load info in kernel log

[ 4767.435085] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-1f641
[ 4767.435087] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
--
[ 4767.435095] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: used compiler XCC 12:0:8 <RG-2017.8-linux> used optimization flags -O2
[ 4767.435104] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 3 size 0x30
--
[ 4767.535751] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-1f641
[ 4767.535753] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
--
[ 4904.321166] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 1:8:0-1f641
[ 4904.321167] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:19:0 Kernel ABI 3:18:0
--

[ 4767.435091] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware debug build 65535 on dtermin.-extman - options:
                        GDB: disabled
                        lock debug: disabled
                        lock vdebug: disabled
--
[ 4767.535757] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware debug build 65535 on dtermin.-fwready. - options:
                        GDB: disabled
                        lock debug: disabled
                        lock vdebug: disabled
--
[ 4904.321171] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware debug build 65535 on dtermin.-extman - options:
                        GDB: disabled
2021-08-05 03:38:04 UTC Sub-Test: [INFO] Test Result: PASS!
2021 08 05Thursday 03:38:04 UTC
2021-08-05 03:38:04 UTC [INFO] ==== firmware boot complete: 1 of 25 ====
2021-08-05 03:38:04 UTC [INFO] ===== Starting iteration 2 of 25 =====

```
From the console , we can see a time gap of over 4 minutes , for checking SOF Firmware load info in kernel log.

Run mere sof_remove.sh and dmesg shows a log of ACPI Errors

lgirdwood, 2021-08-06:
> From the console , we can see a time gap of over 4 minutes , for checking SOF Firmware load info in kernel log.
> more investigation is needed . But DUTs offline now ,will check later

4 minutes is way too long, btw, can you retry with lock debugging enabled in the kernel (@keyonjie can show you how to enable this in kernel kconfig).

lgirdwood, 2021-08-06:
@plbossart fyi, may be FW or kernel, unsure atm. ruling out locking.

plbossart, 2021-08-06:
we already enable all this by default for the kernel

````
more lock-stall-defconfig 
CONFIG_LOCKUP_DETECTOR=y

CONFIG_HARDLOCKUP_DETECTOR=y
CONFIG_HARDLOCKUP_DETECTOR_PERF=y

CONFIG_SOFTLOCKUP_DETECTOR=y

CONFIG_DETECT_HUNG_TASK=y

CONFIG_DEBUG_RT_MUTEXES=y
CONFIG_DEBUG_SPINLOCK=y
CONFIG_DEBUG_MUTEXES=y

CONFIG_DEBUG_RWSEMS=y

CONFIG_LOCKDEP=y
CONFIG_DEBUG_LOCK_ALLOC=y

CONFIG_LOCK_STAT=y

CONFIG_DEBUG_ATOMIC_SLEEP=y
CONFIG_WQ_WATCHDOG=y

````