thesofproject/sof #4597: [BUG]ipc timed out when CORE_ENABLE, when kmod unload load on TGLU_RVP_NOCODEC_CI

XiaoyunWu6666, 2021-08-04:
**Describe the bug**
when kmod unload load for multiple rounds on TGLU_RVP_NOCODEC_CI , ipc timed out when CORE_ENABLE

**To Reproduce**
 TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tplg ~/sof-test/test-case/check-kmod-load-unload.sh -l 25

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 32657609
SOF Branch: main
SOF Commit: e2fc7082e1be
 Topology: /lib/firmware/intel/sof-tplg/sof-tgl-nocodec-ci.tpl
 Platform: TGLU_RVP_NOCODEC_CI

**Screenshots or console output**
[dmesg]
[dmesg_CI.txt](https://github.com/thesofproject/sof/files/6929351/dmesg_CI.txt)

XiaoyunWu6666, 2021-08-04:
related issues: #4552 #4586 
ipc timed out may happen in different phases

lgirdwood, 2021-08-04:
@plbossart looks like kernel issue
```
[29202.727243] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[29202.727249] sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.6.SSP2.IN setup complete
[29202.727255] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[29202.727461] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[29202.727474] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[29202.727602] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x30030000 size 12
[29202.727746] BUG: unable to handle page fault for address: ffffb495c0f2d567
[29202.727747] #PF: supervisor read access in kernel mode
[29202.727747] #PF: error_code(0x0000) - not-present page
[29202.727748] PGD 100000067 P4D 100000067 PUD 100102067 PMD 104724067 PTE 0
[29202.727750] Oops: 0000 [#1] SMP NOPTI
[29202.727750] CPU: 1 PID: 27829 Comm: rmmod Not tainted 5.14.0-rc3-daily-nocodec-20210803 #16d5c452
[29202.727751] Hardware name: Intel Corporation Tiger Lake Client Platform/TigerLake U DDR4 SODIMM RVP, BIOS TGLSFWI1.R00.3264.A00.2006251828 06/25/2020
[29202.727751] RIP: 0010:string_nocheck+0x34/0x50
[29202.727752] Code: 66 85 c0 74 3c 83 e8 01 4c 8d 5c 07 01 31 c0 eb 19 49 39 fa 76 03 44 88 07 48 83 c7 01 41 8d 71 01 48 83 c0 01 4c 39 df 74 0f <44> 0f b6 04 02 41 89 c1 89 c6 45 84 c0 75 d8 4c 89 d2 e9 15 ff ff
[29202.727753] RSP: 0018:ffffb495c0e3b950 EFLAGS: 00010046
[29202.727754] RAX: 0000000000000000 RBX: ffffb495c0e3b9e0 RCX: ffff0a00ffffff04
[29202.727754] RDX: ffffb495c0f2d567 RSI: ffffb495c0e3bb0e RDI: ffffb495c0e3bb2f
[29202.727755] RBP: ffffb495c0e3bb0e R08: ffffffff987b7d18 R09: 00000000ffffffff
[29202.727755] R10: ffffb495c0e3bb0e R11: ffffb496c0e3bb2e R12: ffffb495c0f2d567
[29202.727756] R13: ffff0a00ffffff04 R14: 0000000000000000 R15: ffffffffc07133cb
[29202.727756] FS:  00007fa163193540(0000) GS:ffff94a4e6e00000(0000) knlGS:0000000000000000
[29202.727756] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[29202.727757] CR2: ffffb495c0f2d567 CR3: 0000000144d92005 CR4: 0000000000770ee0
[29202.727757] PKRU: 55555554
[29202.727758] Call Trace:
[29202.727758]  string+0x42/0x50
[29202.727758]  vsnprintf+0x273/0x4f0
[29202.727759]  va_format.isra.0+0x6e/0xa0
[29202.727759]  vsnprintf+0x1ec/0x4f0
[29202.727759]  vprintk_store+0xa5/0x430
[29202.727760]  ? vsnprintf+0x273/0x4f0
[29202.727760]  vprintk_emit+0x6e/0x200
[29202.727760]  dev_vprintk_emit+0x134/0x158
[29202.727761]  dev_printk_emit+0x49/0x60
[29202.727761]  ? sof_ipc_tx_message_no_pm+0x6a/0x90 [snd_sof]
[29202.727762]  _dev_err+0x67/0x7e
[29202.727762]  sof_set_up_pipelines.cold+0x28/0x3a [snd_sof]
[29202.727762]  sof_resume+0xdf/0x160 [snd_sof]
[29202.727763]  pci_pm_runtime_resume+0x65/0x80
[29202.727763]  ? pci_pm_default_resume+0x20/0x20
[29202.727763]  __rpm_callback+0x3c/0x110
[29202.727764]  ? pci_pm_default_resume+0x20/0x20
[29202.727764]  rpm_callback+0x54/0x60
[29202.727764]  rpm_resume+0x536/0x740
[29202.727764]  __pm_runtime_resume+0x45/0x90
[29202.727765]  device_release_driver_internal+0x45/0x1d0
[29202.727765]  driver_detach+0x42/0x90
[29202.727765]  bus_remove_driver+0x53/0xd0
[29202.727766]  pci_unregister_driver+0x36/0x80
[29202.727766]  __x64_sys_delete_module+0x18c/0x250
[29202.727767]  do_syscall_64+0x38/0x90
[29202.727767]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[29202.727767] RIP: 0033:0x7fa1632dfa3b
[29202.727768] Code: 73 01 c3 48 8b 0d 55 84 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa b8 b0 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 25 84 0c 00 f7 d8 64 89 01 48
[29202.727768] RSP: 002b:00007ffdb496c698 EFLAGS: 00000206 ORIG_RAX: 00000000000000b0
[29202.727769] RAX: ffffffffffffffda RBX: 000055fd73843790 RCX: 00007fa1632dfa3b
[29202.727770] RDX: 000000000000000a RSI: 0000000000000800 RDI: 000055fd738437f8
[29202.727770] RBP: 00007ffdb496c6f8 R08: 0000000000000000 R09: 0000000000000000
[29202.727771] R10: 00007fa16335bac0 R11: 0000000000000206 R12: 00007ffdb496c8d0
[29202.727771] R13: 00007ffdb496e649 R14: 000055fd738432a0 R15: 000055fd73843790
[29202.727771] Modules linked in: snd_hwdep snd_sof_nocodec snd_sof_pci_intel_tgl(-) snd_sof_intel_hda_common snd_sof_pci snd_sof_acpi_intel_bdw snd_sof_acpi_intel_byt snd_sof_intel_ipc snd_sof_acpi snd_sof_intel_atom snd_sof snd_sof_xtensa_dsp snd_soc_core snd_compress snd_soc_acpi_intel_match snd_soc_acpi snd_intel_dspcfg snd_pcm ax88796b fuse asix usbnet ledtrig_audio snd_usbmidi_lib wmi_bmof snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer x86_pkg_temp_thermal intel_powerclamp snd soundcore i915 mei_me mei processor_thermal_device_pci_legacy processor_thermal_device processor_thermal_rfim processor_thermal_mbox int340x_thermal_zone intel_soc_dts_iosf i2c_algo_bit ttm drm_kms_helper wmi int3400_thermal acpi_thermal_rel drm drm_panel_orientation_quirks efivarfs intel_lpss_pci intel_lpss xhci_pci idma64 mfd_core xhci_hcd [last unloaded: snd_sof_pci_intel_icl]
[29202.727788] CR2: ffffb495c0f2d567
[29202.727788] ---[ end trace 52a1bc9b0de048a2 ]---
```

plbossart, 2021-08-04:
@keyonjie you are the owner of this 'nocodec-ci' topology file, I'll let you look into this.

XiaoyunWu6666, 2021-08-09:
duplicate as #4552 #4586   ,all happened when runtime resume

XiaoyunWu6666, 2021-08-16:
in inner daily 5882 , Tglu-rvp-nocodec-ci ipc timed out when runtime resume to record 100 sec