thesofproject/sof #4598: [BUG] `mixer_trigger()` aborts playback PAUSE trigger propagation

lyakh, 2021-08-04:
**Describe the bug**
Playback PAUSE triggers are unconditionally terminated by the mixer component.

First of all this code in `mixer_trigger()`
```
	/* don't stop mixer on pause or if at least one source is active/paused */
	if (cmd == COMP_TRIGGER_PAUSE ||
	    (cmd == COMP_TRIGGER_STOP &&
	     (mixer_source_status_count(dev, COMP_STATE_ACTIVE) ||
	     mixer_source_status_count(dev, COMP_STATE_PAUSED)))) {
		dev->state = COMP_STATE_ACTIVE;
		ret = PPL_STATUS_PATH_STOP;
	}
```
contrary to what the comment above it says, will prevent the mixer from stopping on PAUSE even if no other pipelines, using the mixer are currently active. Because capture is handled separately above that code, this only affects playback. That means, when you hit pause during playback on a pipeline with a mixer, the PAUSE command will not be propagated to components downstream from it. E.g. on UP2 I see, that the SSP DAI isn't getting PAUSE on hw:0,0.

Besides, keeping the state as ACTIVE also generates a "comp_set_state(), state already set to 5" message in the trace when `comp_dbg()` is called earlier in the same function, and that also aborts the trigger propagation.

lgirdwood, 2021-08-04:
@ranj063 fyi - looks like it's needed for low power pipes.

plbossart, 2021-08-04:
IIRC this was a feature, not a bug. When we pause a mixer we generate zeroes, isn't it?

dcbee3cdf3 (Adrian Bonislawski   2020-11-18 09:05:17 +0100 239) mixer: dont stop mixer on pause

ranj063, 2021-08-04:
@lyakh I also think this was specifically added as a feature and is not a bug

lyakh, 2021-08-04:
@plbossart @ranj063 So when the pipeline with a mixer is paused, the DAI isn't stopped, right? Doesn't that mean, that it keeps consuming power? DMA, clock, etc.?

plbossart, 2021-08-04:
Pause is usually not a scenario tracked in power KPIs.
Very few solutions actually enable pause_push/release, it's not used by CRAS/PipeWire/PulseAudio.
And in theory we shouldn't even support INFO_PAUSE in linux since we can't restart exactly at the same point.
I suggested removing PAUSE support a while back but it's supported in Windows, so long term this stays as a requirement.

lgirdwood, 2021-08-04:
> I suggested removing PAUSE support a while back but it's supported in Windows, so long term this stays as a requirement.

I think this is no longer true as I've not seen in IPC4. @mwasko can you confirm ? We could remove this and simplify some logic.

plbossart, 2021-08-04:
I suggested this back in 2019 and we decided to keep PAUSE: https://github.com/thesofproject/linux/issues/1339

lgirdwood, 2021-08-04:
> I suggested this back in 2019 and we decided to keep PAUSE: [thesofproject/linux#1339](https://github.com/thesofproject/linux/issues/1339)

Maybe time to revisit if no use by Windows is confirmed.

mwasko, 2021-08-05:
> > I suggested removing PAUSE support a while back but it's supported in Windows, so long term this stays as a requirement.
> 
> I think this is no longer true as I've not seen in IPC4. @mwasko can you confirm ? We could remove this and simplify some logic.

There are three pipeline states that can be set by the host in Windows:

RESET Resets pipeline.
PAUSED Pauses pipeline.
RUNNING Starts pipeline.